openscad/openscad #1594: Default function arguments are evaluated for every function call

kintel, 2016-03-03:
Consider the following example.
The default function argument ```(arg = expensive())``` is evaluated for each function call, making the apparently trivial function calls take a lot of time.
In this example, we wouldn't need to evaluate the default value at all, but even evaluating it only once would be an improvement.

When fixing this, keep in mind that functions can be called before they're defined, as all function definitions are bound before assignment expressions are evaluated.
```
function expensive(n = 20) = n == 0 ? 1 : (expensive(n - 1) + expensive(n - 1));
function myFunc(arg = expensive()) = arg;

echo(myFunc(1));
echo(myFunc(2));
echo(myFunc(3));
```

kintel, 2016-03-06:
Another use case discussed on the [mailing list](http://forum.openscad.org/children-and-extra-unwanted-evaluations-tt16291.html) is calling ```rands()``` to get the default parameter

doug-moen, 2016-03-14:
Evaluating default argument expressions on every function call is the correct behaviour for the general case. We want dynamically scoped variables (like $foo) to be looked up at function call time, not fixed at the time the function is defined, or the first time it is called.

Evaluating a default argument expression at definition time is an optimization that is only safe if we can prove that the expression does not depend on the value of a dynamically scoped variable. We could perform this optimization at compile time if we had infrastructure for compile time optimization. Or we could optimize this at run time by caching the value after it's computed the first time, if we had some kind of run-time optimization infrastructure (like in a JIT compiler) for proving at run time that an expression doesn't depend on dynamic variables.

The forum comment about calling rands() to specify a default parameter value is not right, IMO. The rands() function is broken: it's an imperative function with side effects in a declarative language. I don't want to break the semantics of default argument values to work around the broken semantics of rands(). My proposed replacement for rands() is called random(): it's a deterministic function that hashes its argument together with $seed to return a pseudo-random number, also described in the same forum thread. It would make sense to call random() to provide a default argument value, but then it would be necessary for the default argument value to change whenever $seed changes. That's because the result computed by a function that uses random() must change whenever $seed changes: it's part of the contract of random().

closed 2025-03-09 by jordanbrown0