openscad/openscad #1596: Instantiating children() multiple times also re-evaluates the children

kintel, 2016-03-06:
The following example shows that when instantiating children more than once, the children are re-evaluated, resulting in potentially different geometry.
This sounds counter intuitive and may be considered a bug - more investigation needed.

```
module TwoChildren()  {
  translate([10,0,0]) children();
  children();
}

TwoChildren() cube(rands(1,10,1)[0]);
```

davidson16807, 2016-03-07:
I think there are certain circumstances where preserving this behavior is critical. One such circumstance occurs when a module renders differently depending on the state of a special variable. I rely on this to implement the [attach](https://github.com/davidson16807/relativity.scad/wiki/CSG-operations#attach) function in relativity.scad.

	module TwoChildren()  {
	  translate([10,0,0], $class="foo") children();
	  children();
	}
	
	module ChildBase(){
	  if($class=="foo")
	  	cube([1,2,3]);
	  if($class!="foo")
	  	cube([4,5,6]);
	}

	TwoChildren() ChildBase();

doug-moen, 2016-03-08:
I am surprised that setting $class in the translate() arguments sets a dynamically scoped variable that's visible in the children.

The more obvious place to set $class is in the children() call. But if you rewrite the code to use children($class="foo") and children($class="bar"), then it doesn't work at all, and you get an error message. The error message suggests to me that this is not intended to be a supported feature.

In OpenSCAD2, functions and modules are unified into a single concept, and translate([10,0,0]) becomes an ordinary function call that returns a transformation value, which is another function. For example,
```
   translate([10,0,0]) cube(10);
```
becomes equivalent to
```
   t = translate([10,0,0]);
   t cube(10);
```
and t is a function. In this implementation, translate([10,0,0], $class="foo") isn't going to do anything useful; at best the $class parameter is ignored.

If we want to officially support what the relativity library does in OpenSCAD, then we should extend the children() module to accept dynamically scoped $foo= arguments.

However, the current design of OpenSCAD2 has no way to support this use case. The children() module is deprecated, and no longer works the same way internally. Modules and functions are unified into a single concept, which means they use the same argument passing mechanism, which means that shape arguments to modules are not reevaluated each time they are referenced.
* When a function is called, the argument expressions are evaluated to produce an argument list of values. Then the function is entered, the argument values are bound to formal parameters, and the body is evaluated.
* With modules, it's the same thing, since they are compiled into functions. The children are first evaluated into a shape or group, and then the function is entered, the children are stored in a parameter slot (which can be referenced by children()), and the body is evaluated.

One more example. In OpenSCAD2, shapes are values, and can be assigned to variables, and
```
TwoChildren() {
  if($class=="foo")
    cube(10);
  if($class=="bar")
    sphere(d=10);
}
```
is equivalent to
```
kids = {
  if($class=="foo")
    cube(10);
  if($class=="bar")
    sphere(d=10);
};
TwoChildren() kids;
```
OpenSCAD2 has compositional semantics; these two fragments are required to be equivalent. I'd expect a simple optimizing compiler to compile both into the same code.

So my question is whether relativity.scad is supported in future versions of OpenSCAD, and in particular, is it supported in OpenSCAD2?

doug-moen, 2016-03-08:
Relativity.scad seems to use $class to pass "style sheet" information from a parent node in the CSG tree, down to its children. It turns out that I need a mechanism to accomplish the same thing in my F-Rep geometry kernel, so I personally need to find some way to express this in OpenSCAD2.

davidson16807, 2016-03-08:
@doug-moen Well, there's actually a few "private" special variables that work behind the scenes to pass the style sheet information and record ancestry information. `$class` doesn't have to be a special variable, but the other private variables do have to be special for relativity.scad to work, so all the points you make above still apply.

jordanbrown0, 2024-02-08:
I use the ability to pass $ variables to children to allow an outer module to pass information to its children.  (My "wall" module can take children that create trim around windows and doors; it passes the window/door dimensions in a $ variable.)

jordanbrown0, 2024-11-27:
I think it's pretty well established that people regard this as a feature.  @kintel , if you agree you might close it.

closed 2024-11-27 by kintel