openscad/openscad #1597: SVG export no longer includes width or height

thinkl33t, 2016-03-10:
As of commit ab338a2c20bf7e94317d43a93e1c013fc531c629, where the export formats were split into separate files, openscad is no longer adding the width or height to exported SVGs.

As mentioned in this forum thread http://forum.openscad.org/SVG-Export-td12216.html ,  If SVGs are exported using openscad, other software will generally assume the SVGs are in pixels (which they normally aren't for laser cutter work - SVGs are commonly preferred for lasers as they are nice closed paths).  

Previously I was able to work around this issue by opening the SVG in a text editor and adding 'mm' to the end of the units in the width and height tag.  This causes all the SVG software to assume all other measurements are in mm.

Any chance of getting this back, and can we reopen the discussion on adding a "Default SVG export units" option somewhere? (I should probably open a separate issue for that?)

t-paul, 2016-03-10:
Oh, that change was probably not intentional. As for the options, that should be covered by #649.

closed 2016-03-10 by kintel

JamesNewton, 2017-10-23:
I'd like to point out that /units/ are not added to height and width which was mentioned by the OP but not addressed with the issue was closed. e.g. we now get:
```html
<svg width="10" height="10" ... />
```
but we really should get:
```html
<svg width="10mm" height="10mm" ... />
```

nophead, 2017-10-23:
That is what the latest version does: https://github.com/openscad/openscad/blob/b48b525adeeaa5a320c5d81ec575a9ae83cf0987/src/export_svg.cc#L81

PR #1600 (2016-03-10): Restore width & height parameters and use mm in SVG header
Fixes  #1597.

This also sets the export unit to mm which seems to be a bit more useful than the default pixel units.
Ultimately this (and some additional options) should be user configurable (see #649).
commit 6ecf8ed3 (2016-03-10): Restore width & height parameters and use mm in SVG header (fixes #1597).