openscad/openscad #1604: Meta data state

Gregwar, 2016-03-21:
Hello,

I think the meta-data case is summed up well here:
https://github.com/openscad/openscad/wiki/Meta-Data-Use-Cases

There is multiple proposals of ways to add meta-data in scad files, but:
* What will be the correct one chosen to be released?
* What is the roadmap for it?

Gregwar, 2016-03-21:
If there is no already chosen proposal, here is a new one:
https://github.com/openscad/openscad/pull/952#issuecomment-199342162

kintel, 2016-03-22:
To answer your questions: There is no roadmap. We're mostly waiting for enough time on our hands to properly design and implement something.

Gregwar, 2016-08-18:
Any news on this?

Gregwar, 2017-05-02:
Bump, is there any news on the state of adding Meta data in scad files?

kintel, 2017-05-03:
There was a bunch of discussion on this topic in #1608 a while back, but nobody has had enough time to come up with a good mechanism covering the use-cases.
Another idea which came up was to try to enable experiments like this to live in user-space, e.g. by offering some sort of lower-level API to traverse the AST or CSG trees. There are still lots of unknowns in that domain though.