openscad/openscad #1605: Wrong instruction execution order and variable reset in loops

plushvoxel, 2016-03-24:
I'm using openscad 2015.03.02-3

Executing the following code:
``` scad
left = ["CTS", "RTS", "RXD", "TXD", "DSR", "DTR", "GND"];

difference(){
    square([15.5, 34.5]);
    pos = [1, 3.75];
    for(i = left){
        echo(pos);
        echo(i);
        translate(pos)
        text(i, 2, "Nimbus Mono", valign="center");
        pos = [pos[0], pos[1]+2.54];
        echo(pos);
    }
}
```
I get the following output:
```
ECHO: [1, 6.29]
ECHO: "CTS"
ECHO: [1, 6.29]
ECHO: [1, 6.29]
ECHO: "RTS"
ECHO: [1, 6.29]
...
```
Where I would expect something like:
```
ECHO: [1, 3.75]
ECHO: "CTS"
ECHO: [1, 6.29]
ECHO: [1, 6.29]
ECHO: "RTS"
ECHO: [1, 8.83]
...
```

Do you have any idea what this could be?

kintel, 2016-03-24:
All assignments in an openscad block is hoisted to the start of the block, so ```pos``` will have the same value everywhere in that block. You can give it another name if you want to reference two different values.

doug-moen, 2016-03-24:
OpenSCAD is not an imperative language with mutable variables, which is
what you are used to if you normally program in Python, Javascript, etc.

Instead, OpenSCAD is a declarative language where each variable is defined
exactly once. Think back to high school algebra. Imagine that you have the
equation
   x = x + 1
and you are solving for x. This equation defines a number 'x' which is
equal to 1 + itself, and there is no such number. It is the same in
OpenSCAD. A definition like 'x = x + 1' makes no mathematical sense, and
OpenSCAD is a mathematical language.

In your program, you think you are initializing 'pos' to an initial value
outside the loop, then updating its value on each iteration, but that is
not what OpenSCAD sees. Instead, OpenSCAD sees two different definitions of
'pos' which denote different variables. One definition is outside the for
loop. The other definition is inside the for loop, and the inner definition
is redefined on each iteration. The inner definition of 'pos' shadows the
outer definition.

You need to use a different style of programming.

Try something like this:

doug-moen, 2016-03-24:
My mailer screwed up when I pasted the example code. One more try:

left = ["CTS", "RTS", "RXD", "TXD", "DSR", "DTR", "GND"];

difference(){
    square([15.5, 34.5]);
    for (i=[0:len(left)-1]) {
        pos = [1, 3.75 + i*2.54];
        translate(pos)
          text(left[i], 2, "Nimbus Mono", valign="center");
    }
}

closed 2016-04-02 by kintel