openscad/openscad #1607: Canceling some exports reports to console

MichaelAtOz, 2016-03-24:
Residual issue from closure of #1552 

When selecting File/Export/(the 3D formats), clicking cancel does just that, and nothing else.
However for 2D and image a message is logged to the console:

Image - No filename specified. Image export aborted.
CSG - No filename specified. CSG export aborted.
DXF - No filename specified. DXF export aborted.
SVG - No filename specified. DXF export aborted. (sic)

a. The cancel button was pressed, not 'no filename specified' - as it was pre-filled.
b. Regardless I don't think it should log anything for a cancel.

Remove messages.

MichaelAtOz, 2016-03-24:
Consider what happens for command-line export?

kintel, 2016-03-25:
@MichaelAtOz Do you build yourself or are you testing using the latest Windows snapshots?

closed 2016-03-25 by kintel

MichaelAtOz, 2016-03-25:
Using latest Windows snapshot 2016.02.09/64.