openscad/openscad #1608: color(), material() and part()

doug-moen, 2016-03-25:
This feature request generalizes our export mechanism so that we can export models with:
* multiple named parts (which are allowed to overlap). For AMF, this means multiple &lt;object> elements. For DXF, and perhaps also SVG with Inkscape extensions, this means multiple layers. For STL and OFF, this means multiple output files are created by a single export.
* multiple colors and materials. The goal is to support 3D printing with multiple extruders, so colors are volumetric, rather than applied to the surface of a mesh. A single 3D part may have multiple non-overlapping volumes, each volume having a different color or material. A single 2D part may have multiple non-overlapping areas, each area having a different color. For AMF, this means each &lt;object>/part has multiple &lt;volume> elements. For DXF and SVG, each layer has multiple non-overlapping shapes, each of which can have a different solid color. For STL and OFF, this means multiple output files are created by a single export, one for each color/material.

The reason for bundling these three features together into one FR is to show how they interact with each another, and with the CSG operators. I also want to make the point that these 3 features are not special cases of some more general "metadata" feature. Different kinds of "metadata" interact with the CSG operators in different ways, depending on the role of that "metadata" within the system. For example, OpenSCAD2 has an "object" data type, which can be interpreted as a way to attach "metadata" to a shape. But the purpose and use of that metadata is different, and CSG operators must ignore OpenSCAD2 object attributes, while they must pay attention to the color(), material() and part() attributes. The CSG operators interact with the part() attribute differently than they interact with the color() and material() attributes.

`part("name")shape` associates the same part name with all of the components of `shape`, overriding any previously specified part names. If no part name is specified, the part name defaults to `""`. So `square(10)` is equivalent to `part("")square(10)`.

The concept of a 2D or 3D shape is generalized so that each shape has zero or more distinctly named parts, which may overlap. (As a special case, the empty OpenSCAD script produces the empty shape, which has no parts.) The CSG operations are generalized so that each distinct named part contained within the children list is operated on separately, and thus retains its identity.

For example,
```
union {
   part("p1") cube(10);
   part("p1") sphere(5);
   part("p2") cylinder(h=15,r=2);
}
```
This produces a new shape containing two parts named "p1" and "p2". The cube and sphere are unioned together into a single composite shape, since they belong to the same part. However, the cylinder is not unioned with anything, it retains its identity since it belongs to a separate part.

As a future extension, we'll be able to import an AMF file containing multiple &lt;object> elements, and preserve the identity of each &lt;object> as a separate part. Likewise for importing a DXF file with multiple layers.

A part name is zero or more characters chosen from the set [a-zA-Z0-9_]. That's intended to be a lowest common denominator design that will be compatible with all export formats.

The color() operator already exists, but now it is extended so that it affects the F6 rendering semantics. Shapes are generalized so that each part contains one or more disjoint regions, each region having a different colour. The CSG operators are generalized so that, when an operator is passed two shape arguments within the same part that have different colours, then the result is computed as some sort of composite, usually containing multiple regions with different colours. But these semantics vary from one operation to the next.

Colours are volumetric, and are considered to apply to each geometric point within a region, both at the surface and within the interior. If no colour is specified, then it defaults to `undef`. For example, `cube(10)` is equivalent to `color(undef)cube(10)`. You can remove all colours from a shape using `color(undef)shape`. This distinction matters for export: if a region has no colour, then no colour is specified for that region in the exported file.

For example, the `union` operator is extended so that the order of the shape arguments matters. If two shape arguments overlap, and belong to the same part, but have different colours, then the colour of the shape closest to the beginning of the argument list takes precedence for the colour of the overlapping region in the final result.

The material() attribute is far less important than part() or color(). I'm only suggesting it because the AMF file format allows you to specify a colour, a material, or both, for each volume in an object. The material attribute doesn't provide any benefit for any other currently supported export file format. We could leave it out without losing much. I'll update this paragraph later based on the discussion.

kintel, 2016-03-25:
Thanks for the write-up doug - I'm interested in moving this forward and this is a good first step.
A few comments:
* multi-body STL (ASCII only) appear to be reasonably well supported by external software. We should consider doing the same, as a user-selectable export option. Similarly for the future extension of preserving part names on import.

* Have you considered alternative management of hierarchical naming? e.g.
```
MyModule() {
  part("p1") cube();
  part("p2") sphere();
}
part("ObjectA") translate([-10,0,0]) MyModule();
part("ObjectB") translate([10,0,0]) MyModule();
```

* This has some similarities to marker nodes (#952). Should we extend the .csg file format to include part names?

doug-moen, 2016-03-26:
@kintel:

Part names are included in the CSG tree: they must be, since they change the semantics of CSG operations. Part names should appear in the *.csg output.

However, marker nodes (#952) are different from part names. The key is that marker nodes have no effect on CSG semantics. Marker nodes satisfy the same use case as OpenSCAD2 object attributes. They permit model- and application-specific annotations to be added to the node of a model: annotations which have no meaning to built-in operations. For example, a wheel component could have metadata giving the wheel radius, the width, the rim thickness, the number of spokes.

Hierarchical part names sound cool, but it's an extra complication that needs to be justified by use cases. It's something that could be added later: I've rigidly restricted the part namespace; a hierarchy separator character could be added later as an extension. Also, we'd need to look at each import and export format and see how hierarchical part names map in each case.

t-paul, 2016-03-26:
Why does the actual use of the meta data need to influence the syntax? All meta data will need to be in the CSG tree anyway. Some would be only exported like the markers or BOM information and other can be handled either by the core (color), or the GUI (parameters, ...). In all other cases we try to move into the direction of unifying things, it seems strange not to do that for the meta data.

doug-moen, 2016-03-27:
@t-paul: I don't really understand your comment "Why does the actual use of the meta data need to influence the syntax". What syntax? And why do you think that my proposal fails to move in the direction of unifying things? After all, I've unified #1041 and #1044 into a single more general concept of "part".

t-paul, 2016-03-27:
Using the `part()` syntax is only possible for module instantiations, not for variable, function or modules definitions. At least I don't see how that would be possible.

doug-moen, 2016-03-27:
I don't agree that the proposed color() and part() operators have anything to do with metadata. [Maybe you could claim that the *old* color() operator is metadata, since it has no effect on the model, only on the preview. But the *new* color() operator proposed here isn't metadata.]

The AMF file format has support for metadata via a &lt;metadata&gt; element. You can specify a textual description, author, copyright information, etc. This is clearly metadata, since it is information *about* the data, and it has no semantic effect on the model. It has no effect on what actually gets manufactured.

Similarly, if we can tag a module definition with a human readable explanation of what the module does, that is also metadata, since it has no effect on the operation of the module.

But in this proposal, color() and part() convey semantic information that is part of the model. They directly affect what gets manufactured. They aren't metadata. A corollary is that they do need to be included in the *.csg file.

doug-moen, 2016-03-27:
@t-paul: The part() operator is a module; it takes a shape as an argument, and returns another shape. Ditto for color(). Semantically, they have the same domain and range as other modules like translate().

doug-moen, 2016-03-27:
t-paul: Are you suggesting that the part() operator should be applicable to variable, function or modules definitions? If so, this makes no sense to me. In my mind, part() is a geometric operation. It sounds like you may have a counterproposal that is radically different from mine.

kintel, 2016-03-27:
Some thoughts on the confusion of model attributes vs. metadata. I'm not sure if this really addresses the confusion/comments, but it's an attempt:

Essentially,  ```part```, ```color``` and ```material``` are all used to attach some sort of attribute to a node in the CSG tree. This proposal suggests some semantics for how this affects the underlying geometry, possibly hierarchically.

Now, we _could_ view ```part```, ```color``` and ```material``` as pure attributes, and also allow other metadata attributes. We then define that some attribute names have built-in behavior which affects the design in various ways. As a side note: One could argue that all transformation nodes are really just attributes as well. This could be useful in the future if we want to associate transformations with objects in a GUI.

This leads me to two issues:
* How do we attach such other metadata attributes to CSG tree nodes?
* How do we attach attributes to non-CSG tree nodes (e.g. modules, functions, variables)?

I think @t-paul's comments relate to these two issues.

..so the question to @doug-moen would be: Can we (later on) solve those two issues in a clean way, allowing us to move forward without doing it all at the same time?

doug-moen, 2016-03-28:
@kintel: The proposed API for the part(), color() and material() modules is a pure functional interface that is completely abstract. This means the interface is independent of the underlying implementation. The underlying implementation could change in the future, and we wouldn't be forced to change the API or cause old code to stop working.

As soon as we extend OpenSCAD to allow arbitrary attributes (name/value pairs) to be attached to any entity, then the problem of data abstraction arises. There is now a distinction between stored and computed attributes. "Stored" attributes are stored directly in the data structure as name/value pairs, while "computed" attributes are computed from the values of other attributes when they are requested. We could decide to expose the underlying implementation, by requiring the user to use different syntax for stored vs computed attributes. Or we could hide the underlying implementation by providing an abstract, pure functional interface for accessing the attributes.

kintel, 2016-03-28:
@doug-moen I don't quite follow your on what a "computed attribute" is..

t-paul, 2016-03-28:
Yes, I'm not talking about the behavior, e.g. how color will be handled by the CSG operations.

In my view the color is a property of a specific node / mesh, and has no node identity in itself. Using `part()` and the already existing `color()` it create a new node and applies it's properties to the child nodes (that's what's visible to the user writing the code, yes we could hide this internally, but that's not my point). I'd prefer to see in the code and enforced by syntax that the information is only attached to a node, just like the '#' modifier right now.

The second point is currently coming from the Customizer topic which wants to annotate variables. I don't know if it's possible to handle `param(desc = "box width", values = [1:6]) size = 3;`. Also having no way to directly distinguish meta data from other things makes it more complicated to read and understand the code.

Also I think all meta data must go into the CSG file, that's the whole points of most of the meta data attributes (e.g. for the marker nodes). We will handle some of it internally, e.g. the GUI would handle the export related values. Basically every processor will just ignore all the attributes it does not know.

kintel, 2016-03-28:
@t-paul Just a small comment on modifier characters: My current view is that this is syntactical sugar for setting a modifier attribute on a node. We could offer a full syntax for the same if we offer a general purpose attribute module.

kintel, 2016-03-28:
Since customization UI is a much larger topic, I'd like to explore if we could decouple this proposal from the customization parameter annotations and let these concepts peacefully co-exist. In the future, they could potentially use the same underlying mechanism, or even use the same language front-end.

I think moving all these features forward in sync is a bit utopic with the current resource situation. ..so if we can see a clean break I think we should go for it.

t-paul, 2016-03-28:
Yes, the modifier `#` could be a short-cut, but what I mean is, it's attached to the node, which is why it's not possible to write `# { cube(); sphere(); }` it needs a node to attach to, e.g. `union` and becomes an attribute of that node.

doug-moen, 2016-03-28:
@t-paul wrote "Yes, the modifier # could be a short-cut, but what I mean is, it's attached to the node, which is why it's not possible to write # { cube(); sphere(); } it needs a node to attach to, e.g. union and becomes an attribute of that node."

See, this is an example of something that I consider a bug in OpenSCAD. As a user, I don't care about what a "node" is in the underlying C++ implementation. I just want things to work. The fact that this doesn't work because "it's not a node" is not something I want to hear. This is something that I had proposed to fix in the OpenSCAD2 design document.

Similarly, I don't want to be forced to care about how colours and parts are implemented internally.

I prefer the language to be simple, general, high level, abstract, and not shove implementation details into my face when it's not necessary.

kintel, 2016-03-28:
@t-paul ah, ok, _now_ I understand what you meant an hour ago.

doug-moen, 2016-03-28:
@t-paul wrote "Yes, the modifier # could be a short-cut, but what I mean is, it's attached to the node, which is why it's not possible to write # { cube(); sphere(); } it needs a node to attach to, e.g. union and becomes an attribute of that node."

Right now, I can write
```
color("red") {square(10);circle(5);}
```
and it works.

From your comments about color(), "I'd prefer to see in the code and enforced by syntax that the information is only attached to a node, just like the '#' modifier right now.", it seems you would like to introduce a new syntax for color in which the above code will no longer work.

kintel, 2016-03-28:
I tend to agree with @doug-moen. Just because ```part``` doesn't create a traditional CSG node doesn't mean that it behaves differently than a CSG node from a user's perspective. The only place a user gets a whiff of what a "node" is, is when we save to a .csg file.

You can think of it as creating a "part node" and add the underlying nodes as children (this might very well also be the cleanest way of implementing it in the current codebase).

This is an implementation detail, until we write to file. When writing to file we need to transform our "node" concept into something compatible with each file format.

t-paul, 2016-03-28:
Well, I guess we will just not agree on what's easier to read. That's not a big deal.

Back to the main question "how do I do that for module definitions and variables"? Having an annotation style syntax like a number of newer languages provide seems both useful and works for all cases I see right now.

doug-moen, 2016-03-28:
The original implementation of `color()` did not support an alpha channel, and did not support symbolic color names. So the original interface was `color([r,g,b])`. When we upgraded the language to support those things, we extended `color()` so that you can write `color("red")`, `color("red",0.5)` and `color([r,g,b,a])`. The original interface still works for backward compatibility.

If we stop using an abstract, functional interface for specifying colour, and use "attribute" syntax instead, then does it become impossible to enhance the interface in a backward compatible way, as we did for the `color` module?

kintel, 2016-03-28:
@doug-moen I my mind, attribute specification is still strongly rooted in the current syntax, making these two equivalent (note: attribute syntax popped off the top of my head for the sake of the argument):

```
   color([r,g,b]) someShape();
```

```
   attribute('color', [r,g,b]) someShape();
```

-> does that make sense?

@t-paul:

**When** we have a fully featured attribute syntax, including the proposed way of attaching any user-defined object dict ("POJO") to a shape, I think we would be close to satisfying @t-paul's request for a Customizer UI syntax. To make that complete, we need a way of attaching attributes to non-shape objects, which shouldn't be a problem as these attributes are really only currying the underlying value. We might, in addition, need some syntactical sugar for making this all more readable (e.g. the proposed javadoc-style syntax).

See a few messages above: If we can agree that this feature is orthogonal (for now) to the Customizer UI stuff, that would make it a lot easier to push this forward.

bobc, 2016-04-11:
FWIW, I am entirely behind @doug-moen 's proposal. In my experimental Carve-CSG fork I implemented the same semantics, and I believe is the only way it can work.

It is important to realise that once you go to a volumetric representation, _the color/material is as fundamental part of the node as the geometry_. The two cannot be separated, therefore it is not possible to view it as an attribute. Unless you also regard the geometry as merely another attribute.

It is easy to get lost in semantics, but unless you can get over the mental hurdle of what volumetric semantics implies then no progress can be made.

One slight quibble, I think that AMF specifies non-overlapping volumes. In practice, this may not be too much of a problem, because slicers already handle overlapping STL files by using a priority scheme. So unless they are very strict, they could/are likely to apply priority scheme to AMF objects as well.

Otherwise, I'm fully with it.

doug-moen, 2016-04-11:
Thanks, Bob, I based the color() proposal on your work. Maybe your fork can
be used to create the color() implementation.

In AMF, there is a two level hierarchy. There are multiple named objects
that can overlap. Within an object, there are multiple non-overlapping
volumes.

>

guowcas, 2017-04-23:
@doug-moen : it looks like an assembly tree to me.

doug-moen, 2017-04-24:
@guowcas: it's not an assembly tree, because there isn't a tree. The parts
are disjoint, they are not related by parent/child relationships. Maybe
what's confusing is my choice of the word "part" to describe the partition
of a model into disjoint, non-overlapping pieces which can be independently
exported.

An assembly tree would be a nice feature to have, but it's an entirely
separate feature from the "parts" feature. The closest we have to an
assembly tree right now is the CSG tree, but you can't annotate it with
high level information.

guowcas, 2017-04-24:
Sure it's not an assembly tree yet, but maybe a tree like structure should be the final form of this evolution. However, I agree that a loose collection of parts is suitable for 3D printing by now since high level information is not required yet. 

best regards, 
from GuoW's iPhone

alromh87, 2020-10-18:
Hello, any help needed here?

t-paul, 2020-10-18:
Yes, there's a number of topics to solve to get this feature in. The 2 main parts would be the ability to pass along color information inside the geometry engine and defining how the language extension can look like.

traverseda, 2021-01-24:
I wrote a short script that lets you export multiple parts, figured this is the issue to mention it in as anyone searching for a way to do this will find it. It doesn't handle basically any of these edge cases and is only the most simple example, but it solved a problem for me.

http://localhost:8000/code/partsScad/index.html

tcurdt, 2021-01-24:
@traverseda that's a link to `localhost`

traverseda, 2021-01-24:
@tcurdt I have no idea what you're talking about :p

Thanks.

cathalgarvey, 2021-04-27:
This issue is tracking two things, but are they closely related in implementation? I'm just wondering about colour support - I discovered today while hacking this little thing up:
https://git.hackers.town/seachaint/sheetstruder
..that colours are not supported in the output STLs. It looks like this is a long-standing thing without a clear priority, so I'm just wondering should I expect this limitation of OpenSCAD to change or not. :)

I don't actually have a multi-extruder printer in any case, but it would be nice if I could export models for gamedev or something, if I ever got the time to actually get into that.

doug-moen, 2021-04-27:
@cathalgarvey The initial description describes very clearly why the `colour` and `part` are closely related in implementation. Just read it, please.

> it would be nice if I could export models for gamedev or something.
But in game dev, coloured models usually have photographic images texture mapped onto the triangles in the mesh. Or, alternatively, some mesh file formats like X3D allow you to specify a colour for each face, or a colour for each vertex. This face colouring is quite different from the volumetric colour model being proposed here, where each volumetric part in an assembly has a single colour attached to it.

This proposal is designed for multi-extruder FDM 3D printing, which has very different requirements than gamedev.

gingerbeardman, 2021-05-12:
+1 for material

my use case would be to mark certain objects as reflective so they don't go black at some angles (given fixed lighting direction see #3339)

nophead, 2021-05-12:
Or too bright at some angles. Threads look too bright compared to their base material because they face the light.

gingerbeardman, 2021-05-12:
In my situation I don't mind if they blow out to white because after rendering I convert to black and white anyway. I just want all my whites to stay white, and not become some level of gray that will sometimes be crushed to black.

KeithSloan, 2021-11-12:
+1 for material. I would like to convert OpenSCAD shapes to GDML solids. The only real thing stopping this is the lack of a material name that would be in the CSG definition.

This is a very old requirement request like five years at least, what is the likelyhood of an implementation.

Happy to donate if material name is addressed.

traverseda, 2021-11-12:
@KeithSloan I was using my code above to convert stuff to GLTF, although that part of the code was never quite completed.

Here's the blender code I used to convert the files to GLTF, although it currently doesn't deal with multi-parts.

```
import sys
argv = sys.argv
infile, outfile = argv[argv.index("--") + 1:]

import bpy
bpy.ops.wm.read_factory_settings(use_empty=True)
bpy.ops.import_mesh.stl(filepath=infile)
material = bpy.data.materials.new(name="base")
material.use_nodes = True
principled_bsdf = material.node_tree.nodes['Principled BSDF']
principled_bsdf.inputs[0].default_value = (0.01, 0.2, 0.0, 1.0)
for o in bpy.context.editable_objects:
    o.data.materials.append(material)
bpy.ops.export_scene.gltf(filepath=outfile)
```

The problem with my approach is that it's slow, it needs to invoke a new instance of blender for every single part. It's always going to have that limitation and I'm not inclined to develop it further because it is a terrible hack happening at the wrong level in openscad.

Still, maybe the general approach will work for your use case? Let me know if you have any questions about it.

KeithSloan, 2021-11-12:
Thanks but it does not work for me. If material was added to the scripting language, then I would use FreeCAD OpenSCAD importer to import the SCAD or CSG file. The FreeCAD OpenSCAD workbench runs the code to produce the CSG file under the covers. The plan would be to then use FreeCAD GDML workbench to export the model to GDML. Trouble is the objects need a GDML material allocated and this currently would have to be a manual process through the Gui, where as if one could specify a material when scripting things would be a lot easier. What is GDML used for? Its used for  Montecarlo simulation using  applications such as Geant4. Medical applications, satellites and stuff that goes into space need to be modelled to make sure sensitive parts are protected in a space environment. Some people design such stuff with CAD and export to Tessellated Objects but modelling then takes a long time where as people could script with OpenSCAD. GDML has a richer set of Computer Solids than OpenSCAD so it should not be a problem converting. Hull & Minkowski are a challenge but rich models can still be created without them or they could be converted to Tessellated. One of the guy's that starred the GDML repro was a MSc student designing a CubeSat. Now would it not be cool to have CubSat's and other stuff going to space, designed with OpenSCAD.

pca006132, 2023-10-31:
Note that manifold supports vertex properties which are preserved across boolean operations. This is used in the [web demo](https://manifoldcad.org/) for color example (Tetrahedron puzzle):
[image]

We also support finding the original mesh id of different faces via mesh relation.

There are some issues (https://github.com/elalish/manifold/issues/536) regarding vertProperties in manifold because it is less tested, but we can easily fix them. I can perhaps try to integrate it in OpenSCAD...

t-paul, 2023-11-02:
> We also support finding the original mesh id of different faces via mesh relation.

So this could also be used for viewport-selection -> select in code editor, like we have for the preview. That would be very nice.

jsteinhofff, 2024-09-13:
> I wrote a short script that lets you export multiple parts, figured this is the issue to mention it in as anyone searching for a way to do this will find it. It doesn't handle basically any of these [...]

very easy and gets the job done, thanks for sharing @traverseda !

gsohler, 2024-09-13:
Or you could approach a single-program solution
[image: image.png]

This is openscad with builtin python, and you get a 3mf file with both
parts included as a bonus
I am still happy to share

jsteinhofff, 2024-09-13:
The image is broken for me, but I assume you pointed to pythonscad, which I just discovered in parallel coming across a reddit post from you. Love it, will give it a try asap, I am a big fan of python anyway!

gsohler, 2024-09-13:
sorry for that. let me try it better

try this link:
https://imgur.com/a/heUN84M

Code is:

from openscad import *

rube=cube([10,10,10]).color("red")
blegg=sphere(10).resize([10,10,20]).translate(
    [5,5,10]).color("blue")

export(
    {"rube":rube,"blegg":blegg},
    "combined.3mf")

show([rube, blegg])

cheers

jsteinhofff, 2024-09-19:
finally i found some time to look into this... lots of chaning the order of lines, few less curly brackets, few more comma ... it works!
Amazing work @gsohler , this boosts OpenSCADs capabilities by a huge margin :star_struck:
ps: i have to mention that i crashed OpenSCAD few times along the way, i will think a bit about it and give notice in your fork

rob-miller, 2024-11-03:
It seems like someone should reference this April, 2023 thesis here: [Multi-Material Support in OpenSCAD](https://tams.informatik.uni-hamburg.de/publications/2023/BSc_Nicolaus_Johannes_Leopold.pdf).

His implementation is at [https://github.com/nle1990/openscad-mm/](https://github.com/nle1990/openscad-mm/).

t-paul, 2024-11-03:
> His implementation is at https://github.com/nle1990/openscad-mm/.

Oh, my that's not even a fork :cry:. I hope at least the analysis will be useful, it looks like quite some effort went into the discussion of behavior.

jordanbrown0, 2025-04-17:
In #5785, @jimb15122 said:
> May I suggest using the otherwise useless (for stl export) COLOR tag as the basis for the split. the resulting stl files would append the color (possibly in hex format) to the base filename. UI wise, all that would be needed is a new "export stl by color" option.

The problem is that the rendering engine historically hasn't, and still doesn't quite, maintain volumetric information about color.

With the old CGAL engine, all color is lost.  With the new Manifold engine some color is retained, but (a) I don't believe it's volumetric - that is, the faces are colored, not the interior, and (b) based on a model I'm working on at the moment I don't think it's quite right yet.

The previewer shows color, but it cheats.

jimb15122, 2025-04-18:
I am by no means familiar with openscad internals.  But recently, I have been working on a Blender/Python script that reads a csg, and populates a blender scene.  It uses a stack to push and pop colors (and transforms) as they go in and out of scope. Then as objects are encountered, they get assigned a material based on the current color on the stack.   My point is (npi) that color is simply a state, and does not need to be intertwined with geometry.

pca006132, 2025-04-18:
I just don't understand why the geometry engine has to know the volumetric information. Mesh is not supposed to maintain this. Can't we just process each material individually? E.g. maintain a mesh for each material, as well as a final aggregated mesh so we can perform subtraction/intersection correctly.

jordanbrown0, 2025-04-18:
```
color("red") cube(10, center=true);
color("green") cube(10);
```

There is a 5-unit cube that is inside both the red and green cubes.  What color is it?  On the display, it doesn't matter, but when we turn it into plastic what color is it?

```
difference() {
    union() {
        color("red") cube(10, center=true);
        color("green") cube(10);
    }
    rotate(-135) translate([-15,0,-10]) cube(30);
}
```
Now, with one half cut away, you can see that formerly-interior cube.  What color is it?  Again, what color of plastic gets laid down?

```
intersection() {
    color("red") cube(10);
    color("green") cube(10, center=true);
}
```

What color is the result?  What color plastic is laid down?

jordanbrown0, 2025-04-18:
Another way to look at it is that when you do a boolean operation (notably union; the others are easier) between meshes of two colors, the result usually needs to be two disjoint meshes of different colors.  In general, if you union an object of N colors with an object of M different colors, you end up with N+M meshes, each of a different color.

(There are exceptions for cases where two of the meshes are the same, or where one mesh is completely enclosed, but if we answer the questions for the simple cases then I think those cases are answered automatically.)

The matter is especially concrete for 3D printing, but still applies if all you want to do is display.  If you build a multi-colored object, and then you cut away parts of it, or you drill through it, you need to know and display the colors of the new faces.  The current preview logic paints the revealed faces with the color of the negative object, but that seems more like an implementation artifact than like a desirable and understandable outcome.

gingerbeardman, 2025-04-18:
If it helps, I don't turn any of my designs into plastic :) just PNGs

jordanbrown0, 2025-04-18:
If all you produce is PNGs, then the existing preview logic is mostly enough.  (Except Z-fighting.)

jordanbrown0, 2025-04-18:
Well, give or take the cut-away / drill-through questions.  The existing logic has an answer for those, but it's arguably not the rightest answer.

jimb15122, 2025-04-18:
Jordan, your questions are very valid.  They are at the heart of why no progress is being made.  The "stateful color" approach however can answer the question, albeit in a way you may not prefer.  I view the differencing union and intersect operations as resulting in a single mesh. same way cube or extrudes create a single mesh.  (This view is probably a benefit of my ignorance). Even when nested, the outermost bool operation is viewed as creating a single mesh.  Therefore the meshes entire color is the color that was in play at the time the outermost bool operation was encountered.

jimb15122, 2025-04-18:
implicit / lazy unions can be used to keep colors separate.

burunduk3, 2025-04-18:
Last time I had something similar to do, instead of color I used conditions, something like
```scad
if (target == "part1")
  square(10);
if (target == "part2")
  difference() {
    square(10, center=true);
    square(10);
  }
```

Then I ran `openscad -Dtarget=...` with different values and combined the results with an external script.

It wasn't too straightforward, though. At some point I found myself parsing CSG tree just to get coordinates of circle's center.

[15 more replies]