openscad/openscad #1609: Parameter name consistency

twinone, 2016-03-28:
Is there a reason to choose height in `linear_extrude(height=10)` and h in `cylinder(h=10)` instead of both h (or height), or accepting both?

kintel, 2016-03-28:
No reason. We could probably implement aliases. That potentially opens the floodgates for allowing such aliases in all other modules..

twinone, 2016-03-28:
Adding an alias in this particular case would improve consistency without breaking compatibility. This h/height is the only case I've encountered so far. Any other ideas?

t-paul, 2016-03-28:
There's at least the case of `center` vs. `centre` in addition.

Things are not as simple as providing just an alias as there are also positional parameters. It seems for `linear_extrude()` those are too obscure to create a problem. For `center` it's a bit more difficult as for example `cube(size, center)` has `center` as second positional parameter.

As this topic comes up now and then, maybe it's just a matter of clearly defining what takes precedence and also doing much better parameter validation (e.g. #1574,  #927, ...) and documentation.

doug-moen, 2016-03-28:
The main inconsistency I see is between single letter abbreviations, and words. Eg,

rotate(a=) vs rotate_extrude(angle=)
cylinder(h=) vs rotate_extrude(height=)

I see that a= means 'angle' for rotate, but 'alpha' for color.

My preference is to standardize on words, and support letters for backward compatibility only.

There is also alignment. Many geometric primitives have a center= option. Text has more sophisticated alignment: halign= and valign=. My preference is to provide a separate align() module, which operates on alignable shapes (those shapes for which a bounding box is trivially computable at preview time, without invoking CGAL). The align() module permits you to separately align the bounding box with respect to the X, Y and Z axes. For each axis, you can specify either over, center or under.

square(size=,center=)
circle(r=|d=)
polygon(points=,paths=,convexity=)
text(text=,size=,font=,halign=,valign=,...)
projection(cut=)
linear_extrude(height=,center=,...)
rotate_extrude(angle=,convexity=)
scale(v=)
resize(newsize=)
rotate(a=,v=)
translate(v=)
mirror(v=)
multmatrix(m=)
color(c=,a=)
offset(r=|delta=, chamfer=)
cube(size=,center=)
sphere(r=|d=)
cylinder(h=,r=,d=,r1=,r2=,d1=,d2=)
polyhedron(points=,faces=,convexity=)

kintel, 2016-03-28:
@doug-moen On using modules vs. parameters to set properties/attributes: Perhaps we should consider supporting both syntaxes (ref. the discussion in #1608)?

doug-moen, 2016-03-28:
@kintel said: On using modules vs. parameters to set properties/attributes:
Perhaps we should consider supporting both syntaxes (ref. the discussion in
#1608 <https://github.com/openscad/openscad/issues/1608>)?

Please give an example of the alternate syntax.

kintel, 2016-03-29:
@doug-moen You just gave an example, but let me extrapolate:
```
// these could be equivalent
align("center") cube();
cube(center=true);
```

..and for part name?
```
part("MyPart") sphere(r=10);
sphere(r=10, part="MyPart");
```

Not sure if that makes sense, but it could be helpful in terms of how we think about attributes.

doug-moen, 2016-03-29:
@kintel: Actually, this is the opposite of the idea I was trying to convey.

OpenSCAD has a history of loading too much functionality into each operation. It is better to break up these overly complex operations into multiple simpler, orthogonal operations. These new, simpler operations can then be composed together in richer ways. If done right, the language becomes simpler and more powerful. See also, the philosophy of Composable Building Blocks in OpenSCAD2: https://github.com/doug-moen/openscad2/blob/master/rfc/Composable_Building_Blocks.md

For example. linear_extrude() was originally designed to operate on a DXF file, and the DXF filename was specified as an argument, as in linear_extrude(file="foo.dxf", ...); In a later release, this complex operation was split up into two simpler, orthogonal operations: linear_extrude with a child argument, and dxf_import. This made the language more powerful, because these two primitives can be composed in richer ways. And each primitive is simpler.

I am attempting to point out a similar situation: many geometric primitives have alignment arguments. Most of them have a center= argument, but text() has halign= and valign=, which are more powerful. The circular primitives don't have alignment options at all: center is all you get. Instead of programming alignment logic into each primitive (and having more or fewer alignment options in each primitive, depending on the personal taste of the author who wrote that primitive), it is better to centralize that code into an align() operator. Now individual primitives do not need to implement their own alignment options.

align() is not intended to "set an attribute" in the "node" that is passed as its child argument. That's the opposite of the idea of simple, orthogonal operations that I'm trying to convey. It's an independent geometric operation that translates its child, based on alignment options passed as arguments, and also based on the child's bounding box. So it's similar to resize(), which scales its child argument relative to the child's bounding box.

kintel, 2016-03-29:
@doug-moen Thanks for the clarification. I agree, with a small fear that this may make code less readable for non-professionals. But I think that's better addressed after splitting things up into separate/orthogonal concerns.

doug-moen, 2016-03-29:
If you have an align() operator, then you can use it to implement a stack() operator, which takes an arbitrary number of children as arguments, and stacks them along some axis.

Other requests for more generalized alignment features: #265, #586, #621

t-paul, 2016-03-29:
Having `align()` is probably more readable than any of the proposed `center` extensions that allow different options.
Note that `text()` still would benefit from using `align()` as the `halign`/`valign` are based on glyph properties and not on the generated geometry.

juliendorra, 2017-03-28:
I second the idea of progressively deprecating single letter parameters. It's confusing and defeat the purpose of named parameters (Named parameters are really a great thing in a language, it's beginners friendly, it makes the code self-documenting… great force of Python, great sadness of Javascript… but single letter named parameters? Well, not so useful ;-)

As a regular user, h and height trip me up nearly everytime, and force me to go check on the cheatsheet.

What would be the path to deprecate single letter parameters?

kintel, 2017-03-28:
The path would be to rename all single-letter parameters, and add code to support the legacy single-letters ones for backwards compatibility. ..then consider if it's worth issuing a deprecation warning.
..then do the same for all docs and examples.

nophead, 2017-03-28:
I can't see the point of deprecating r,d, h, etc. It is pretty obvious what r and h would mean for a cylinder and overly verbose to use radius and height.

MichaelAtOz, 2017-03-28:
I'm with nophead. I would also like to see alias's for existing full words to first letters, where that makes sense. Particularly height in Linear_extrude... maybe c=convexity too.
Can you imagine the volume of deprecation errors you'd get from all those programs from Thiniverse which won't get updated.

jordanbrown0, 2025-01-06:
Note that with PR#5559 removing the file-based extrusion, height is now the first positional parameter, making it reasonable to specify it that way.

commit 6a25abe2 (2025-01-19): Parameter aliases in linear_extrude, rotate_extrude.  Fixes #1609. (#5572)

* Parameter aliases in linear_extrude, rotate_extrude.  Fixes #1609.
linear_extrude allows "h" as an alias for "height".
rotate_extrude allows "a" as an alias for "angle".
Provides a general-purpose alias-capable lookup function in Parameters.
Cleans up linear_extrude parameter processing.
closed 2025-01-16 by kintel