openscad/openscad #1610: align(): generalized alignment

doug-moen, 2016-03-29:
The `align()` operator aligns a shape with respect to the X, Y and Z axes. For each axis, you can align the shape to be centred, above (in the positive range), below (in the negative range), or specify no change. To match the capabilities of `halign` and `valign` in `text()`, you can also request to be centred, above or below a specific coordinate value on a particular axis.

The `align()` operator is more general than the `center=` argument supported by some geometric primitives, and can also replace most uses of the `halign=` and `valign=` arguments of `text()`.

In general, the arguments are `align(all=aspec, x=aspec, y=aspec, z=aspec)child` where each *aspec* argument defaults to `undef`. An *aspec* value is one of the following:
* `"center"`, which is short for `["center",0]`
* `["center",n]` -- move the center of the bounding box to coordinate n on the specified axis
* `"above"`, which is short for `["above",0]`
* `["above",n]`-- move the bottom of the bounding box to coordinate n on the specified axis
* `"below"`, which is short for `["below",0]`
* `["below",n]`-- move the top of the bounding box to coordinate n on the specified axis
* `undef`, meaning no change

`align("center") cube(10)` is the same as `cube(10, center=true)`. It centres the cube on the origin.

`align("above") cube(10)` is the same as `cube(10, center=false)`. It places the cube in the +X,+Y,+Z octant, adjacent to the origin.

The `align` operator has the same power as the `center=[x,y,z]` argument proposed by #265. For example,
```
align(x="center", y="center") cube(10)
```
is the same as the proposed
```
cube(10, center=[true,true,false])
```

Issue #621 proposes to add a `center=` option to `import()`. This is a more general solution.

Issue #586 proposes to either add a `center=` option to `resize()`, or to add a separate `center()` primitive. This is a more general solution.

Using `align()`, it is possible to implement a `zstack()` module which stacks a list of shapes along the Z axis. This is because calls to `align()` can be nested, which is an extra degree of flexibility that you don't get from alignment options in each primitive shape module.

Initial discussion of `align()` took place in issue #1609.

t-paul, 2016-03-29:
In addition we could review if https://github.com/devlaam/openscad/blob/loop_extension/doc/Loop.pdf from the old pull request #450 has some more ideas that could be included...

doug-moen, 2016-03-29:
The `align()` operator is more general than the generalized `center=` arguments of #450. For example,
```
sphere(4, center=[0,1,-1]);
```
from the Loop.pdf document is equivalent to
```
align(x="center",y="above",z="below") sphere(4);
```

doug-moen, 2016-03-29:
The `align()` operator has the same power as the `position()` operator with 1 shape argument from #450. For example, these are equivalent:
```
position(xmin=1, ymid=2, zmax=3) shape
align(x=["above",1], y=["center",2], z=["below",3]) shape
```

The `position()` operator with 2 shape arguments is a different beast. It can't be implemented in terms of `align()`. I would not suggest incorporating the 2-shape `position` semantics into `align`; I'd be happier if it was a separate operation in a separate change request. I think the 2-shape `position` is kind of a weird operation, which is different from anything I've seen requested before. By contrast, the existing design for `align` consolidates a set of features that many people have independently asked for, so it should be easier to build consensus behind the current design.

MichaelAtOz, 2016-03-29:
Above and Below connotate height, perhaps 'plus' & 'minus', or just '+' & '-',
```
align(x="+",y=["-",2],z="center") shape();
```

doug-moen, 2016-03-31:
Here's the rationale for "above" and "below".

@kintel said "I agree, with a small fear that this may make code less readable for non-professionals."

In order to make code readable for non-professionals, the best practice is to design a "fluent interface". The key idea is that, when you read the code out loud, it reads like English, or at least like some kind of pidgin English. It's English-enough that you can guess the meaning, using your knowledge of English.

Reference: http://www.martinfowler.com/bliki/FluentInterface.html, "The key test of fluency, for us, is the Domain Specific Language quality. The more the use of the API has that language like flow, the more fluent it is."

I think `align("center")` is fluent, and so is `align(["above", 0])`. The latter reads out loud as "align above zero". Unfortunately, the abbreviation `align("above")` is not as fluent.

"Above" and "below" don't just connote height, they are generally used for comparing magnitudes or quantities, such as temperatures, or dollar amounts. I haven't been able to come up with better words than these, which preserve fluency. `align(["plus",0])` is less fluent because "align plus zero" isn't English for the concept we want to convey, it sounds more like algebra, "align+0", where "align" is in the position of a variable.

doug-moen, 2016-03-31:
[relativity.scad](https://github.com/davidson16807/relativity.scad/wiki) has a really nice interface for aligning objects relative to the axes, or relative to each other.

The `anchor=` argument in Relativity is identical to the `center=` argument in #450. It's `[x,y,z]` where each axis value is -1 (below the origin), 0 (centered on the origin) or 1 (above the origin), same as "below", "center" and "above" in my proposal.

clothbot, 2016-04-02:
I say keep it simple. Apply the align() operator relative to the origin-axes and let the designer translate()/rotate()/multmatrix() as required.

Fleshing this out in my own mind, for a given blob() with bounding-cube corners [minx,miny,minz], [maxx,maxy,maxz]:

align(x="min",y="min",z="min") blob();

...or...

align(x="above",y="above",z="above") blob();

...would be equivalent to:

translate([ (minx>0)?-minx:minx,(miny>0)?-miny:miny,(minz>0)?-minz:minz]) blob();

...where the new bounding-cube corners are now [0,0,0], [maxx-minx,maxy-miny,maxz-minz].

align(x="max",y="max,z="max") blob();

...or...

align(x="below",y="below",z="below") blob();

...would be equivalent to:

translate([(maxx>0)?-maxx:maxx,(maxy>0)?-maxy:maxy,(maxz>0)?-maxz:maxz]) blob();

...where the new bounding-cube corners are now [minx-maxx,miny-maxy,minz-maxz], [0,0,0].

align(x="min",y="max",z="center") blob();

...or...

align(x="above",y="below",z="center") blob();

...would be equivalent to:

translate([
  (minx>0)?-minx:minx
  ,(maxy>0)?-maxy:maxy
  ,((minz>0)?-minz:minz)-(maxz-minz)/2
]) blob();

Does that seem about right?

Andrew.

doug-moen, 2016-04-02:
@clothbot: If the bounding box of the child has corners [xmin,ymin,zmin] and [xmax,ymax,zmax], then
```
   align(x="above",y="below",z="center") child
```
is equivalent to
```
   translate([-xmin, -ymax, -(zmax-zmin)/2]) child
```

clothbot, 2016-04-03:
Yeah, that's what I meant... Got my thought processes tangled up in knots thinking in terms of absolute values.  D'oh!

jordanbrown0, 2021-02-15:
My personal modules use `[jx,jy,jz]` where each is -1/0/+1.  That means that all eight "simple" alignments can be represented in a maximum of 10 characters.  The text variants are several times that verbose.  I understand and appreciate the desire for readability, and by all means include that option, but once I got this feature I used it a *lot* and verbosity matters too.

Allowing for fractions and values greater than +1/-1 is a natural extension, but not one I've felt the need for.

I find 0 to be a more natural "center" than 0.5.  I find -1 and +1 to be natural to put the object left and right of the centerline.  I find -2 and +2 natural to put the object left and right of that.  But when I start trying to do the math it gets unnatural, because the first +1/-1 moves the object half its size, and the second and later move it all of its size.  I'd rather abandon fractional alignment than abandon what I consider to be the naturalness of the integer behavior.

(But:  are "above" and "below" really appropriate for the X axis, which is usually oriented left-right?  And if you support "left" and "right" for X, what do you use for Y?)

MoonCactus, 2021-05-25:
> My personal modules use `[jx,jy,jz]` where each is -1/0/+1. 

I fully agree, and  it also helps a lot when alignment itself needs to be parametric.

I love the align() idea, but I really dislike using literal keywords. I mean, it would be as if scale(x="negative",size=10) instead of scale(x=-10). Sure scaling by -10 is an arguable choice in real life, but it was implemented like this, which I find parametric, compact and still obvious to read.

jordanbrown0, 2021-05-25:
I'm not sure exactly how it fits into this discussion, but BOSL2 has "anchors" that can, in addition to obvious points, translate an object to a *named* point on the object.  A screw, for instance, could have the origin be at the obvious top or bottom, but could also have it be on the bottom of the screw head.  It might be worth looking at what's been done there.

nophead, 2021-05-25:
The screws in my library always have the origin at the mating surface, so for a cap head it is under the head but for a countersunk screw it is at the top of the head, which matches the way screws are measured. I.e. a 20mm cap head does not include the head but a 20mm countersunk one does.

I never need anchors, just a useful origin and enough info to be able to find other useful features.

MoonCactus, 2021-05-27:
Anchors are worth mentioning here, but they are probably something else in my mind (*)

Actually I think the excellent reason to have an align() would be to generalize the `,center=true` specific argument that are both inconsistent and polluting to every function that implements it.
Then, imo, it would be "beautiful" though arguable that such an `align` works as `scale()` or `translate()` with a 3d vector consisting of values -1,0,+1 (matching the "below", "center" and "above" proposed by @doug-moen).

(*) BTW when I need *named* 3D position + alignment, I usually use a module + `children()`. What I'd love to have is the option to symbolically name such a "position" within a shape I am building from which I can refer and build later on (eg. to carve holes within a complex, positive shape afterwards). Not sure it suits CSG well and not sure it is pretty; in any case not related to `align()`

jordanbrown0, 2024-03-03:
[ Correction:  I said above 'all eight "simple" alignments'.  Should be 27:  plus, center, minus, in each of three axes. ]

General align() is hard, because you have to fully calculate the underlying operations to calculate the bounding box for an object.  There is no simple operation that will tell you the bounding box resulting from a difference or intersection.

But for many operations (union, translate, scale, hull, rotate_extrude, linear_extrude without twist, minkowski(?)) calculating the bounding box is trivial, and for many others (rotate, skew, multmatrix, linear_extrude with twist) it's only linear on the number of points.

I think the "hard" cases are difference, intersection, and maybe offset.

Note that render() turns an underlying "hard" case into an "easy" case.

Would it be worthwhile to have a limited align() that only promised good results on the "easy" cases (however defined)?

bitbasher, 2025-07-19:
sorry for coming late to the discussion .. but i feel the need to point out that in all of the other CAD tools i have used the term "align" refers to the operation of bringing multiple objects into alignment with each other, with the first item selected being the reference.

If i have read the discussion correctly "align" in this issue is about how to place a single object with respect to its own origin point or, possibly, arbitrary point in 3 space?  i am not clear on that.

I am just  trying to put out there that there is an existing industry standard  (sort of) meaning for 'align" that we would do well to follow.

I can tell i need to take a close look at BOSL2 ..