openscad/openscad #1616: Create signed version of Mac binaries

larsrc, 2016-04-06:
I have a company-issued MacBook and would like to use OpenSCAD on it. Unfortunately, the security policy only allows installing apps from App Store and identified developers. While I can understand not wanting to deal with the App Store weirdnesses, could you perhaps start signing the DMGs you make available for download? Instructions are here: https://developer.apple.com/library/mac/documentation/IDEs/Conceptual/AppDistributionGuide/DistributingApplicationsOutside/DistributingApplicationsOutside.html

t-paul, 2016-04-06:
Closing as duplicate of #880.

closed 2016-04-06 by t-paul

kintel, 2016-04-06:
@larsrc Please follow the original issue #880. I've been fearing a showstopper related to this. I'll see what I can do..