openscad/openscad #1619: OSX build fails

giantpune, 2016-04-09:
I have a freshly reset macbook running 10.11.4.  I checked out this git and ran the 2 suggested commands.  The first time, it built some of the really big libraries like Qt and boost.  Eventually it showed some popup prompt saying that i needed java installed and then the build failed.  Now, every time I run the scripts, they exit in this spot.

```sh
JVD-MacbookAir:openscad bob$ source setenv_mac.sh
JVD-MacbookAir:openscad bob$ ./scripts/macosx-build-dependencies.sh
Detected El Capitan (10.11) or later
Using clang compiler
Building for 10.7 or later
Setting number of CPUs to 4
Using basedir: /Users/bob/code/openscad/../libraries
eigen gmp mpfr boost qt5 qscintilla cgal glew gettext libffi glib2 opencsg freetype ragel harfbuzz libxml2 fontconfig
Building packages: eigen gmp mpfr boost qt5 qscintilla cgal glew gettext libffi glib2 opencsg freetype ragel harfbuzz libxml2 fontconfig

eigen already installed - not building
gmp already installed - not building
mpfr already installed - not building
boost already installed - not building
qt5 already installed - not building
qscintilla already installed - not building
cgal already installed - not building
glew already installed - not building
gettext already installed - not building
libffi already installed - not building
Building glib2 2.46.1...
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... ./install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether UID '503' is supported by ustar format... yes
checking whether GID '20' is supported by ustar format... yes
checking how to create a ustar tar archive... gnutar
checking whether to enable maintainer-specific portions of Makefiles... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... clang
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether clang accepts -g... yes
checking for clang option to accept ISO C89... none needed
checking whether clang understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of clang... gcc3
checking how to run the C preprocessor... clang -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking build system type... x86_64-apple-darwin15.4.0
checking host system type... x86_64-apple-darwin15.4.0
checking for Win32... no
checking for the Android... no
checking for Mac OS X Carbon support... yes
checking for Mac OS X Cocoa support... yes
checking OSX version >= 10.9.0... yes
checking whether to enable garbage collector friendliness... no
checking whether to disable memory pools... no
readlink: illegal option -- f
usage: readlink [-n] [file ...]
checking whether we are using the GNU C++ compiler... yes
checking whether clang++ accepts -g... yes
checking dependency style of clang++... gcc3
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... no
checking for pkg-config... no
checking for gawk... (cached) awk
checking for perl5... no
checking for perl... perl
checking for indent... indent
checking for perl... /usr/bin/perl
checking for a Python interpreter with version >= 2.5... python
checking for python... /usr/bin/python
checking for python version... 2.7
checking for python platform... darwin
checking for python script directory... ${prefix}/lib/python2.7/site-packages
checking for python extension module directory... ${exec_prefix}/lib/python2.7/site-packages
checking for iconv_open... no
checking for libiconv_open in -liconv... no
checking for iconv_open in -liconv... yes
checking for ZLIB... no
checking for inflate in -lz... yes
checking zlib.h usability... yes
checking zlib.h presence... yes
checking for zlib.h... yes
checking for LIBFFI... no
configure: error: in `/Users/bob/code/libraries/src/glib-2.46.1':
configure: error: The pkg-config script could not be found or is too old.  Make sure it
is in your PATH or set the PKG_CONFIG environment variable to the full
path to pkg-config.

Alternatively, you may set the environment variables LIBFFI_CFLAGS
and LIBFFI_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

To get pkg-config, see <http://pkg-config.freedesktop.org/>.
See `config.log' for more details
```

kintel, 2016-04-09:
Java? that sounds weird. Any idea where that popup came from?

The log shows that glib fails to build. I'll boot into 10.11 and see if I can reproduce..

giantpune, 2016-04-09:
I did `brew install pkg-config` and i'm making it past that spot.  It installed version 0.29.1.  I also installed JDK for whatever was causing the `install java` prompt.

spicetraders, 2016-04-09:
After install on my IMAC desktop, I installed on my MACBOOK running 10.11.4 this was done two nights back.  I am searching to find my install file (for double checking version as I remember reading some statement on version issues being possible) but late night install not sure what was used.   BUT I do remember a need to install JAVA or I should say JDK, as I have multiple JAVAs on my systems.  But install was clean and working after that.

kintel, 2016-04-10:
@giantpune Thx for the extra info. I'll double-check the dependencies on pkg-config and autotools and update the docs.

FYI: It's safe to ignore that Java message. I haven't figured out how to silence that dialog. Somehow OS X pops it up when one of the libraries tries to determine if Java is installed, bit doesn't offer an obvious way of turning that off..

kintel, 2016-04-10:
@giantpune The Java popup looks like a bug in gettext. Their build system has a ```--disable-java``` flag, but somehow they don't actually respect that flag. There are a bunch of discussions on this in various places online. If someone feels like tracking it down and potentially getting it reported to gettext, that would be awesome. The workaround right now is to just ignore the popup. That's also the workaround recommended by e.g. homebrew.

closed 2016-04-10 by kintel