openscad/openscad #1623: crash on windows 10 when using openscad.exe (but not openscad.com)

jandrieu, 2016-04-13:
I'm running version 2016.02.09 (git 9950e6a), so maybe its known that it won't work well in WIndows 10, but I didn't see any bugs exactly like this.

If I try opening from **Program Files/OpenSCAD/openscad.exe**, nothing happens other than a spinner for a few seconds. OpenSCAD crashes before even a window pops up.

However, if I click on **openscad.com**, it works great.  Unfortunately the "pin to taskbar" thinks the app in the .exe, so I can't click on the pinned icon, but at least opening files from file explorer works.

kintel, 2016-04-13:
@jandrieu Does the latest official release have the same problem? (2015.03-2)

jandrieu, 2016-04-13:
> @jandrieu[1] Does the latest official release have the same problem?

Yes, actually. I tried both.
 
And I just double checked to make sure.
 
No luck with the .exe, but the .com worked fine.

kintel, 2016-04-13:
@jandrieu There are a few things you could try:

* On the cmd-line, run ```openscad.exe --info``` and post the output here
* When OpenSCAD crashes, check the _Windows Event Log_ for any messages related to OpenSCAD
* Sometimes, bad graphics drivers tend to trip up OpenSCAD. Making sure that you've got the latest ones tend to fix some issues.

MichaelAtOz, 2018-04-17:
Hi Adrian,

Exception Code 0xc0000005 is Access Violation, usually a pointer is bad and tries to access memory it shouldn't.
It is occurring in the driver (ig9icd64.dll), not OpenSCAD.

You should try the latest development snapshot [here](http://www.openscad.org/downloads.html#snapshots).
If that doesn't work, there is little OpenSCAD can do.
Perhaps see if a slightly older driver fixes both problems?

closed 2024-12-22 by kintel