openscad/openscad #1624: Problem in editor when using two monitor with different scaling (dpi) on Win10

BADF00D, 2016-04-15:
Hi,
I'm struggling with a bug in the editor when using two screens with different dpi settings. One is a Full-HD 15" monitor with a scaling of 125% (120dpi I think)and an external 24" Full-HD monitor with a scaling of 100% (96dpi). I made different screenshots with the same scirpt to show the problem.

Only first screen (at 120dpi) active => all fine
[image]
Only second screen (at 96dpi) active => all fine
[image]
Both active, OpenScad at 15" screen => all fine
[image]
Both active, OpenScad at 24" screen=> **text rendering in editor is weird**
[image]
When I switch to 96dpi on my 15" screen => all fine on both screens
[image]
[image]

By the way, the 15" screen is my primary screen.

doug-moen, 2016-04-15:
This may be a Scintilla bug on Windows.

"SciTE’s support of HiDPI displays on Windows is still incomplete despite
quite a bit of effort. When there are multiple monitors where each has a
different DPI setting, SciTE does not adapt well when moved between
monitors." Nov 2015

source:
https://groups.google.com/forum/#!msg/scintilla-interest/EdY6kjQuFgA/2fl17ivyBQAJ

>

kintel, 2016-04-15:
@BADF00D Could you try if you get similar errors with NotePad++ (https://notepad-plus-plus.org/). Then we'd know where to report this.

BADF00D, 2016-04-18:
Hi, 
I just copied the source code from OpenScad to Nodepad++. The letters are a little bit blurry in Notepad++, but there are sharp in OpenScad. But main difference is, that wihtin OpenScan the letters overlap each other, but they do not overlap in Notepad++. I attached some screenshots to show this.

15 inch at 120dpi and 24 inch at 96dpi Notepad++ on 15inch monitor
[image]

15 inch at 120dpi and 24 inch at 96dpi OpenScad on 15 inch monitor
[image]

kintel, 2016-04-18:
@BADF00D Thanks for hanging in there :)

Some more things to try: In Preferences, there are a bunch of editor settings. Could you play around with them a bit and see if that changes anything?

* I'd be interested in whether font changes make a difference.
* It's also possible to enable the old editor. This is likely to fix the issue, but you'd lose a lot of functionality. 
* Syntax highlighting could influence this too - try turning that off.

BADF00D, 2016-05-06:
Hi,
I was a little busy lately, but finally I tried you suggestions.
- I changed the Editor to simple, without solving the problem.
- I changed the font (within simple editor and QScintilla), without solving the problem. I also tried mono and non mono fonts.
- Disabling the syntax highlighting dosn't solve the problem either.

SteveDC2, 2016-08-16:
Any update on this issue? I am getting similar, unusable, editor behavior with multiple monitors where one is a different scale factor. 

In my case the buffer is changed in text locations that are different to the cursor position. Often the cursor will jump to a position to the right of where the buffer is being changed.

The displayed text will also end up being changed at the wrong place and updated when scrolled etc...

Basically it looks like the displayed cursor and the actual position in the buffer are mis-aligned.

Occasionally I also get large gaps between the characters and/or text lines.

Strange thing is that it does not always seem to happen, but once it starts happening it does not clear up.

Not managed to determine exactly what triggers the behavior though yet.

t-paul, 2016-08-17:
I'm not aware of any progress. Due to lack of both a developer on Windows and the needed hardware setup, it's hard to say how and when this can be solved.
The tests seem to indicate it's not (or at least not only) a SciTE/QScintilla issue, as it also happens with the Simple editor which is basically just a Qt text edit control.

kintel, 2016-08-17:
@BADF00D @SteveDC2 When switching the editor to simple, you need to restart OpenSCAD for the changes to take effect. Could you double check that this problem is present also with the simple editor?

SuperRoach, 2016-11-13:
If it helps, this issue occurs only only maximised. If the screen is a partial window, it moves between screens fine. Once you maximise on the non primary screen, the problem occurs (and stays). If the app is maximised on startup the error is there right away.

NoobyToosday, 2017-01-04:
@kintel I had the same problem with overlapping text. Using the simple editor worked for me after restarting.

MichaelPFrey, 2017-01-06:
this issue could be related to #1522

NathanJPhillips, 2021-04-19:
Tried simple editor - that fixes it as per other report.

closed 2026-02-15 by github-actions[bot]