openscad/openscad #1629: Any interest in a UML diagram loosely describing OpenSCAD code?

mcraveiro, 2016-04-22:
Hi,

As part of another project, I had to spend some time understanding the internals of OpenSCAD. This is an ongoing effort and does not capture every detail - rather mainly the parts that I am mostly interested in. It is also not really a "pure" UML diagram, but instead uses a lot of "practical" and expedient notation.
The diagram is now good enough for my needs. It is a Dia[1] diagram and is available here[2]. 

Is there any interest in the diagram by the OpenSCAD comunity? 

Cheers

Marco

[1] https://wiki.gnome.org/Apps/Dia/. For Windows see http://dia-installer.de/
[2] https://github.com/UHBiocomputation/neurite/blob/master/doc/misc/open_scad.dia

FluxIX, 2016-04-22:
I think it would be good, so long as it was maintained by someone. Having a diagram available would make it easier to determine the impact of changes and ease the introduction of new developers into the project.

kintel, 2016-04-22:
We already have some diagrams here: https://github.com/openscad/openscad/tree/master/doc
I agree that diagrams ideally need to be maintained. One workaround could be to put a visual update timestamp in all diagrams to give an indication of age.

The larger problem is how to display and edit such diagrams. To my knowledge there isn't a decent standard for representing diagrams which can be edited by reasonable tools. Dia might make sense, but sadly doesn't work on Mac (it theoretically works I guess, but their binaries don't). Also, diagramming tools tend to have horrific or unintuitive UIs. As a result most people tend to personally learn and standardize on one tool for all their diagramming needs. Including me - I eventually bit the bullet and bought OmniGraffle for Mac and use it for virtually everything (including the diagrams in this repo; I just exported to PDF). This is counter-productive for multi-platform collaboration, and the diagrams is mostly for personal use.

Nevertheless, I'd be happy to accept a pull request and put your diagram into the doc folder, in case more people finds value in it.

doug-moen, 2016-04-22:
1. I managed to get Dia running on my Mac (OS X 10.11) by starting XQuartz,
then in the XQuartz xterminal window, set DISPLAY=:0, then run Dia from the
command line. See also http://xkcd.com/1671/

2. The diagram seems useful enough. Thanks Marco.

3. It's a class diagram, so in the long term, it will get out of date as
the code changes. If we wanted to maintain this kind of documentation, I'd
suggest using Doxygen comments in the code. Then use Doxygen to generate
the UML, as well as other online documentation. The results may or may not
be as good as Marco's hand generated diagram, but the benefit is
automation. Here's the feature set:
https://www.stack.nl/~dimitri/doxygen/manual/diagrams.html

thehans, 2018-02-21:
An updated overall class hierarchy diagram would be really useful to me right now.  Its tough to get an view of how things fit together just from looking individual source files, and the most recent diagram in /doc looks like its missing a lot.

mcraveiro, 2018-02-22:
Hi @thehans,

I'm afraid my diagram has not been updated since I opened this ticket and I will not be able to maintain it going forward as unfortunately I do not require OpenSCAD for my project any longer :-( but it may still be worthwhile looking at as a starting point.

Cheers

Marco

t-paul, 2018-04-06:
I guess that also makes the ticket obsolete for now.

closed 2018-04-06 by t-paul