openscad/openscad #1630: Build error on Mac OS X

xythobuz, 2016-04-23:
I'm trying to build the current master on Mac OS X 10.10.5 with XCode 7.2.1. First, I tried to use the libraries installed with MacPorts, but I've also tried the included dependency install scripts `setenv_mac.sh` and `macosx-build-dependencies.sh`. Both result in the same error when trying to compile `src/OpenSCADApp`:

    src/OpenSCADApp.cc:15:31: error: cannot initialize a new value of type 'EventFilter'
          (aka 'bool (*)(void *, long *)') with an rvalue of type 'OpenSCADApp *'
            this->installEventFilter(new EventFilter(this));
                                         ^           ~~~~

There are other setenv scripts, am I using the wrong one? Any other ideas?

kintel, 2016-04-24:
Confirmed. This must be a recent regression. I'm using the same configuration.

kintel, 2016-04-24:
This build issue appears to be a conflict between your default shell environment and the manually built libraries. It works for me if I run ```source setenv_mac.sh``` before executing a clean ```qmake && make```.

xythobuz, 2016-04-24:
Thank you very much, I've solved it.

I was still using the MacPorts qmake binary instead of the one built by the dependency script. Calling the correct qmake and it builds without problems.

I'm sorry, please excuse the non-issue :)

closed 2016-04-24 by xythobuz