openscad/openscad #1631: ERROR: CGAL error in CGALUtils::applyBinaryOperator difference: CGAL ERROR: assertion violation!

gskillas, 2016-04-25:
Slicing an object produces the above error in 2 of 20 slices:

---------------------------------------------------
> sh slices.sh 
CGAL Cache insert: difference(){import(file="curveRoundedEn (4595744 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (5164304 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (5164304 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (5192864 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (5229464 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (5114416 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (5196376 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (5413096 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (4614688 bytes)
ERROR: CGAL error in CGALUtils::applyBinaryOperator difference: CGAL ERROR: assertion violation!
Expr: target(s1).vertex()==target(s2).vertex()
File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h
Line: 218
CGAL Cache insert: difference(){import(file="curveRoundedEn (5840184 bytes)
ERROR: CGAL error in CGALUtils::applyBinaryOperator difference: CGAL ERROR: assertion violation!
Expr: target(s1).vertex()==target(s2).vertex()
File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h
Line: 218
CGAL Cache insert: difference(){import(file="curveRoundedEn (7118776 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (4605968 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (5409416 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (5191096 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (5109136 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (5224184 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (5187584 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (5159024 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (5159024 bytes)
CGAL Cache insert: difference(){import(file="curveRoundedEn (4592304 bytes)
---------------------------------------------------

Please see attached tarfile for STL and SCAD files.

[curve.tar.gz](https://github.com/openscad/openscad/files/234042/curve.tar.gz)

kintel, 2016-04-25:
How did you create your STL file? This STL file, while the connectivity is manifold, has a few Zero Area Triangles (aka. T-vertices). I believe that CGAL sees these as self-intersections and fail to perform operations which touch these faces. All of the face in question are located inside some of the horizontal holes in the model. If I manually clean that up in Blender, it works as expected.

I think we could slim down this into a clean test case which we could file with CGAL. See related issues #802 #1525 #1184

WZ9V, 2016-05-15:
I'm wondering if I'm having a similar problem.  I imported a STL and it renders OK, I can rotate it and move it just fine.  However if I try try to union something to it or difference something from it it fails to render with F6 and does draw with F5.  It's a STL I got from thingiverse that I'm trying to remove some features from and extend some others.

WZ9V, 2016-05-15:
Think I figured out my problem, the STL mesh has issues according to FreeCAD when I import it there.

stefano2734, 2026-09-20:
Related issue is here with led diffuser solved. Is here progress after 8 years.