openscad/openscad #1632: Animate with dump pictures from command line

nairaner, 2016-04-25:
Adding option to use animation would be useful for specifying output image resolution, which would allow downsampling for better animation, as well as animation of more complicated models.

t-paul, 2016-04-25:
Animation is basically a GUI feature, but it's possible to just give the $t parameter manually on command line like `openscad -D$t=0.2 -o out.png in.scad` which should have pretty much the same effect when calling multiple times with varying values.

nairaner, 2016-04-25:
It works, but it is very slow due to reload of all files when making each file

t-paul, 2016-04-25:
The idea is to provide some means for persistent caching which would help for multiple use cases including this one.

kintel, 2016-05-02:
I guess it's conceivable to have a cmd-line option to specify animation parameters, and run the preview-to-png renderer in a loop. It's kind of crossing into application scripting though.

Hexelena, 2016-05-14:
I would like to add that when running in a shell like bash, you have to put the $t=0.2 in quotation marks like:
`openscad` -D'$t=0.2' -o out.png in.scad`
Otherwise it will be interpreted as shell variable.

closed 2020-11-28 by t-paul

spuder, 2021-12-30:
For anyone else looking how to animate an STL into still images (which could be converted into a video). I've [published a bash script on S.O.](https://stackoverflow.com/a/70468907/1626687)

PR #3280 (2020-11-28): Add --animate command line option.
Fixes #1632.

Notes:
- I split up the huge `cmdline` function because for animation the second half of this function (which does the actual exporting) needs to be run once for each frame.
- If you don't specify an output filename and only use `--animate N`, the frame filenames will be `frame00000.png`, `frame00001.png`, etc, same as the filenames picked by the GUI's "Dump pictures" feature.
- If you _do_ specify an output filename, e.g. `-o bla.png --animate 20`, the frames are output as `bla00000.png`, `bla00001.png`, etc.