openscad/openscad #1633: Buggy comparision operators in floating point numbers

ghpenguin, 2016-04-26:
Hi, 

64bit AMD Devuan kernel 
version 2016.03.10.nightly

I encountered weird problem of the comparision operators - less than, equal to, and etc.
 
V is the original vector that contains 3D coordinates. 
draw_edges() creates a vector of floating point numbers.  Function quicksort() creates a sorted vector. 

If the vector contains floating numbers, 'less than' operator gives inconsistent result. So I had to convert all floating numbers to integer for the comparison. 
     edge_distances = [for (i = fromto) 
                let (e = Edge[i], 
                    i1 = e[0],   // index of 1st vertex 
                    i2 = e[1],   // index of 2nd vertex
                    v = V[i1],  // coordinate of 1st vertex
                    w = V[i2],  // coordinate of 2nd vertex
                    _d = distance(v, w),
                    d = round(decimal_factor * _d)) [v, w, d]
            ];

function get_edge_types(vv, tt, i) = i <= end(vv) ?
        let (
            prev = vv[i-1],
            curr = vv[i],
            t = (curr > prev) ? [curr] : []
            ) get_edge_types(vv, concat(tt, t), i+1) : tt;    

get_edge_types should collect unique edge types like below. 
ECHO: 0, "Type: ", 10.6437
ECHO: 4, "Type: ", 10.6438
ECHO: 10, "Type: ", 10.7532
ECHO: 33, "Type: ", 10.7533
ECHO: 39, "Type: ", 12.0387
ECHO: 49, "Type: ", 12.1509

Link to the source code.
https://github.com/ghpenguin/workshop/blob/OpenSCAD/R50cm_dome3.scad

V = 
[
    [0, 0, 19.69],                  // 0
    [3.19405, -9.8303,16.7243],     // 1
    [10.3362, 0, 16.7243],          // 2
    [5.44218, -16.7493, 8.80564],   // 3    
    [13.5302, -9.8303, 10.3362],    // 4
    [17.6113, 0, 8.80564],          // 5
    [-8.36215, -6.07546, 16.7243],  // 6
    [-14.2478, -10.3516, 8.80564],  // 7
    [-5.1681, -15.9058, 10.3362],   // 8
    [-8.36215, 6.07546, 16.7243],   // 9
    [-14.2478, 10.3516, 8.80564],   // 10
    [-16.7243, 0, 10.3362],         // 11
    [3.19405, 9.8303, 16.7243],     // 12
    [5.44218, 16.7493, 8.80564],    // 13    
    [-5.1681, 15.9058, 10.3362],    // 14
    [13.5302, 9.8303, 10.3362],     // 15
    [18.6983, -6.07546, 0],         // 16
    [11.5562, -15.9058, 0],         // 17
    [14.2478, -10.3516, -10.3362],  // 18 * z: -8.80564
    [0, -19.6606, 0],               // 19
    [-5.44218, -16.7493, -10.3362], // 20 * 
    [5.1681, -15.9058, -10.3362],   // 21
    [-11.5562, -15.9058, 0],        // 22
    [-18.6983, -6.07546, 0],        // 23
    [-17.6113, 0, -10.3362],        // 24 * 
    [-13.5302, -9.8303, -10.3362],  // 25
    [-18.6983, 6.07546, 0],         // 26
    [-11.5562, 15.9058, 0],         // 27
    [-5.44218, 16.7493, -10.3362],  // 28 * 
    [-13.5302, 9.8303, -10.3362],   // 29
    [0, 19.6606, 0],                // 30
    [11.5562,15.9058, 0],           // 31
    [14.2478,10.3516,-10.3362],     // 32 * 
    [5.1681,15.9058,-10.3362],      // 33
    [18.6983, 6.07546, 0],          // 34
    [16.7243, 0, -10.3362]          // 35
];

// Quicksort 
function quicksort(vector) = !(len(vector)>0) ? [] : 
        let (
            pivot = floor(len(vector)/2),
            pivot_item = vector[pivot],  // [0, 1, 10.7532]
            lesser = [ for (y = vector) if (y < pivot_item) y ],
            equal = [ for (y = vector) if (y == pivot_item) y ],
            greater = [ for (y = vector) if (y > pivot_item) y ]
        ) concat(quicksort(lesser), equal, quicksort(greater));

// Find all edge types to assign colors 
function get_edge_types(vv, tt, i) = i <= end(vv) ?
        let (
            prev = vv[i-1],
            curr = vv[i],
            t = (curr > prev) ? [curr] : []
            ) get_edge_types(vv, concat(tt, t), i+1) : tt;    

module draw_edges(fromto)
{   
    ECHO_STR("draw_edges(): START", true);      
 
    // Create edge and distance vector 
    // [[v, w, distance], ...] 
    // [[0, 0, 19.69], [3.19405, -9.8303, 16.7243], 107532], ... ]
    // convert distance value of floating point to integer 
    // integer = round(decimal_factor * ##.####) 
    // i.e., 10.7532 to 107532 
    edge_distances = [for (i = fromto) 
                let (e = Edge[i], 
                    i1 = e[0],   // index of 1st vertex 
                    i2 = e[1],   // index of 2nd vertex
                    v = V[i1],  // coordinate of 1st vertex
                    w = V[i2],  // coordinate of 2nd vertex
                    _d = distance(v, w),
                    d = round(decimal_factor * _d)) [v, w, d]
            ];
   
    // Sort edges
    temp = [for (i = fromto) edge_distances[i][2]];  
    e_d = quicksort(temp);    
      
    // Find all edge types
    t1 = e_d[0];         
    edge_types = get_edge_types(e_d, [t1], 1);   
  
    echo(len(edge_types));
    for (i = indices(edge_types))
    {
        t0 = edge_types[i]/decimal_factor;
        t = edge_types[i];
        result = len(search(t, e_d, 0, 0));
        echo(str("Type", i, " ", t0, ": ", result));
    }
    
    /*
        Color names
    ECHO: "Type0 10.6437: 4"  red
    ECHO: "Type1 10.6438: 6"  black
    ECHO: "Type2 10.7532: 23" lime 
    ECHO: "Type3 10.7533: 7"  maroon
    ECHO: "Type4 12.0387: 10" purple 
    ECHO: "Type5 12.1509: 45" skyblue 
    */
    
    r1 = 1/2;
    for (i = indices(edge_distances))
    {
        v = edge_distances[i][0];
        w = edge_distances[i][1];
        t = edge_distances[i][2];
        if (t == edge_types[0]) draw_edge(v, w, r1, "red"); 
        if (t == edge_types[1]) draw_edge(v, w, r1, "blue");
        if (t == edge_types[2]) draw_edge(v, w, r1, "lime");
        if (t == edge_types[3]) draw_edge(v, w, r1, "maroon");
        if (t == edge_types[4]) draw_edge(v, w, r1, "purple");
        if (t == edge_types[5]) draw_edge(v, w, r1, "orange");
        // draw_edge(v, w, 1, cname);
    }
        
    ECHO_STR("draw_edges(): END", true);         
}

draw_edges(all_edges);

kintel, 2016-04-26:
It's hard to read from your code what the problem really is. Do you have a minimal example of a failing comparison operator?

nophead, 2016-04-26:
Not that echo() doesn't show the full precision of floating point numbers,
so it is possible to get two numbers that print the same but fail an
equality test.

ghpenguin, 2016-04-27:
@kintel 

I made a short code that generates the comparison operator errors.
ff is a vector that contains floating point numbers.
ii is a vector that contains integer numbers of the floating point numbers. 

 echo("Compare floating point numbers");
    echo();
    for (i = [1 : end(ff)])
    {
        prev = ff[i-1];
        curr = ff[i];
        if (curr > prev)
            echo(i, "=======>",  curr, "> ", prev);
        else if (curr == prev)
            echo(i, curr, "==", prev);
    }
    
    echo();
    echo("Compare integer numbers");
    echo();
    
    for (i = [1 : end(ii)])
    {
        prev = ii[i-1];
        curr = ii[i];
        if (curr > prev)
            echo(i, "save =====>",  curr, "> ", prev);
        else if (curr == prev)
            echo(i, "throw");
    }

ECHO: 
ECHO: "draw_edges(): START"
ECHO: 
ECHO: "Compare floating point numbers"
ECHO: 
ECHO: 1, 10.6437, "==", 10.6437
ECHO: 2, "=======>", 10.6437, "> ", 10.6437
ECHO: 3, 10.6437, "==", 10.6437
ECHO: 4, "=======>", 10.6438, "> ", 10.6437
ECHO: 5, 10.6438, "==", 10.6438
ECHO: 6, "=======>", 10.6438, "> ", 10.6438
ECHO: 7, 10.6438, "==", 10.6438
ECHO: 8, "=======>", 10.6438, "> ", 10.6438
ECHO: 9, 10.6438, "==", 10.6438
ECHO: 10, "=======>", 10.7532, "> ", 10.6438
ECHO: 11, 10.7532, "==", 10.7532
ECHO: 12, "=======>", 10.7532, "> ", 10.7532
ECHO: 13, 10.7532, "==", 10.7532
ECHO: 14, "=======>", 10.7532, "> ", 10.7532
ECHO: 15, 10.7532, "==", 10.7532
ECHO: 16, "=======>", 10.7532, "> ", 10.7532
ECHO: 17, 10.7532, "==", 10.7532
ECHO: 18, "=======>", 10.7532, "> ", 10.7532
ECHO: 19, 10.7532, "==", 10.7532
ECHO: 20, "=======>", 10.7532, "> ", 10.7532
ECHO: 21, 10.7532, "==", 10.7532
ECHO: 22, "=======>", 10.7532, "> ", 10.7532
ECHO: 23, 10.7532, "==", 10.7532
ECHO: 24, "=======>", 10.7532, "> ", 10.7532
ECHO: 25, 10.7532, "==", 10.7532
ECHO: 26, "=======>", 10.7532, "> ", 10.7532
ECHO: 27, 10.7532, "==", 10.7532
ECHO: 28, "=======>", 10.7532, "> ", 10.7532
ECHO: 29, 10.7532, "==", 10.7532
ECHO: 30, "=======>", 10.7532, "> ", 10.7532
ECHO: 31, "=======>", 10.7532, "> ", 10.7532
ECHO: 32, 10.7532, "==", 10.7532
ECHO: 33, "=======>", 10.7533, "> ", 10.7532
[... 137 lines cut ...]
ECHO: 74, "throw"
ECHO: 75, "throw"
ECHO: 76, "throw"
ECHO: 77, "throw"
ECHO: 78, "throw"
ECHO: 79, "throw"
ECHO: 80, "throw"
ECHO: 81, "throw"
ECHO: 82, "throw"
ECHO: 83, "throw"
ECHO: 84, "throw"
ECHO: 85, "throw"
ECHO: 86, "throw"
ECHO: 87, "throw"
ECHO: 88, "throw"
ECHO: 89, "throw"
ECHO: 90, "throw"
ECHO: 91, "throw"
ECHO: 92, "throw"
ECHO: 93, "throw"
ECHO: 94, "throw"
ECHO: 
ECHO: "draw_edges(): END"
ECHO: 
[ffnumber_bug.scad.txt](https://github.com/openscad/openscad/files/238540/ffnumber_bug.scad.txt)

kintel, 2016-04-27:
It looks OK to me. As @nophead pointed out, ECHO doesn't print with full precision, so that might throw off your visual comparison.

nophead, 2016-04-27:
If you change the echo to show the difference you see it is always zero when they are equal and positive when they differ.
```
        if (curr > prev)
            echo(i, "=======>",  curr, "> ", prev, curr - prev);
        else if (curr == prev)
            echo(i, curr, "==", prev, curr - prev);
```
```
ECHO: "Compare floating point numbers"
ECHO: 
ECHO: 1, 10.6437, "==", 10.6437, 0
ECHO: 2, "=======>", 10.6437, "> ", 10.6437, 2.1745e-006
ECHO: 3, 10.6437, "==", 10.6437, 0
ECHO: 4, "=======>", 10.6438, "> ", 10.6437, 4.03579e-005
ECHO: 5, 10.6438, "==", 10.6438, 0
ECHO: 6, "=======>", 10.6438, "> ", 10.6438, 2.61092e-005
ECHO: 7, 10.6438, "==", 10.6438, 0
ECHO: 8, "=======>", 10.6438, "> ", 10.6438, 4.01972e-006
ECHO: 9, 10.6438, "==", 10.6438, 0
ECHO: 10, "=======>", 10.7532, "> ", 10.6438, 0.109411
ECHO: 11, 10.7532, "==", 10.7532, 0
ECHO: 12, "=======>", 10.7532, "> ", 10.7532, 2.15236e-006
ECHO: 13, 10.7532, "==", 10.7532, 0
ECHO: 14, "=======>", 10.7532, "> ", 10.7532, 6.35597e-006
ECHO: 15, 10.7532, "==", 10.7532, 0
ECHO: 16, "=======>", 10.7532, "> ", 10.7532, 9.01784e-006
ECHO: 17, 10.7532, "==", 10.7532, 0
ECHO: 18, "=======>", 10.7532, "> ", 10.7532, 9.59402e-006
ECHO: 19, 10.7532, "==", 10.7532, 0
ECHO: 20, "=======>", 10.7532, "> ", 10.7532, 9.43565e-006
ECHO: 21, 10.7532, "==", 10.7532, 0
ECHO: 22, "=======>", 10.7532, "> ", 10.7532, 2.34354e-006
ECHO: 23, 10.7532, "==", 10.7532, 0
ECHO: 24, "=======>", 10.7532, "> ", 10.7532, 2.36958e-006
ECHO: 25, 10.7532, "==", 10.7532, 0
ECHO: 26, "=======>", 10.7532, "> ", 10.7532, 6.20353e-007
ECHO: 27, 10.7532, "==", 10.7532, 0
ECHO: 28, "=======>", 10.7532, "> ", 10.7532, 2.10072e-007
ECHO: 29, 10.7532, "==", 10.7532, 0
ECHO: 30, "=======>", 10.7532, "> ", 10.7532, 5.03865e-006
ECHO: 31, "=======>", 10.7532, "> ", 10.7532, 1.41568e-006
ECHO: 32, 10.7532, "==", 10.7532, 0
ECHO: 33, "=======>", 10.7533, "> ", 10.7532, 5.59264e-006
ECHO: 34, "=======>", 10.7533, "> ", 10.7533, 1.37965e-005
ECHO: 35, 10.7533, "==", 10.7533, 0
ECHO: 36, "=======>", 10.7533, "> ", 10.7533, 3.97879e-006
ECHO: 37, 10.7533, "==", 10.7533, 0
ECHO: 38, "=======>", 10.7533, "> ", 10.7533, 1.64971e-005
ECHO: 39, 10.7533, "==", 10.7533, 0
ECHO: 40, "=======>", 12.0387, "> ", 10.7533, 1.28538
ECHO: 41, 12.0387, "==", 12.0387, 0
ECHO: 42, "=======>", 12.0387, "> ", 12.0387, 1.69978e-005
ECHO: 43, 12.0387, "==", 12.0387, 0
ECHO: 44, "=======>", 12.0387, "> ", 12.0387, 2.0933e-006
ECHO: 45, 12.0387, "==", 12.0387, 0
ECHO: 46, "=======>", 12.0387, "> ", 12.0387, 8.62348e-006
ECHO: 47, 12.0387, "==", 12.0387, 0
ECHO: 48, "=======>", 12.0387, "> ", 12.0387, 3.56084e-005
ECHO: 49, 12.0387, "==", 12.0387, 0
ECHO: 50, "=======>", 12.1509, "> ", 12.0387, 0.112153
ECHO: 51, 12.1509, "==", 12.1509, 0
ECHO: 52, "=======>", 12.1509, "> ", 12.1509, 1.79932e-005
ECHO: 53, 12.1509, "==", 12.1509, 0
ECHO: 54, "=======>", 12.1509, "> ", 12.1509, 1.06083e-006
ECHO: 55, 12.1509, "==", 12.1509, 0
ECHO: 56, 12.1509, "==", 12.1509, 0
ECHO: 57, 12.1509, "==", 12.1509, 0
ECHO: 58, "=======>", 12.1509, "> ", 12.1509, 1.23606e-005
ECHO: 59, 12.1509, "==", 12.1509, 0
ECHO: 60, "=======>", 12.1509, "> ", 12.1509, 2.707e-007
ECHO: 61, 12.1509, "==", 12.1509, 0
ECHO: 62, "=======>", 12.1509, "> ", 12.1509, 9.16169e-006
ECHO: 63, 12.1509, "==", 12.1509, 0
ECHO: 64, 12.1509, "==", 12.1509, 0
ECHO: 65, 12.1509, "==", 12.1509, 0
ECHO: 66, "=======>", 12.1509, "> ", 12.1509, 1.68547e-006
ECHO: 67, 12.1509, "==", 12.1509, 0
ECHO: 68, 12.1509, "==", 12.1509, 0
ECHO: 69, 12.1509, "==", 12.1509, 0
ECHO: 70, "=======>", 12.1509, "> ", 12.1509, 1.97368e-007
ECHO: 71, 12.1509, "==", 12.1509, 0
ECHO: 72, 12.1509, "==", 12.1509, 0
ECHO: 73, "=======>", 12.1509, "> ", 12.1509, 2.62553e-006
ECHO: 74, 12.1509, "==", 12.1509, 0
ECHO: 75, 12.1509, "==", 12.1509, 0
ECHO: 76, 12.1509, "==", 12.1509, 0
ECHO: 77, "=======>", 12.1509, "> ", 12.1509, 6.33001e-006
ECHO: 78, 12.1509, "==", 12.1509, 0
ECHO: 79, "=======>", 12.1509, "> ", 12.1509, 1.24394e-006
ECHO: 80, 12.1509, "==", 12.1509, 0
ECHO: 81, "=======>", 12.1509, "> ", 12.1509, 2.78296e-007
ECHO: 82, 12.1509, "==", 12.1509, 0
ECHO: 83, 12.1509, "==", 12.1509, 0
ECHO: 84, 12.1509, "==", 12.1509, 0
ECHO: 85, "=======>", 12.1509, "> ", 12.1509, 2.69321e-006
ECHO: 86, 12.1509, "==", 12.1509, 0
ECHO: 87, 12.1509, "==", 12.1509, 0
ECHO: 88, 12.1509, "==", 12.1509, 0
ECHO: 89, "=======>", 12.1509, "> ", 12.1509, 7.59622e-006
ECHO: 90, 12.1509, "==", 12.1509, 0
ECHO: 91, "=======>", 12.1509, "> ", 12.1509, 1.85837e-006
ECHO: 92, 12.1509, "==", 12.1509, 0
ECHO: 93, 12.1509, "==", 12.1509, 0
ECHO: 94, 12.1509, "==", 12.1509, 0
```

The small differences get lost when you round them to integers.

closed 2016-05-02 by kintel