openscad/openscad #1635: Ubuntu 16.04 failing tests

doug-moen, 2016-04-30:
On April 29 I cloned the OpenSCAD master branch and built OpenSCAD on Ubuntu 16.04. I ran the default set of tests using "ctest" with no arguments.
There are 3 failures:

The following tests FAILED:
    469 - opencsgtest_issue1165 (Failed)
    471 - opencsgtest_issue1215 (Failed)
    781 - throwntogethertest_issue1215 (Failed)

Although this looks like an OpenGL driver issue, in further tests I am unable to conclusively pin it on the driver.

I'm not using a VM. I'm using an NVidia graphics card, GT 218 [GeForce 210], GL_RENDERER is "Gallium 0.4 on NVA8". I'm using the default Nouveau open source driver.

If I switch to the NVidia-340 proprietary driver, the same tests fail.

Force Goldfeather does *not* fix the problem: the same tests fail with both of the above drivers. So I've tested 4 variations with no change in behaviour.

MichaelPFrey, 2018-03-01:
I have pretty much the same issue. Ubuntu 17.10 with NVidia proprietary driver.

[image]

related to #1412

thehans, 2018-03-01:
I also can confirm that the issue1215 tests fail for me, on Linux Mint 18.3 (based on Ubuntu 16.04).  
Also using Nvidia proprietary driver.

issue964 does not fail on my specific setup though.

thehans, 2018-03-01:
The travis build already excludes these and other tests from its test run:
https://github.com/openscad/openscad/blob/master/scripts/travis-ci.sh#L62-L84

I think it would be good to maybe reconfigure the test suite so that the default matches with what the travis server runs, and then just change the travis script to run the default?

closed 2026-02-16 by github-actions[bot]