openscad/openscad #1636: New 2-D Polygon Offset Function Feature Request

to-the-nth, 2016-05-01:
OpenSCAD already has a 2-D Polygon Offset Transform that returns a 2-D polygon object.  The new function would use CGAL to return an offset polygon path list from a polygon path list.  One use is creating 3D polyhedron skins.  

http://forum.openscad.org/Polygon-Offset-Function-td17186.html

3D Minkowski-universe-difference for offsetting 3D objects is many orders of magnitude too slow for offsetting complex 3D objects.  3D offset doesn't exist yet.

Other uses might include laser kerfs, gear mesh clearance, thin wall vases and CAM tooling paths.

A more general solution would be to have a function that could extract the polygon path list of an arbitrary 2-D object, but that would require the object to be rendered, which breaks the single pass execution flow.

kintel, 2016-05-01:
Note: We use Clipper, not CGAL, for 2D offset.

This would be a stop-gap measure until we properly support assigning shapes to variables, and thus support using any 2D shape as a parameter to e.g. skin(), which may or may not include being able to query 2D shapes in user-space.

I think we should rather use our scarce resources towards a proper solution, especially considering that we'd have to keep supporting such a stop-gap feature in the future if we include it now.

to-the-nth, 2016-05-15:
Agreed.  It appear to be straight forward to call clipper from a standalone program and precompute data sets to include in OpenSCAD.

kintel, 2016-05-15:
@to-the-nth btw., one potential solution could be to allow OpenSCAD to export .scad/.csg files with 2D object resolved in to a single polygon() statement. Not quite sure how the interface should look though.

to-the-nth, 2016-05-16:
An export .scad/.csg or even an echo statement would be great!  Thanks

kintel, 2016-05-16:
We've briefly discussed that earlier. This also applies to 3D objects. Since it's an interactive feature, it would later be possible to do this for any subtree (convert it into it's polygon()/polyhedron() representation).

to-the-nth, 2016-05-24:
Marius, do you think this will show up on the radar screen anytime soon?

kintel, 2016-05-24:
Hard to say. Corner cases tend to make such things time consuming. First step is to make sure we're building the right thing.
Some help would be appreciated:

* Is adding an File->Export->polygon and a ```-o <out.scad>``` option make sense? This would export an .scad file with a single ```polygon([...]);``` statement.
* Do you have a relatively simple but concrete example of what you'd like to build?
* If you want to extract the outlines, you'd have to somehow parse the data. Do you have any ideas how this best would be done?
* How would you deal with multiple-outlines-multiple-holes polygons?

to-the-nth, 2016-05-30:
I'm still looking at different tool flow options.

Dan wrote an interesting converter that can export OSCAD polygons from Inkscape that has to deal with a lot of the issues.  He used multiple polygons for each outline and differences instead of a path list.  I am going to see how far I can get with inkscape scripting for initial data set generation.  And inkscape does have a full up offset function. If I am not successful, at least I'll have a more crisp problem statement.

http://www.thingiverse.com/thing:25036

`
// Module names are of the form poly_<inkscape-path-id>().  As a result,
// you can associate a polygon in this OpenSCAD program with the corresponding
// SVG element in the Inkscape document by looking for the XML element with
// the attribute id="inkscape-path-id".

// fudge value is used to ensure that subtracted solids are a tad taller
// in the z dimension than the polygon being subtracted from.  This helps
// keep the resulting .stl file manifold.
fudge = 0.1;

module poly_path3336(h)
{
  scale([25.4/90, -25.4/90, 1]) union()
  {
    difference()
    {
       linear_extrude(height=h)
         polygon([[-29.426757,-302.441410],[-98.268557,-99.050780],[-304.844723,-40.457030],[-132.682617,87.869140],[-140.791987,302.441410],[34.452143,178.359380],[236.016603,252.378910],[172.163083,47.367190],[304.844723,-121.458980],[90.135743,-124.082030],[-29.426757,-302.441410]]);
       translate([0, 0, -fudge])
         linear_extrude(height=h+2*fudge)
           polygon([[-58.729497,-156.115230],[-15.411137,-116.193360],[42.032223,-129.246090],[17.450193,-75.710940],[47.616213,-25.109370],[-10.895507,-31.945310],[-49.698247,12.378910],[-61.276367,-45.380860],[-115.422857,-68.587890],[-64.067387,-97.449220],[-58.729497,-156.115230]]);
    }
  }
}

poly_path3336(5);`

Neon22, 2016-05-30:
Dan's plugin is good. THe V7 I made has not yet been accepted by him but IMHO it is an improvement as it brings the shapes out to variables, and allows resizing.
Give it a try:
- http://www.thingiverse.com/thing:1065500

stefano2734, 2026-09-20:
Is this solved or improved after 10 years without messages?

jordanbrown0, 2026-09-20:
The proposal is to use CGAL's offset mechanism, and we still use Clipper's, so I'll go with "no".

Neon22, 2026-09-24:
I appreciate the prompt to update my Inkscape plugin from so long ago.

gsohler, 2026-09-24:
are youreferring to skin ?

https://pythonscad.org/reference/extrusions/#skin