openscad/openscad #1639: OpenSCAD + Bullet Physics Simulation

koppi, 2016-05-06:
Hi, I've created a marble run with Duplos,
see https://www.youtube.com/watch?v=IwdbtfiEo0A .

Background: Yesterday, I've integrated OpenSCAD into my physics simulation software:
* The Bullet Physics Playground: https://github.com/bullet-physics-playground

Here's some recent Video recordings:
* A basic CSG testing scene: https://www.youtube.com/watch?v=WbElbcZ6Sa0
* Experiments with gears and motors: https://www.youtube.com/watch?v=xy7qDgStMXk
* Experiment with various OpenSCAD scripts I've found on Thingieverse
https://www.youtube.com/watch?v=c8glPDbG1FI

Gitter chat at https://gitter.im/bullet-physics-playground/bpp

t-paul, 2016-05-06:
Very nice! I'm not sure why this is posted here. I'd suggest to post on the Forum and/or G+ Community.

closed 2016-05-06 by t-paul

koppi, 2016-05-06:
@t-paul where is the G+ community?

I have this link: https://plus.google.com/+openscad/posts but there is no content.

t-paul, 2016-05-06:
[OpenSCAD Google+ Community](https://plus.google.com/u/0/communities/112968258499418200895)

koppi, 2016-05-06:
@t-paul see https://github.com/openscad/openscad/issues/1640 .

polkovnikov-ph, 2017-05-12:
This place is handy too.