openscad/openscad #1640: Fix link to G+ community

koppi, 2016-05-06:
The link under "g+" on the left on the page links to https://google.com/+openscad

[image]

But the community is acually found at: https://plus.google.com/u/0/communities/112968258499418200895

Please fix the link on the page accordingly.

kintel, 2016-05-15:
The owner of that Google+ community is missing in action. We're trying to get in touch with him so we can get that link updated. Any help would be appreciated.

koppi, 2016-05-15:
@kintel https://google.com/+openscad is not a community but a personal G+ user account.

kintel, 2016-05-15:
@koppi I know - the goal was to use that link for the G+ community, but without admin access I cannot do that. A workaround would be to accept the weird URL for now.

MichaelPFrey, 2018-10-27:
for those that have not yet gotten the news:
https://www.blog.google/technology/safety-security/project-strobe/

> Finding 1: There are significant challenges in creating and maintaining a successful Google+ product that meets consumers’ expectations.
> 
> Action 1: We are shutting down Google+ for consumers.

So, within in a year, this issue will self resolve.

closed 2018-11-06 by kintel