openscad/openscad #1642: Please enable package builds for Ubuntu 16.04

koppi, 2016-05-09:
As of 2016-05-09 the following link is 404:
* http://download.opensuse.org/repositories/home:/t-paul/xUbuntu_16.04/

t-paul, 2016-05-09:
The first attempt to enable this failed as OBS never had the Universe repo and the list of build dependencies from there is now too big to just rebuild those too.

I just found that the latest comments in https://bugzilla.novell.com/show_bug.cgi?id=977234 indicate OBS added Universe meanwhile, so I'll have another look.

t-paul, 2016-05-09:
OBS indeed added the Universe repo which makes things much easier. The 64bit build is though already, so I'm closing the issue. The 32bit build seems to be stuck somewhere in the scheduler, but OBS should fix that hopefully automatically.

closed 2016-05-09 by t-paul

koppi, 2016-05-09:
Thank's for the quick fix, I just installed openscad-nightly (2016.05.08.nightly (git d3db6a6)) on Ubuntu 16.04 amd64, looks good!

koppi, 2016-05-09:
Please do not forget to update the list at http://www.openscad.org/downloads.html#snapshots