openscad/openscad #1643: Is there a 2015.03-3 install package for Linux?

WZ9V, 2016-05-12:
The most recent install I could find to use with Linux was 2015.03

I'd like to get the missing syntax highlighting that looks like it was fixed in 2015.03-2

kintel, 2016-05-13:
@hyperair Looks like the latest PPA release is 2015.03-1. Is there an easy way to bump that to 2015.03-2?

FYI: 2015.03-3 is a Mac-only security fix.

WZ9V, 2016-05-14:
I don't know who maintains the PPA but yes it is one version behind and that version has syntax highlighting broke when using the new editor.  I'm trying to compile my own right now but it would be much easier if the PPA were up to date.

hyperair, 2016-06-09:
Mm okay. I didn't realize there was an update.

leoheck, 2016-07-02:
I am using, version 2015.03-1 from the OpenScad PPA in Ubuntu 16.04 and the syntax highlight is ok and it is a very good stuff... However, I found a plugin for sublime text that has command completion, and syntax highlight which can also be used.

leoheck, 2016-07-02:
Maybe this issue can be closed.

closed 2018-04-06 by t-paul