openscad/openscad #1645: Create persistent cache feature

MarPiRK, 2016-05-14:
It would be great to add persistent cache, so when openscad is run in shell all cache is shared between separate runs. It'd be very useful regarding #1632.
Also I'd like to announce, that I'm planning to create CMake scripts supporting compiling OpenSCAD files and there it would be even more useful, especially when project has shared modules.
It'd be nice if there was shell argument with directory, where cache would be saved – it'd ease my work with CMake script.

t-paul, 2016-05-14:
See also related discussion in #1437.

kintel, 2016-05-15:
See also https://github.com/openscad/openscad/wiki/Ideas-for-Development-Tasks#persistant-caching

MarPiRK, 2016-05-22:
I have idea how to do it. There may be used some sort of protocol buffer (like [Cap'n Proto](https://capnproto.org/), which I'm familiar with), which would manage entire cache.

I'd use protobuf for three reasons:

1. It's fairly easy to create whatever, platform independent cache structure we'd like to have.
2. There would be possibility, within single library, to implement single process memory cache (like it's done currently), local file cache (what was proposed in this issue) and remote cache (via RPC calls), what would enable distributed compiling over network.
3. Multiple processes could use the same cache simultaneously without any kind of collisions and conflicts.

t-paul, 2016-05-22:
It would be nice to use something where we don't need to write the server part ourselves (e.g. like [memcached](https://memcached.org/)). The page you linked mentions sandstorm.io, so is there also some server side implementation that could be used (and installed independently) by just implementing the client side in OpenSCAD? I've seen the talk about [sandstorm.io](https://sandstorm.io/) from Asheesh Laroia at DebConf15 which was interesting.

MarPiRK, 2016-05-22:
In Cap'n Proto server side is application specific, so there is no universal server. I also thought about some kind of NoSQL database (like serverless [UnQLite](https://unqlite.org/)), but serialization-deserialization part would need to be handled in OpenSCAD code. Cap'n Proto handles this process automagically (and platform independent) basing on schemes, which are conveniently translated into C++ headers (look [there](https://capnproto.org/cxx.html)).

MarPiRK, 2016-05-22:
Or we can use Cap'n Proto just for serialization of data structures (like Geometry, CGAL), which would be saved in memcached.

MarPiRK, 2016-05-22:
The problem with memcached (if I understand how it works correctly) is that it requires separate process to be run. With capnp entire cache would be transparently handled by capnp no matter what type of cache it would be. Server part would simply receive and send capnp objects, so it's nothing complicated.

kintel, 2016-05-22:
@t-paul Didn't you play around with a cmake target a while back (or was that someone else)?

kintel, 2016-05-22:
@MarPiRK There are a few different components to this, some of which require some more thought/

These should probably be written up somewhere, just need someone to start chipping away at it.
A simple version is a start, we should just keep in mind what we're _not_ doing and why:

1) What do we cache? We have multiple internal representations, which depend on what external libraries are used. We tend to lose data when converting between those (e.g. by downsampling). The internal cache isn't _one_ cache, but several independent caches. Some internal cleanup would be good.

2) In what format do we cache data? While binary memory dumps are fast, it's not generally possible to reconstruct in-memory data structures from them, e.g. since we're using external libraries which require said data structures to be built in a certain way. Storing serialized, parseable, data on a known/standard format seems like a cleaner starting point. Keep in mind that caching is primarily there to save minute-long computations. Deserializing/parsing is orders of magnitude cheaper, even if done naively.

3) How do we manage compatibility between versions/platforms of OpenSCAD? Should we aim to allow central platform/version independent caching, or lock a cache to a certain OpenSCAD installation? Offering caching as a service could be interesting, but we might want to start out with something less involved.

3) How do we manage (store/retrieve) the cache. Separate vs. in-process. Database vs. filesystem. What transport should we support, how does this translate to OS installers?

t-paul, 2016-05-22:
The cmake stuff is in the `cmake-build` branch, but it's really just a beginning. https://github.com/openscad/openscad/pull/1048

MarPiRK, 2016-05-27:
1. Different caches could be described with different capnp schemes, so it's basically no problem.
2. Data construction would be done on the fly. There would be one method to retrieve/save to cache, which would translate between library data structure and capnp data structure transparently.
3. Backwards compatibility is one of features of capnp.
4. With capnp it could be implemented as separated and in-process. There is no such thing as database, but file storage (or basic memory storage) is possible. Capnp is platform and OS independent, so there would be no problem with OS-OS connection and instalation.

kintel, 2016-05-27:
@MarPiRK Sounds reasonable. The CGAL cache is the most important one, so that would be a good start. To my knowledge, the easiest way of getting good bidirectional data to/from CGAL's Nef polyhedron is using iostreams with the .nef3 file format.

capnp is an implementation detail. We should probably have a some reasonable meta-data layer above that payload.

MarPiRK, 2016-05-27:
It's definitely not just an implementation detail. `capnp` binary creates (from schemes) actual C++ interface sources with actual data structures, so creating another layer is in my opinion pointless.

I'll look at it more in-depth probably next week.

t-paul, 2016-05-27:
So the build process will need to extended too. We need to make sure this will work on all 3 supported platforms, Linux (I see Debian has version 0.5.3 in testing), MacOS and Windows (means support in [MXE](http://mxe.cc/) which is used as build environment for the Windows binaries).

dumblob, 2019-01-20:
Any news on this? I need to use slicing a lot in my models and one render takes at least 15 minutes and when not careful enough, it takes infinity and therefore I need to kill openscad, open the file again, adjust some of `$fs` `$fn` etc. and try again while waiting for the many same computations again and again...

stefano2734, 2026-09-20:
Is this problem solve after 7 years? Related issues are solved.

jordanbrown0, 2026-09-20:
> Is this problem solve after 7 years?

No.