openscad/openscad #1646: scale should automatically set z component to 1 for 2D objects

thehans, 2016-05-15:
Previewing the example code shows no cutouts of the 2d object.  However, the f6 render is correct.
This is due to the z being scaled and holes being completely surrounded by the parent object 

One workaround is to explicitly scale the z to 1:    scale([1/3, 1/3, 1])

OpenSCAD should automatically set z scale to 1 for 2D objects

    //_sierpinski(0,0,100,3);
    sierpinski(100,3);

    module sierpinski(s, n) {
      difference() { 
        square(s, center=true);
        _sierpinski(0,0,s,n);
      }
    }

    module _sierpinski(x, y, s, n) {
      translate([x,y]) {
        if (n>0) {
          i = n-1;
          scale(1/3) {
            square(s, center=true);
            if (i>0) {
              _sierpinski(-s,-s, s, i);
              _sierpinski(-s, 0, s, i);
              _sierpinski(-s, s, s, i);
              _sierpinski( 0,-s, s, i);
              //_sierpinski( 0, 0, s, i);
              _sierpinski( 0, s, s, i);
              _sierpinski( s,-s, s, i);
              _sierpinski( s, 0, s, i);
              _sierpinski( s, s, s, i);
            }
          }
        }
      }
    }

MichaelAtOz, 2017-02-06:
Note also should be done when using this form: `scale[x,y,z]`
Should probably also generate a warning if z<>1? in this instance.

MichaelPFrey, 2018-11-17:
I would suggest a slightly different workaround. The workaround `scale([1/3, 1/3, 1])` can be simplified to `scale([1/3, 1/3])`, the Z-Axis silently being defaulted to 1.

MichaelPFrey, 2019-02-22:
see also #166

jordanbrown0, 2024-11-23:
The reason that Z scale is a problem for 2D objects is that the previewer doesn't actually handle 2D objects - instead, it turns them into 3D objects with a height of 1, that then get affected by Z scale.

A better solution would be to present 2D objects as infinitely thin, like a polyhedron with only one face.  The problem with doing that is that the previewer uses its 3D OpenGL-based schemes for doing Boolean operations on 2D objects, and those schemes do not work on infinitely-thin objects.  We could render 2D objects before preview (as we do for linear extrude), but that loses color.

closed 2026-05-06 by jordanbrown0