openscad/openscad #1647: Kill Boost Filesystem V2 workarounds

kintel, 2016-05-15:
We have a bunch of workaround for compiling against Boost Filesystem V2.
V2 was replaced with V3 as default in Boost 1.46. I don't believe we support building on any systems which use Such an old version of Boost any longer, so we can start killing off these workarounds.

See https://github.com/openscad/openscad/blob/master/src/boosty.h

@donbright I believe you're the original author of boosty.h. If you have any opinions, feel free to chip in.

donbright, 2016-08-12:
" in a few years, this file should be deleted as unnecessary."

The Prophecy has come to pass... the Wizard has arrived to heal the Codebase... all praises to the Open Source...

donbright, 2016-09-16:
just to be clear... yes, take it all out, remove it.. . . i would have done it myself by now but I was waiting to hear what @DrCodeWizzard  to do a pull request or make a comment...

DrCodeWizzard, 2016-09-16:
sorry I've been a bit busy with my regular work. I thought i already merged this in into the code base? Also while developing this change i realized that not all of boosty.h can be removed due to version of boost installed on the test rig.

kintel, 2016-09-16:
Yes, this was already merged from #1717. I just forgot to close this.

closed 2016-09-16 by kintel