openscad/openscad #1649: lookup returns junk for nested lists

fukanchik, 2016-05-23:
This code:
    a=[[0,1],
       [1,33],
   [2,[[0,17],
       [3,3]]],
   [3,71]];

echo(a);

v1 = lookup(0, a);
echo("[0]=",v1);
v2 = lookup(1, a);
echo("[1]=",v2);
v3 = lookup(2, a);
echo("[2]=",v3);
v4 = lookup(3, a);
echo("[3]=", v4);`

prints:

ECHO: [[0, 1], [1, 33], [2, [[0, 17], [3, 3]]], [3, 71]]
ECHO: "[0]=", 1
ECHO: "[1]=", 33
ECHO: "[2]=", 52
ECHO: "[3]=", 71

expected:
ECHO: [[0, 1], [1, 33], [2, [[0, 17], [3, 3]]], [3, 71]]
ECHO: "[0]=", 1
ECHO: "[1]=", 33
ECHO: "[2]=", [[0, 17], [3, 3]]
ECHO: "[3]=", 71

doug-moen, 2016-05-23:
Garbage in, garbage out. `lookup` takes a number as first argument, a list of pairs of numbers as the second argument, and returns a number. It simulates a continuous function f:number->number by doing linear interpolation on members of a lookup table. In your example, the second argument is not a list of number pairs.

OpenSCAD does not have a built in function that does what you want (look up a value in a list of [key,value] pairs). (The `search` function looks like it should do this, but many people find it unusable, and avoid it.)

What most people do is implement it themselves.
```
a=[
  [0, 1],
  [1, 33],
  [2, [[0,17],[3,3]]],
  [3, 71]
];

echo(find(2, a));

function find(key, alist, i=0) =
    i >= len(alist)
        ? undef
    : alist[i][0] == key
        ? alist[i][1]
        : find(key, alist, i+1);
```

fukanchik, 2016-05-24:
I see. Thanks

closed 2016-05-24 by fukanchik