openscad/openscad #1650: Open Source CAD exchange format.

Evil-Spirit, 2016-05-24:
We are working hard on SolveSpace. SolveSpace use Mesh/NURBS for geometry, but has history-based approaches.
You are working on a good tool which is really not like something else. OpenSCAD uses pixel operations for booleans as fast visualization, and CGAL as boolean operations for generating some STL meshes. You have used scripts for bodies representation. It's some sort of history based approach.
There is also FreeCAD which is working on NURBS and has history-based approaches.
There is also Antimony which uses F-Rep and visual scripting for building bodies.

So, what do you think about making the ultimate exchange format for history-based geometry. This format must be not snapped to interal format (mesh, b-rep, f-rep, etc), just geometry with building process. It must operate on 2d primitives, 2d/3d sketches of primitives and operations (extrusion, revolve, etc) with combination function (union, subtract, etc).

My first draft:

<cad>
 <sketches>
  <sketch type="2d" id="1">
   <contour id="1" combine="add">
    <start point="0 0"/>
    <line point="0 10"/>
    <line point="10 10"/>
    <line/>
   </contour>
   
   <circle id="2" op="sub" center="5 5" radius="1">
  </sketch>
 </sketches>

 <bodies>
  <body id="1">
   <feature id="1" op="extrude" combine="add">
    <sketch id="1" transform="mtx"/>
   </feature>
  </body>
 </bodies>

</cad>
We can use some general approaches from dxf and svg.

Can you think about it?

doug-moen, 2016-05-24:
I'm not familiar with the term "history based approach", but I think you are talking about the "Constructive Solid Geometry" (CSG) representation, where a solid object is represented by a tree, with operations at the interior nodes, and primitive objects at the leaf nodes. See: https://en.wikipedia.org/wiki/Constructive_solid_geometry

So, I think you want a CSG based exchange format that can be used to exchange 2D and 3D objects between different solid modeling tools, like SolveSpace, OpenSCAD, FreeCAD, Antimony, and so on.

The first question is whether such a format already exists. No sense in reinventing the wheel.

OpenSCAD scripts are already being used as a CSG interchange format, because there are a number of existing solid modeling tools that attempt to read OpenSCAD. There is: ImplicitCAD, OpenJSCAD, IceSL, maybe others. The problem is that it's a lot of work to implement a bug-compatible OpenSCAD implementation. Although we have a user-maintained wiki, there are a lot of dark corners of the language that aren't documented, and, from my experience, it takes a few years to learn all of these dark corners. I kind of doubt that anyone has build a fully backward compatible implementation of the OpenSCAD language, due to complexity and lack of the necessary documentation. Plus, it's a moving target.

OpenSCAD exports a simplified subset of the language, called CSG. This is also used as an interchange format, and it is much easier to implement than the full OpenSCAD language. For example, FreeCAD imports OpenSCAD "CSG" files. Even here, there are drawbacks, due to the fact that OpenSCAD only supports the polyhedron and the polygon as the representation of 3D and 2D shapes. In OpenSCAD, either in the full language, or the CSG subset, there is no way to distinguish between a circle and a regular polyhedron, or between a cylinder and a regular polyhedral prism, because the former is represented by the latter in OpenSCAD. In OpenSCAD, you use the circle primitive to create a regular polyhedron, and you use the cylinder primitive to create a prism. That's not the case for other modeling tools, like Antimony.

A more general purpose and less OpenSCAD-specific set of primitives and operations might be desirable in an "ultimate" interchange format.

There is periodically some discussion about supporting the STEP format (#893). Although STEP looks like it is general enough to support a wide variety of modeling tools, it has two major problems. One is that it's a proprietary standard, and it would probably cost > $1000 to purchase all of the necessary standards in order to learn the format. The other problem is the size and complexity of the format.

If someone were to create an "ultimate" CSG interchange format for open source solid modeling tools, I think we'd have the following requirements, in order for it to be successful:
* The standard should be distributed as open source.
* It should be simple.
* It should be well documented.
* There should be a group of people who "own" the standard, with representatives from at least 3 open source solid modeling tools. Probably with its own home on github.
* There should be an open source library on github for reading and writing the file format.

As for the content of the standard, as I said, I think it should be simple. A document consists of a CSG tree, with operations at the interior nodes, and primitive shapes at the leaf nodes. The main job of the design committee would then be to agree on a set of operations and primitive shapes. (I don't agree with the "sketches" section of your proposal, because I don't see any natural way for OpenSCAD to generate the contents of this section.)

kintel, 2016-05-24:
It's a good idea. However, I think it will be quite challenging to bridge content from different approaches in one single format, i.e. finding the least common denominator between scad, b-rep, f-rep and meshes.

Did you think about how to evaluate things like f-rep designs in a b-rep system? Would the f-rep system convert to a b-rep or mesh for export, or would the b-rep system simply ignore f-rep objects?

Similarly, how would a b-rep system like SolveSpace evaluate a convex hull of two primitives? We could export to a mesh, but then we lose all the information, so we'd probably have to select a specific feature set for export. b-rep systems are traditionally not very good at processing meshes, making this less desirable.

doug-moen, 2016-05-24:
We can look at how people support the already existing CSG file format. For example, FreeCAD: http://freecadweb.org/wiki/index.php?title=OpenSCAD_CSG

On import, some operations are simply not supported, such as projection, surface, minkowski, glide, path, subdiv, null. (I don't know what glide, path and subdiv are; does some tool other than OpenSCAD implement them?) On export, features not supported by the CSG format are converted to polyhedra.

As I said, many other tools already support CSG, including spline-based and F-Rep based tools, so a new format designed with interoperability in mind should at least do better than the status quo.

We could start by defining a "least common denominator" set of operations and primitives, those known to be supported by all of the modeling tools that we are trying to support. Then we could extend this with operations that are native in some tools, but which could be simulated or approximated in other tools, possibly with library support.

kintel, 2016-05-25:
My main fear is that the "least common denominator" is a bit small to be really useful; a strict subset of .csg. Related to this: We've discussed a few times making the .csg export configurable to be able to select the supported output primitive and operators, to e.g. allow properly exporting to tools like FreeCAD where we know what subset is supported.

Evil-Spirit, 2016-05-25:
"Lest common denominator" can't be small while we have no any denominators. It's infinity greater than nothing. I mean, even simple sketches exchange can be applicable: we have many approaches  in absolutely different software, so we can start to standartize some small subset, while the new standard will grow, some of software will evolute to fit it.

Evil-Spirit, 2016-05-25:
> I don't agree with the "sketches" section of your proposal, because I don't see any natural way for OpenSCAD to generate the contents of this section.

Sorry, I don't even near the OpenSCAD, but I've just found something called "dxf_linear_extrusion". I will wonder if OpenSCAD can't do the same by it's code (not using dxf as source).
http://hackaday.com/2010/01/12/how-to-make-a-printable-ces-badge/

Evil-Spirit, 2016-05-25:
@doug-moen,
> but I think you are talking about the "Constructive Solid Geometry" (CSG) representation
...
So, I think you want a CSG based exchange format that can be used to exchange 2D and 3D objects between different solid modeling tools, like SolveSpace, OpenSCAD, FreeCAD, Antimony, and so on.

Yes, you are right.

> OpenSCAD scripts are already being used as a CSG interchange format, because there are a number of existing solid modeling tools that attempt to read OpenSCAD. There is: ImplicitCAD, OpenJSCAD, [...]

If OpenSCAD has the good subset of it's CSG format, it is fine. But the format - is the just data with some obvious interpretation rules and if we just need to use OpenSCAD compiler as part of format librariy - it's not good idea.

> there is no way to distinguish between a circle and a regular polyhedron, or between a cylinder and a regular polyhedral prism, because the former is represented by the latter in OpenSCAD. In [...]

This is definitely bad. We have to distinguish all the curved things when we can. For OpenSCAD importing it's not the problem, but if we will export this as piecewise segments - it's the big problem. But if we could have any exchange formats/standarts, OpenSCAD and other software will get some ambitions/requirements to fit and support format. When we have circle - we must directly assign radius to it, if we have circle extrusion - we must have cylinder and there can't be any options. May be even this will be just bounded set of functions inside openscad and other software to be standart-compatible. May be even it must be an option "Show/Enable/Use only The Open Standard compatable Tools/Functions/Features".

> There is periodically some discussion about supporting the STEP format (#893). Although STEP looks like it is general enough to support a wide variety of modeling tools, it has two major problems. [...]

STEP is a one big issue. This is designed for a "BIG and POWERFUL" proprietary CAD/CAM/PLM/BIM with a huge functionality list. FOSS CADs is far behind it, and far ahead it in some directions, and definitely can't use such formats for exchange. We have to design new format to fit the realy existent features. We don't want to just look at STEP features list which we can't support now or even never wanted to implement in future.

>If someone were to create an "ultimate" CSG interchange format for open source solid modeling tools, I think we'd have the following requirements, in order for it to be successful:
* The standard should be distributed as open source.
* It should be simple.
* It should be well documented.
* There should be a group of people who "own" the standard, with representatives from at least 3 open source solid modeling tools. Probably with its own home on github.
* There should be an open source library on github for reading and writing the file format.

I am absolutely AGREE, sounds like a plan to start moving in this direction.
IRC: #make-cad-format

Evil-Spirit, 2016-05-25:
@kintel 
> Did you think about how to evaluate things like f-rep designs in a b-rep system? Would the f-rep system convert to a b-rep or mesh for export, or would the b-rep system simply ignore f-rep objects?

I think Antimony uses f-rep for internal representation (generate body function and ray-tracing). It seems what functional blocks (script execution) can produce any result (mesh, csg, etc). It's challange for the developers to make it.

> Similarly, how would a b-rep system like SolveSpace evaluate a convex hull of two primitives? We could export to a mesh, but then we lose all the information, so we'd probably have to select a [...]

Solvespace uses history-based approach which can be easily converted to CSG representation.

> OpenSCAD scripts are already being used as a CSG interchange format
I've learned what is .csg and this is nearly what I thinking about, but this not supports sketches as set of primitives (segments, arcs, beziers, etc). All this things is performed through "polygons".
So, while the export or import this can be converted in polylines and loose all the primitives data. UnitedCSG can contain code for picewise linear approximation of the sketches.

doug-moen, 2016-05-25:
I am familiar with the internals of Antimony and ImplicitCAD. They are both CSG tools using the F-Rep internal representation. They are similar to OpenSCAD, in that they provide a basic set of primitive shapes, and operations on those shapes. There is a "lowest common denominator" of primitives shared by OpenSCAD and the F-Rep CSG tools: sphere, cuboid, etc, intersection, translate, etc. There are F-Rep operations not easy to duplicate in OpenSCAD, like filleting and rounded corner primitives, or the twist operator, while OpenSCAD has some operations not easy to duplicate in F-Rep, like minkowski sum and convex hull.

If you wanted to support a superset containing all of the primitives found both OpenSCAD and in the F-Rep tools, then you need to deal with questions like, how does Antimony import a minkowski sum, how does OpenSCAD import a twist operator.

The simple answer is that the import fails if it contains unsupported primitives. That will have to be our starting point.

The complex answer is that a library could be written to convert unsupported operations into a simpler form that can be imported.
* One strategy: render unsupported primitives into polytopes, which I think is the lowest common denominator representation that everybody can read.
* In some cases, a sophisticated, general operation can be reduced to a simpler, more specialized operation.
   * For example, ImplicitCAD doesn't support general transformation matrices (aka multmatrix in OpenSCAD). However, it's not hard to reduce a transformation matrix into a sequence of simpler transformations: translate, rotate, scale, skew. ImplicitCAD supports all of these except skew.
   * For example, minkowski sum is hard to implement in an F-Rep system, but the most common idiom used by OpenSCAD is minkowski sum of an object with a sphere, and that is equivalent to the 'dilate' operator found in some F-Rep systems, which is relatively easy to implement. So we could automatically reduce minkowski sum of a sphere to dilate.

mkeeter, 2016-05-25:
(Antimony developer here)

I strongly agree with @doug-moen and @kintel – the difference in underlying representations makes this near-impossible to do in a complete fashion, and reducing the space to make it feasible makes it less useful / powerful. 

I'm not sure about the shape of the feasible / useful tradeoff curve – there may be a subset that's both feasible and useful, but I'm not yet convinced.

Translating history / design intent between representations is also incredibly challenging.  For example, if we wanted OpenSCAD to *properly* open a `.sb` file, it would need to flattens the set of graph nodes into functions.  We'd lose 2D layout and nested hierarchy, and each node's script would need to be translated into an OpenSCAD function.

Anything less would be more feasible, but would lose most of the design intent.   With this meta-information removed, a CSG interchange representation isn't much more useful than a simple mesh.

Evil-Spirit, 2016-05-25:
@mkeeter,

It's not the main point to support the full design intent, but to have just ability to exchange some tings through the FOSS CAD. It's also as something like the point of crystalisation, the theme where developers of the different CADs can mave in the same direction and create compatible infrastructure.

doug-moen, 2016-05-25:
Hi @mkeeter, glad you joined our party. I admire your work.

I think there are two reasons for a standard open-source CSG interchange format to exist.
 1. Generally much more compact than an STL/AMF/3MF file, and also resolution independent. I remember reading that the AMF working group is working on a CSG based extension to AMF. Converting F-Rep models to STL files can be quite horrendous. It would be nicer to export to a CSG file, then let the slicer directly render the CSG file for 3D printing.
 2. The way that "design intent" is represented in OpenSCAD vs Antimony vs FreeCAD is so different that an automatic translation is infeasibly difficult. But importing a mesh really sucks, because ALL STRUCTURE IS LOST. A nice reasonable compromise is to throw away the "design intent", but import the primitive objects and operations. Then you can add the design intent back by hand. A CSG tree is much more reusable than a mesh. Unlike a mesh, a CSG tree is resolution independent. Consider the fact that FreeCAD offers OpenSCAD import, despite the limitations: they must find this more valuable than importing a mesh, or it wouldn't have been implemented.

kintel, 2016-05-25:
As a side note: The .csg format originated in a small project by Adrian from RepRap, who wanted to perform CSG in the slicer and wanted a simple way of avoiding massive STL files when it would be significantly faster (and higher resolution) to reduce it to a 2D problem during slicing.

kintel, 2016-05-25:
Anyway, it sounds like a least common denominator would be the three CSG operators, a subset of the affine transformations, the three basic primitive (cube, sphere, cylinder), as well as the corresponding 2D operations. In addition, we could include a subset of the 2D-to-3D extrusions (rotate_extrude around a fixed origin and linear_extrude without twist, scale or rotate). Everything else would have to be rendered out as a mesh. ..and possibly import from certain types of files.
As it stands today, this could be implemented as a strict subset of our .csg file format.

@Evil-Spirit Does SolveSpace support full CSG between imported meshes and NURBS?

Evil-Spirit, 2016-05-25:
@kintel,
SolveSpace can convert NURBS into mesh and make booleans with it.

Evil-Spirit, 2016-05-25:
> who wanted to perform CSG in the slicer and wanted a simple way of avoiding massive STL files when it would be significantly faster (and higher resolution) to reduce it to a 2D problem during slicing.

This is good point.

> Anyway, it sounds like a least common denominator would be the three CSG operators, a subset of the affine transformations, the three basic primitive (cube, sphere, cylinder), as well as the [...]

Yes, this is enough for the begining. 
Just CSG tree and ability to make extrusion and revolution from closed sketch(contour, loop) of primitives (segments, arcs, beziers, etc). It's like the current .csg, but with extensions like primitives (not polygons). We could use easy to parse(json) or more general(xml) structure as container.  Format must be self-descriptable.

doug-moen, 2016-06-09:
@Evil-Spirit said "We could use easy to parse(json) or more general(xml) structure as container. Format must be self-descriptable."

I prefer JSON because it is simpler and easier to deal with. I prefer to design the interchange format in JSON. Then, if it becomes necessary later on, it's easy to specify a mapping of the JSON schema onto XML. Going the other direction is not as easy, because XML is more complex than JSON and has more features.

So, here's what a JSON based format might look like. This format uses all 6 of the JSON types (null, bool, number, string, array, object). Each node in the CSG tree is represented by an object with a "name" field (name of the primitive), plus additional parameter fields. The parameters can have any JSON type.

OpenSCAD:
   translate([1,2,3]) sphere(10)

as JSON:
  {"name":"translate", "offset":[1,2,3], "object":
     {"name":"sphere", "radius":10} }

Evil-Spirit, 2016-06-09:
@doug-moen,
Actually, I don't like JSON because of it's less expressivity. But we can just make even simplier than JSON and with the ununique keys:
```json
{
    "file"   : "test.ucsg",
    "author" : "Evil-Spirit",

    "union": {
        "difference": {
            "extrusion": {
                "transform" : [
                    1, 0, 0, 0,
                    0, 1, 0, 0,
                    0, 0, 1, 0,
                    0, 0, 0, 1
                ],
                "sketch2d": {
                    "start" : [-10, -20],
                    "line"  : [ 10, -20],
                    "line"  : [ 10,  20],
                    "line"  : [-10,  20],
                    "line"  : "start"
                },
                "length" : 20
            },
            "extrusion": {
                "transform" : [
                      0, 1, 0, 0,
                      1, 0, 0, 0,
                      0, 0, 1, 0,
                    -10, 0, 0, 1
                ],
                "circle2d": {
                    "center" : [-10, -20],
                    "radius" : 10
                },
                "length" : 30
            }
        },
        "extrusion": {
            "transform" : [
                1, 0, 0, 0,
                0, 1, 0, 0,
                0, 0, 1, 0,
                0, 0, 0, 1,
            ],
            "sketch2d": {
                "start" : [-20, -30],
                "line"  : [ 20, -30],
                "line"  : [ 20,  30],
                "line"  : [-20,  30],
                "line"  : "start"
            },
            "length" : -20
        }
    }
}
```

The same as XML
```xml
<ucsg version="0.0">
	<info>
		<name>detail</name>
		<author>Evil-Spirit</author>
		<description>
			This detail is simple test
			for the new file format.
		</description>
	</info>
	<geometry>
		<union>
			<difference>
				<extrusion length="20" transform="1 0 0 0 0 1 0 0 0 0 1 0 0 0 0 1"> 
					<start pos="-10 -20"/>
					<line pos=" 10 -20"/>
					<line pos=" 10  20"/>
					<line pos="-10  20"/>
					<line pos="start"/>
				</extrusion>
				<extrusion length="20" transform="0 1 0 0 1 0 0 0 0 0 1 0 -10 0 0 1"> 
					<circle pos="-10 -20" radius="10"/>
				</extrusion>
			</difference>
			<extrusion length="-20" transform="1 0 0 0 0 1 0 0 0 0 1 0 0 0 0 1"> 
				<start pos="-20 -30"/>
				<line pos=" 20 -30"/>
				<line pos=" 20  30"/>
				<line pos="-20  30"/>
				<line pos="start"/>
			</extrusion>
	</geometry>
</ucsg>
```