openscad/openscad #1655: Crash using mirror

kintel, 2016-05-30:
From the [Debian tracker](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=825295):

Using mirror can cause a hard crash during rendering (F6), e.g.
```
mirror() {
  difference() {
    cube(size = [2, 15, 18]);
    translate([6, 15, 6]) {
      difference() {
        cylinder(h = 9, r = 10);
        cylinder(h = 9.5, r = 4, $fn = 20);
      }
    }
  }
}
```

Stack trace:
```
#0	0x00000001008ec953 in CGAL::Nef_polyhedron_3<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool>::change_orientation(bool) at /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/CGAL/Nef_polyhedron_3.h:1531
#1	0x0000000100858657 in CGAL::Nef_polyhedron_3<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool>::transform(CGAL::Aff_transformation_3<CGAL::Cartesian<CGAL::Gmpq> > const&) at /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/CGAL/Nef_polyhedron_3.h:1789
#2	0x0000000100853b80 in CGAL_Nef_polyhedron::transform(Eigen::Transform<double, 3, 2, 0> const&) at /Users/kintel/code/OpenSCAD/openscad/src/CGAL_Nef_polyhedron.cc:158
#3	0x000000010026dfe6 in GeometryEvaluator::visit(State&, TransformNode const&) at /Users/kintel/code/OpenSCAD/openscad/src/GeometryEvaluator.cc:604
```

MichaelAtOz, 2016-05-30:
Reproduced in Windows.
OpenSCAD 2015.03-2-64 & 2016.05.12-64

Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	openscad.exe
  Application Version:	0.0.0.0
  Application Timestamp:	564d2b59
  Fault Module Name:	openscad.exe
  Fault Module Version:	0.0.0.0
  Fault Module Timestamp:	564d2b59
  Exception Code:	c0000005
  Exception Offset:	0000000000e0eebd
  OS Version:	6.1.7601.2.1.0.256.1
  Locale ID:	3081
  Additional Information 1:	ebab
  Additional Information 2:	ebabc9d1c0d4746cb7dc148995826f88
  Additional Information 3:	13b8
  Additional Information 4:	13b8cda15070c75ff22f24ccd1c7194d

thehans, 2016-05-31:
If you render it with the mirror operation commented out, it works but gives:
WARNING: Object may not be a valid 2-manifold and may need repair! 

If you turn on edges in the rendered view you can see that there is a edge that doesn't belong to any face of the mesh.
[image]

doug-moen, 2016-05-31:
The inner difference() is returning a mesh that, if converted from rational
to floating point, would probably become non-2-manifold due to loss of
numeric accuracy. Based on earlier experience with this kind of code.

This mesh is then transformed using a floating point transform matrix. Are
we converting the mesh from rational to floating point at this point?

Then, we take the result of translate() and feed it into the outer
difference(). I'm guessing that we are converting a non-manifold mesh to a
NefPolyhedron at this point. I think that the NefPolyhedron has a flag
telling us if it is non-manifold, so we should check that flag and abort
(without crashing) at this point. Ideally, we print a filename/linenumber
of the operation that produced the bad result, at the earliest time
possible, so the user knows where in the script the corruption has occurred.

That's all guesswork. The output of the outer difference() is definitely
non-manifold, and the warning that Hans saw indicates we know it's
non-manifold. We should check and abort at that point.

Looks like mirror() will not tolerate a non-manifold input.

MichaelAtOz, 2016-05-31:
The original has a problem here;
[image]

If you change the 2nd r= to 3.99, ie
```
mirror() {
  difference() {
    cube(size = [2, 15, 18]);
    translate([6, 15, 6]) {
      #difference() {
        cylinder(h = 9, r = 10);
        cylinder(h = 9.5, r = 3.99, $fn = 20);
      }
    }
  }
}
```
Then it renders OK.

kintel, 2016-05-31:
We simply pass an affine transformation matrix to CGAL. This looks like either a bug in CGAL, or a weakness in what types of transformations they accept. I think one decent way forward is to export the inner difference to a .nef3 file, and write (or adapt from CGAL examples) a stand-alone cmd-line tool to demonstrate the issue.

..then, for ourselves, find a way to avoid calling into CGAL when it's known to crash, e.g. by performing a manifold check. That might cause a bunch of valid cases to fail though, so we may want to test that a bit more than this case.

kintel, 2016-06-07:
See https://github.com/CGAL/cgal/issues/1139

closed 2016-06-13 by kintel