openscad/openscad #1656: Question: How to improve preview speed?

justbuchanan, 2016-06-01:
I apologize if this has been asked before, but I've been searching around and haven't found a solution.  When I open a semi-large model in openscad ([Cyclone PCB](https://github.com/CarlosGS/Cyclone-PCB-Factory) for example), the preview mode is unusably slow.  If I try to use the mouse to view from a different angle, it takes several seconds to actually move and I'm on a fairly powerful desktop PC.  However, if I switch to "thrown together" mode, panning and zooming are very fast.  Is there anything I can do to speed up preview mode?  Or is this just a limitation of openscad?  Thanks!

kintel, 2016-06-01:
#1518 is work in progress to improve raw rendering speed.

However, sometimes, even maximum raw rendering speed won't give fast enough results since we do all CSG computation in real-time rather than as a preprocess.
Another solution is to pre-process subtrees by using the ```render()``` module (especially subtrees containing subtractions or intersections which are reused multiple times). Down the road we could also look into opportunistic pre-processing to further improve preview rendering performance.

Rendering using F6 should result in a full-performance rendering, but that is also a time-consuming preprocessing job.

justbuchanan, 2016-06-01:
Thanks for the quick reply!

I tried using `render()` to pre-process a complex component that I have replicated throughout my design and preview mode is very fast now.  Thanks a bunch!

closed 2016-06-01 by justbuchanan