openscad/openscad #1660: Missing non union operator to stack 2D objects

mrvn, 2016-06-04:
I'm using openscad to generate dxf files for laser cutting and I'm trying to cut these shapes (simplified example, view from top):

[test.txt](https://github.com/openscad/openscad/files/298721/test.txt)

The idea is that I'm cutting pieces (green) that will later be assembled into an object. I also need pieces (red) that assemble into a box holding all the small parts (not included) prior to assembly. Building a box that can contain the green parts would require a larger box and extra wood. Instead I want to use the green pieces as part of the box.

I could cut the red piece as difference (blue piece) but that would require placing it far enough from the green piece and waste wood. Instead I want to cut both the green and red piece as a single object as seen when viewed from top.

Note: One way to implement this would be that 2D objects at different heights (as above) are not merged into a single area. When rendering to 2D / exporting to dxf duplicate / overlaping lines should be eliminated though.

t-paul, 2016-06-04:
See also #1044.

doug-moen, 2016-06-05:
I think the extended `color()` feature of #1608 would solve this problem. With this feature, the red and green object both lie in the XY plane and touch each other, but aren't merged into a single area because they are different colours.

mrvn, 2016-06-05:
but wouldn't that mean the dxf then has duplicate lines?

jordanbrown0, 2026-05-06:
This really seems like a duplicate of #99, the need to handle non-solid 2D objects.