openscad/openscad #1661: Dependencies on Ubuntu Xenial

EiNSTeiN-, 2016-06-04:
I ran into a couple issues compiling OpenSCAD on Xenial (16.04), so I thought I'd report them here.

I ran `./scripts/uni-get-dependencies.sh` and then `./scripts/check-dependencies.sh` as per the readme, but there are a few issues with it:
* I have Qt 5.6, and QT_VERSION_STR is located in `qconfig.h` but the check-dependencies script looks in `qglobal.h` so it fails to detect the version number.
* libcgal-qt5-dev libopencsg-dev libglew-dev libeigen3-dev were needed but not installed by ./scripts/uni-get-dependencies.sh
* the uni-get-dependencies script tries to install libXi-dev but the name is libxi-dev

leoheck, 2016-07-02:
Oh, I just saw this now... and it is related to the issue #1725. Sorry for duplicate it but I always forget the Ubuntu nasty names, I prefer to use numbers instead.

donbright, 2016-08-12:
please try again i just updated the build script. thanks.

t-paul, 2019-11-21:
Fixed, script is used for building AppImages: https://github.com/openscad/docker-openscad/blob/master/appimage/appimage-x86_64-base/Dockerfile

closed 2019-11-21 by t-paul