openscad/openscad #1662: Separate differences in the same plane interfer in preview

korpx, 2016-06-04:
Why does multiple differences separate from eachother (difference() {...} difference() {...}) interfer with eachother in preview mode? It makes it hard to do sensible sectional cuts for example.

The interfering "object" does not remove anything from the other, which is to be expected of course but it colors the intersecting face weirdly.

Example with interference;
```
color("Yellow")
difference() {
	// Main shape
	cylinder(d=20,h=10);
	
	// Center hole
	translate([0,0,-0.1])
	cylinder(d=16,h=10+(2*0.1));
	
	// Section cut
	translate([-20/2,-20/2,-0.1])	//22
	cube([20,10,10+(2*0.2)]);
}
color("Green")
difference() {
	// Main shape
	cylinder(d=12,h=10);
	
	// Center hole
	translate([0,0,-0.1])
	cylinder(d=6,h=10+(2*0.1));
	
	// Section cut
	translate([-20/2,-20/2,-0.1])
	cube([20,10,10+(2*0.2)]);
}
```
[image]

Rendering it gives the result to be expected, without any strange interference, but is very time consuming on complicated models especially if changes need to me made so re-rendering is needed.

```
color("Yellow")
render()
difference() {
	// Main shape
	cylinder(d=20,h=10);
	
	// Center hole
	translate([0,0,-0.1])
	cylinder(d=16,h=10+(2*0.1));
	
	// Section cut
	translate([-20/2,-20/2,-0.1])	//22
	cube([20,10,10+(2*0.2)]);
}
color("Green")
render()
difference() {
	// Main shape
	cylinder(d=12,h=10);
	
	// Center hole
	translate([0,0,-0.1])
	cylinder(d=6,h=10+(2*0.1));
	
	// Section cut
	translate([-20/2,-20/2,-0.1])
	cube([20,10,10+(2*0.2)]);

}
```
[image]

Another a bit more complicated example;
[image]

nophead, 2016-06-05:
I suspect it is a consequence of the way OpenCSG works. I work around it by adding render() to all my top level objects like you did.

Other ways to avoid it in your example are to move the yellow cutting plane back 0.001 or reduce the size of the cube that cuts the green part so it doesn't overlap the yellow.

korpx, 2016-06-05:
I agree the problem could easily be worked around in the example, and in my designs I use that method by declaring a "shim" constant equal to 0.1 which I use in all overlapping situations.

But I find that in even moderately complicated designs I tend to spend a surprising amount of time "shimming" things get rid of all interfering faces and for very irregular objects mating with other objects render() seems to be the only sanity preserving way to do it. But the re-renders are so slow. It's okay for a nearly finished design but for a design in heavy development it's a pain. A solution under the hood would have been so nice.

openjscad does not have this problem as far as I can see. Different underlying technologies I assume.

korpx, 2016-06-05:
No shims

[image]

nophead, 2016-06-05:
render() is only slow the first time after a change. After that it improves speed, so for large designs I need to use it anyway.

I don't have render() in the modules that define my objects. That means that when I work on them individually they preview quickly but when I assemble them into a view I render them at that point, so they don't interfere with each other.

Another side effect of the way OpenCSG works is that the negative objects need to be inside the clipping volume for it to work. If you zoom in closer than your section cube the view will break. Adding render fixes that as well.

kintel, 2016-06-05:
Looks like a duplicate of #1000

korpx, 2016-06-05:
I agree. I didn't see it when searching for similar issues.

tinkerology, 2016-06-06:
I have another example of this in gist [HullBetweenIssue,scad](https://gist.github.com/tinkerology/ea6268d41bc31d676cd819a9b5459dc4). If you take out the render on line 30, the resulting object does not appear to be manifold, shows verious issues in Meshlab and does not slice properly.  The easiest way to see the problems in OpenSCAD is by viewing the wireframe.

I'm running version 2016.02.09 (git 9950e6a).

MichaelAtOz, 2016-06-06:
> I'm running version 2016.02.09 (git 9950e6a).

I've had various issues with that version, suggest you get the latest snapshot.

kintel, 2016-06-07:
@tinkerology Your file looks like a totally different issue; it's related to rendering and export to STL, not preview. Perhaps you should report this as a separate issue so we can track it?

kintel, 2016-06-07:
Closing as duplicate of #1000 (I copied this example into that issue).

closed 2016-06-07 by kintel