openscad/openscad #1664: Export to STL includes degenerate triangles and folds.

ioquatix, 2016-06-10:
Hello,

When exporting this mesh: https://github.com/ioquatix/zonestar-p802q-fan-duct/blob/master/source/duct.scad

I get a mesh which has multiple issues as reported by FreeCAD analyze and repair. Is this a bug? I'm going to do some more investigation with meshlab.

t-paul, 2016-06-10:
I'm not getting any issues with 2015.03-1 on Linux (checking with blender and meshlab).

Please add the relevant information listed in ["How do I report bugs?"](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs.3F)

ioquatix, 2016-06-10:
@t-paul I'm using 2015.03-3 running on Arch Linux.

Do you mind giving me a copy of the .stl file you are generating?

ioquatix, 2016-06-10:
[image]

t-paul, 2016-06-10:
Generated STL: [duct.zip](https://github.com/openscad/openscad/files/309459/duct.zip)

AdMesh reports some fixed normal vectors, but that could be Zero-Area faces (which OpenSCAD itself does not like much, but are normally no issue)

```
================= Results produced by ADMesh version 0.98.2 ================
Input file         : duct.stl
File type          : ASCII STL file
Header             : solid OpenSCAD_Model
============== Size ==============
Min X = -27.000000, Max X =  27.000000
Min Y = -27.000000, Max Y =  42.000000
Min Z =  0.000000, Max Z =  70.000000
========= Facet Status ========== Original ============ Final ====
Number of facets                 :  4374                4374
Facets with 1 disconnected edge  :     0                   0
Facets with 2 disconnected edges :     0                   0
Facets with 3 disconnected edges :     0                   0
Total disconnected facets        :     0                   0
=== Processing Statistics ===     ===== Other Statistics =====
Number of parts       :     2        Volume   :  14247.848633
Degenerate facets     :     0
Edges fixed           :     0
Facets removed        :     0
Facets added          :     0
Facets reversed       :     0
Backwards edges       :     0
Normals fixed         :   162
```

ioquatix, 2016-06-10:
I wonder why freecad is reporting so many folds then... hmm

ioquatix, 2016-06-10:
Here is what I am generating... seems like it has issues.. I was trying to use it in OpenFOAM and it reported some issues too..

[duct.stl.tar.gz](https://github.com/openscad/openscad/files/309467/duct.stl.tar.gz)
[image]

ioquatix, 2016-06-10:
This is what I got running admesh

```
 admesh duct.stl
ADMesh version 0.98.2, Copyright (C) 1995, 1996 Anthony D. Martin
ADMesh comes with NO WARRANTY.  This is free software, and you are welcome to
redistribute it under certain conditions.  See the file COPYING for details.
Opening duct.stl
Checking exact...
All facets connected.  No nearby check necessary.
No unconnected need to be removed.
No holes need to be filled.
Checking normal directions...
Checking normal values...
Calculating volume...
Verifying neighbors...

================= Results produced by ADMesh version 0.98.2 ================
Input file         : duct.stl
File type          : ASCII STL file
Header             : solid OpenSCAD_Model
============== Size ==============
Min X = -27.000000, Max X =  27.000000
Min Y = -27.000000, Max Y =  42.000000
Min Z =  0.000000, Max Z =  70.000000
========= Facet Status ========== Original ============ Final ====
Number of facets                 :  4328                4328
Facets with 1 disconnected edge  :     0                   0
Facets with 2 disconnected edges :     0                   0
Facets with 3 disconnected edges :     0                   0
Total disconnected facets        :     0                   0
=== Processing Statistics ===     ===== Other Statistics =====
Number of parts       :     2        Volume   :  14249.413086
Degenerate facets     :     0
Edges fixed           :     0
Facets removed        :     0
Facets added          :     0
Facets reversed       :     0
Backwards edges       :     0
Normals fixed         :   159
```

It looks okay.. perhaps FreeCAD is broken.. it's reporting dengerate faces but admesh isn't right?

t-paul, 2016-06-10:
Also NetFabb does not report errors for that file, so it might be the FreeCAD import causing the issues.

ioquatix, 2016-06-10:
Okay, let me check the mesh in OpenFOAM..

ioquatix, 2016-06-10:
okay it will take a few hours to generate the mesh I will report back tomorrow thanks so much for your prompt reply.

ioquatix, 2016-06-15:
This is simply an issue with FreeCAD. My apologies for consuming your valuable time. i checked the mesh using tools from OpenFOAM and it looks good.

closed 2016-06-15 by ioquatix