openscad/openscad #1666: openscad crashing with segfault when generating this mesh

ioquatix, 2016-06-15:
When trying to render this mesh, openscad crashes. It might be a bug with my code so I will check it but I thought I'd add it here just in case it's an actual issue..

https://github.com/ioquatix/zonestar-p802q-fan-duct/blob/a44654e414a1e1e8ba73939a163b7eed38e93730/source/duct.scad

Requires other files in that directory to run.

ioquatix, 2016-06-15:
Tried twice:

```
[88644.540827] openscad[19880]: segfault at 10 ip 0000000000673122 sp 00007fff6b3a4770 error 4 in openscad[400000+4f1000]
[88670.838953] openscad[21611]: segfault at 10 ip 0000000000673122 sp 00007ffee2960b30 error 4 in openscad[400000+4f1000]
```

t-paul, 2016-06-15:
Even with a bug in the scad code, there should be no crash. This is with Arch/2015.03-3 as mentioned in #1664?

Could you please attach a ZIP with the full collection of required source files that reproduce the issue?

ioquatix, 2016-06-15:
Yes, same system. Here are the files. It's set up to crash, just open `duct.scad`. 

[crash.tar.gz](https://github.com/openscad/openscad/files/316178/crash.tar.gz)

ioquatix, 2016-06-15:
I get an error from CGAL when running on the command line:

```
beemo% openscad duct.scad -o test.stl
CGAL Cache insert: difference(){cylinder($fn=0,$fa=5,$fs=0. (387792 bytes)
CGAL Cache insert: render(convexity=1){difference(){cylinde (387792 bytes)
CGAL Cache insert: group(){render(convexity=1){difference() (387792 bytes)
CGAL Cache insert: multmatrix([[1,0,0,0],[0,1.42308,0,0],[0 (387792 bytes)
CGAL Cache insert: intersection(){multmatrix([[1,0,0,0],[0, (102736 bytes)
CGAL Cache insert: intersection(){multmatrix([[1,0,0,15],[0 (102616 bytes)
CGAL Cache insert: union(){multmatrix([[1,0,0,5],[0,1,0,0], (97456 bytes)
CGAL Cache insert: render(convexity=1){union(){multmatrix([ (97456 bytes)
CGAL Cache insert: group(){render(convexity=1){union(){mult (97456 bytes)
CGAL Cache insert: multmatrix([[1,0,0,0],[0,1,0,26],[0,0,1, (97456 bytes)
CGAL Cache insert: group(){intersection(){multmatrix([[1,0, (102736 bytes)
CGAL Cache insert: multmatrix([[-1,0,0,0],[0,1,0,0],[0,0,1, (97456 bytes)
CGAL Cache insert: union(){group(){linear_extrude(height=1, (592336 bytes)
CGAL Cache insert: group(){multmatrix([[1,0,0,0],[0,1,0,0], (379024 bytes)
CGAL Cache insert: difference(){union(){group(){linear_extr (669448 bytes)
CGAL Cache insert: multmatrix([[1,0,0,0],[0,1,0,0],[0,0,1,1 (669448 bytes)
CGAL Cache insert: render(convexity=1){multmatrix([[1,0,0,0 (669448 bytes)
CGAL Cache insert: group(){render(convexity=1){multmatrix([ (669448 bytes)
ERROR: CGAL error in CGALUtils::applyBinaryOperator UNKNOWN: CGAL ERROR: assertion violation!
Expr: !(h.a()==0 && h.b()==0 && h.c()==0 && h.d()==0)
File: /usr/include/CGAL/Nef_S2/Normalizing.h
Line: 289
CGAL Cache insert: minkowski(convexity=0){group(){render(co (1415080 bytes)
CGAL Cache insert: render(convexity=1){minkowski(convexity= (1415080 bytes)
CGAL Cache hit: group(){render(convexity=1){multmatrix([ (669448 bytes)
CGAL Cache insert: union(){render(convexity=1){minkowski(co (1415080 bytes)
zsh: segmentation fault (core dumped)  openscad duct.scad -o test.stl
```

I wonder if it's because I'm generating two points on top of each other in my spiral extrusion.

ioquatix, 2016-06-15:
okay if I replace the first module with

```
function spiral_lerp(t, a = inner_radius, b = outer_radius) =
	lookup(t, [
		[0, inner_radius+0.1],
		[360, outer_radius]
	]);
```

IT works fine. It's because it's generating two points at the same location sequentially and I guess somehow it's causing some internal inconsistency.

kintel, 2016-06-16:
@ioquatix Yes, that would fix the issue.
FYI: I don't see a crash with the latest snapshot (or any snapshot I've tried later than January 2016), so I assume the crash-bug has been fixed.
Performing a minkowski sum of a non-manifold is still not supported by CGAL though.
We should add a minimal example of this for future testing, unless that's already done in another issue.

ioquatix, 2016-06-16:
Is the spiral non-manafold? I exported a mesh and checked with admesh and it seemed fine, but still causes crash.

kintel, 2016-06-16:
@ioquatix I'm not sure admesh can detect non-manifold surfaces. The spiral is definitely non-manifold; it touches itself, creating an infinitely thin surface as one point.

ioquatix, 2016-06-17:
@kintel Well, I did get it work eventually, the exact same shape, but slightly different method:

```
function spiral_lerp(t) =
	lookup(t, [
		[0, inner_radius],
		[360, outer_radius]
	]);

module spiral(step_size = 10) {
	render()
	difference() {
		linear_extrude(height=outer_height)
		polygon(points=
			[for(t = [360:-step_size:0])
				[spiral_lerp(t)*sin(t),spiral_lerp(t)*cos(t)]]
		);
		
		cylinder(h=outer_height, r=inner_radius);
	}
}
```

Is this also non-manifold?

kintel, 2016-06-17:
@ioquatix Your new design should work, but might be non-manifold for some corner cases. It should work as long as you stay inside OpenSCAD. Once you export to STL, we lose some information and your success depends on which tools you're using downstream.

ioquatix, 2016-06-19:
@kintel you are right, sometimes the STL export is non-manifold. How can I make this shape manifold? I don't mean for them to be sharing the same point. Should I split it into two halves? I don't quite understand why this is a problem or how to fix it  :(

kintel, 2016-06-19:
It's manifold because you have an infinitely thin section on the spiral wall. This basically means that its physical representation is ambiguous. To fix, make sure that the wall either has a small gap, or that it has a minimum thickness, e.g. by making your subtracted cylinder a bit smaller.

ioquatix, 2016-06-19:
@kintel Thanks, I fixed the issue, I didn't cut out the middle of the spiral until much later in the design, after the minkowski and it didn't generate (obviously because there is no zero-width point) a non-manifold mesh. Thanks for your help.

ioquatix, 2016-06-19:
I'm happy to close this issue since the crash looks like it was already fixed in git, and it's up to you guys if you want a minimal test (e.g. the spiral).

closed 2016-06-19 by ioquatix