openscad/openscad #1668: windows 7: screen realestate wasted on top of editor + console (2015.03-2)

sgserg, 2016-06-18:
It's quite clear what the view mean, so their description is rather redundant.
The space, on the other hand, is wasted.
Please see red markings on attached screenshots.
[image]
[image]

t-paul, 2016-06-18:
This is not redundant, it's the handle to drag the windows to other places.

You can disable that in "Preferences->Advanced"

closed 2016-06-18 by t-paul

sgserg, 2016-06-25:
Thanks for the tip!

Not sure I've seen such UI element usage under MS Windows, btw.
Usually, to mark a viewport dock-able and its docking options, well-known icons are used, e.g. debugging window of Firefox:

[image]