openscad/openscad #1678: morphological erosion, closing and opening in addition to minkowski

hm17gh1, 2016-06-24:
would it be possible to have support for the morphological erosion, closing and opening in OpenSCAD ?

doug-moen, 2016-06-24:
As far as I know, morphological erosion, dilation, closing and opening can
all be defined in terms of minkowski sum and set difference, both of which
we have. You should be able to define these yourself. However, this is
outside my area of expertise.

I'm curious, what is your use case for these operations? 3D or 2D?

hm17gh1, 2016-06-24:
I was thinking about 3d cases, but having both 2d and 3d variants would be nice. I think that would be helpful to have them readily available for use, rather than re-implementig each time they are needed. This can be as a included "library", but better as an internal operation, like minkowski, so the person defining a 3d geometry focuses on the main task, not on simulating closing or erosion (or anything) with the minkowsky  (which requires tricks like negative volumes/ holes, etc, which can be cumbersome to implement).

hm17gh1, 2016-06-25:
I'm using this super-slow workaround, but it would be nice to have this built in, and faster....

module morph_dilation(){
    render() minkowski(){
        children(0);
        children(1);
    }
}

module morph_erosion() {
    render() difference() { 
        children(0);
        minkowski() { 
            difference() { 
                minkowski() { 
                    children(0);
                    cube([1,1,1], center=true);
                }
                children(0);
            }
            children(1);
        }
    }
}

module morph_opening(){
    morph_dilation(){
        morph_erosion(){
            children(0);
            children(1);
        }
        children(1);
    }
}
    
module morph_closing(){
    morph_erosion(){
        morph_dilation(){
            children(0);
            children(1);
        }
        children(1);
    }
}

doug-moen, 2016-06-27:
Erosion would be faster if a way could be found to write it using only one call to `minkowski`. There are other implementations of Erode on the OpenSCAD forums that use only one call to `minkowski`, but those seem to be a special case of Erode where the second argument is a radius, not a shape.

There is a forum discussion thread about adding support for "degenerate shapes" like points, paths and surfaces, and using these shapes as arguments to 3D `minkowski`. These degenerate shape operands would be faster than the regular case of 3D shape arguments. @hm17gh1, in your implementation of erosion, the expression
```
         difference() {
            minkowski() {
               children(0);
               cube([1,1,1], center=true);
            }
            children(0);
         }
```
is intended to extract the surface boundary of `children(0)`, and if we had degenerate shape support, this could be replaced by a fast O(1) operation, `boundary()children(0)`. Then, the outer call to `minkowski` would be operating on a degenerate shape, and would run much faster.

The general problem here is that our implementation of Minkowski Sum is slow, both for preview and render, although rendering has become much faster over time for the special case of convex shapes. Simply asking for these operations to be "built in" doesn't make anything go faster unless we can find faster algorithms to use. If we have a faster Minkowski, there may be no need for the other operations to be coded in C++.

My guess is that minkowski sum and the other morphological primitives would be quite fast in preview if we used functional representation (F-Rep) plus ray-marching for previewing. I know of lots of special cases that are fast under F-Rep, and it makes sense that the general case could be made fast as well. But I don't know that for sure without writing a prototype, and I may get around to testing that. (Of course, this implies replacing our OpenCSG based previewer with an all new design.)

Ccing @ronaldopersiano

hm17gh1, 2016-06-28:
The degenerate shape support sound like a very useful thing to have.
F-Rep is something I have read about just after reading your message.

In the mean time, I have tried this (dirty) solution as an 1-dilation implementation of erosion:
(faster, but still slow for complex shapes, and has the disadvantage that mxsize exists)

```
module morph_erosion(mxsize=1000) {
    render() difference() { 
        children(0);
        minkowski() { 
            difference() { 
                cube([mxsize,mxsize,mxsize], center=true);
                children(0);
            }
            children(1);
        }
    }
}
```

amca01, 2017-11-08:
Has there been any movement on this in the last year?  I'm new to OpenSCAD, but I'd like these operations very much.

kintel, 2017-11-08:
Nothing has happened in terms of implementing faster, built-in functions. Did you try the code from this thread though?

amca01, 2017-11-08:
Not yet!  I especially like the idea of 3d morphological opening; it'd be a neat way of - for example - rounding the edges of a box.

unpokeh, 2018-12-24:
I believe rounding the edges of a box can be done using filleting instead of erosion. Not an expert, but just sayin'.