openscad/openscad #1681: creating CTestCustom.cmake : add_test problems

blobule, 2016-06-25:
When building the test environment on Linux (cd tests;cmake .)
I get strange errors:
-- creating CTestCustom.cmake
CMake Error at CMakeLists.txt:1028 (add_test):
  add_test given test NAME "dumptest-examples_children" which already exists in this directory.
Call Stack (most recent call first):
  CMakeLists.txt:1524 (add_cmdline_test)

CMake Error at CMakeLists.txt:1028 (add_test):
  add_test given test NAME "cgalpngtest_children" which already exists in
  this directory.
Call Stack (most recent call first):
  CMakeLists.txt:1525 (add_cmdline_test)

etc...

I can' t find what is wrong, and this is making it impossible to do any testing before I submit a patch.  Does anyone has an idea what is going on?

t-paul, 2016-06-25:
What distribution / Which version? What cmake version? Does it happen with just master?

Maybe also try cleaning the cmake files
```
rm -rf CMakeFiles/ CMakeCache.txt
```

blobule, 2016-06-25:
Linux Mint kernel 3.13.0-24, 32 bit.
cmake 2.8.12.2

These are all the errors...

  add_test given test NAME "dumptest-examples_children" which already exists  in this directory.
   add_test given test NAME "cgalpngtest_children" which already exists in this directory.
   add_test given test NAME "opencsgtest_children" which already exists in  this directory.
   add_test given test NAME "csgpngtest_children" which already exists in this   directory.
   add_test given test NAME "throwntogethertest_children" which already exists

It is strange because I can' t find any directory or file with these names...

t-paul, 2016-06-25:
Hmm, Ubuntu 12.04 has cmake 2.8.7, so I suspect that's what's used on Travis and it's working there too.

kintel, 2016-06-25:
hm, hard to debug remotely, but here are some hints at what to search for:

In your error messages, the “_children” corresponds to a file in the examples/ folder; examples/Advanced/children.scad.
All files in the examples folder needs to have a unique basename. You could try searching through that folder - perhaps there is a stray “childrens.scad” there somewhere.

If that doesn’t help, try adding this line in CMakeLists.txt:1028, just before the main add_test() call.
      message(STATUS ${TEST_FULLNAME})

..then run "cmake . | grep examples_children”, and see if there are actually any duplicate tests.

 -Marius

blobule, 2016-06-26:
hmmmmm... yes, I had added a few temporary examples and one was named children.scad.
I removed it and now it works. Thanks for the suggestion.
Now I'll do some testing... :-)

closed 2016-06-26 by blobule