openscad/openscad #1721: STL file error

swethmandava, 2016-06-29:
minkowski(){
    import("windmill_meshlab.stl", complexity=10);
    sphere(2);
}

ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: e->incident_sface() != SFace_const_handle()
File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h

None of my stl files seem to work. Can't figure out why, is there something I'm missing?
Meshlab says no non manifold vertices or edges. I tried netfabb as well. attached all three stl files

[stl_models.zip](https://github.com/openscad/openscad/files/339605/stl_models.zip)

MichaelAtOz, 2016-06-30:
I'm not sure what this is, but it doesn't look good: (the...repaired.stl)
[image]
The other STLs have it too, and other problems.
Unfortunately CGAL is fussy about such things.
Also note the number of shells, some of the vanes are not connected, that could affect what you are trying to do to it.

swethmandava, 2016-06-30:
I fixed this. still doesn't work :(
[windmill.off.zip](https://github.com/openscad/openscad/files/340763/windmill.off.zip)

MichaelAtOz, 2016-06-30:
That zip file just has a .off file?

closed 2018-10-22 by swethmandava