openscad/openscad #1723: Pow bug

RonaldoCMP, 2016-07-01:
echo(pow(-8,1/3));

nophead, 2016-07-01:
See http://www.cplusplus.com/reference/cmath/pow/

RonaldoCMP, 2016-07-01:
Incredible! After all, pow(-2,3) = -8. The arithmetic of C++ is something else.

RonaldoCMP, 2016-07-01:
I agree it is not an original OpenSCAD bug but it is a bug anyway.
Incredible!

t-paul, 2016-07-01:
It's not just C++, looks like other languages inherit that too instead of having a different interpretation.

```
$ python
Python 2.7.5 (default, Jun 24 2015, 00:41:19) 
>>> pow(-8, 1.0/3.0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: negative number cannot be raised to a fractional power
```

closed 2016-07-01 by t-paul

nophead, 2016-07-01:
Probably because they all use logarithms to do fractional powers. Also,
mathematically, I don't think it is defined except for some special cases.
Yes you can do a cube root of a negative number but 1/3 is 0.333333333...
I.e. not exactly a cube root, so what would the correct answer be?

doug-moen, 2016-07-01:
It's not a bug in OpenSCAD or in C++. It's a problem with the exponential
function itself.

If you want to specify arbitrary real valued arguments (negative bases,
fractional exponents), then the exponential function is only closed over
the complex numbers, not over the real numbers. For example, sqrt(-4), aka
pow(-4, 1/2) has no real-valued solution, but does have a complex solution.
Furthermore, for a given set of arguments, there may be multiple correct
answers. For example, each complex number has 2 square roots, 3 cube roots,
and so on. So we have to choose a convention for deciding which of these
multiple answers to return. According to the standard convention, the
principal square root of 4 is 2 (not -2), and the principal cube root of -8
is 1 + sqrt(3)i, and not -2.

There's no mathematical way to define the 'pow' function to work correctly
and predictably for arbitrary arguments and also behave the way you want
for this specific case.

What I think you need instead is a dedicated cube root function.

function cuberoot(x) = pow(abs(x), 1/3) * sign(x);

RonaldoCMP, 2016-07-01:
Ok, ok, touché. I agree: it is credible! After all, pow(.) receives double
and returns double. Thank you all for the clarification.