openscad/openscad #1724: Usability Improvments

leoheck, 2016-07-01:
Hello all, 

OpenScad is a great tool that I discover recently!

It would be nice to have more improved usability features:
- GUI: Shortcuts (save, save as, ...)
- GUI: Improved/extra preview and render shortcuts
- GUI: White Icons for dark ~~menu~~ themes
- EDITOR: Autocomplete commands
- EDITOR: Toogle comments shortcut (ctrl+/)
- DOC: Offline cheat sheet.

What do you think guys?

t-paul, 2016-07-01:
* GUI: Shortcuts (save, save as, ...)
What's wrong with `CTRL-S` and `CTRL-SHIFT-S`? Or the menu mnemonics `ALT-F` + `S` / `A`?
Are those not working for you?

* GUI: Improved/extra preview and render shortcuts
Could you explain what to improve here?

* GUI: White Icons for dark menu themes
Can you attach a screenshot? Actually there should be white icons, but the dark theme detection is a bit tricky so maybe that does not catch all cases.

* EDITOR: Autocomplete commands
Yep...
* EDITOR: Toogle comments shortcut (ctrl+/)
...and yep, see http://forum.openscad.org/Abbreviations-td17583.html and https://github.com/openscad/openscad/pull/905 - I think those do not cover comment folding yet, but that might be easy to add

* DOC: Offline cheat sheet.
Full offline docs is on the wishlist too, but a bit stuck as the integration into the build process needs to be solved. The scripts scrapes the online documentation and we don't want to do that for every build.
Maybe we can have a simpler solution for the cheat sheet first.

We probably want to split this issue up a bit once clarified...

leoheck, 2016-07-02:
Oh, let me see, 

Just to be crystal clear, I am running Ubuntu 16.04, and I downloaded OpenScad 2015.03-1 from  PAA.

1. Hmm, good, so shortcuts don't work for me! I thought it was a missing stuff.
2. I mean, new (or extra) shortcuts to render/preview. F5 and F6 are too far from typing. It will be nice a shortcut like ``ctrl+shit+somthing`` (or something similar)

3. I am using numic-gtk-theme which has dark menus, but this can also happen to the Ubuntu's default theme. Take a look:
[image]

4. Autocomplete :+1: 
5. I think you mixed some comments here.. but ctrl+/ shortcut to toggle comments is always a good idea like to select multiple lines and indent/unindent.

6. I can help you do that. Do you have some pointers to see that? Maybe I can give you an idea. It is possible to create a makefile for this process.. and add an entry to that in the main build process... 

7. **[NEW]** How to remove this toolbar? I don't use it. And, I think there is no needed to write "Editor" or "Console" in the panel's header. This is a waste of space.

[image]

8. **[NEW]** There is a shortcut to reload file edited externally? I am editing with sublime text for now, because it is far more flexible.

WZ9V, 2016-07-02:
Try building 2015.03-2. The PPA is behind and -2 has several improvements such as syntax highlighting actually working.

leoheck, 2016-07-02:
First step to compile it from the sources... Issue #1725 (done)

leoheck, 2016-07-02:
Someone can point me the script (command, folder, or something) that is used to built offline documentation from the online docs as @t-paul said?

t-paul, 2016-07-03:
https://github.com/runsun/openscad_offliner
http://forum.openscad.org/Use-openscad-offliner-for-offline-documentation-td13096.html

leoheck, 2016-07-03:
I have created a way to generate the offline documentation only once. I put that on ``openscad.pro`` file. I just have to know how I can implement a ``ifdef`` to select from parsing it online or use the .zip file. I will make a pull request for that.

t-paul, 2016-07-03:
Can you explain what your strategy is? E.g. the "nightly" builds are triggered for every check-in for 15 distributions, most having both 32 and 64 bit. Those always clean builds as they run on fresh VMs. We can't really have 30 scrapers running there for every github check-in.
Having an optional way to run the documentation build from `openscad.pro` might be useful nevertheless.

leoheck, 2016-07-03:
Ok, let me see if I can explain.

I have created a new target to build the docs, (which I call ``docs``). The procedure I am using is shown below. Note: I can use the ``ONLINE_DOCS`` flag to trigger if it will parse the content online or just to use the github ``openscad_docs.zip``. For both, it will always clone the ``openscad_offliner`` repo. This is made only once. And this is fast, at least for me.

```
qmake [ONLINE_DOCS=yes]
make docs
```

I thought we could add this target (``docs``) to the main target to build all stuff at the same time.

Considering you said, I think we could have 2 main building targets. The final ``make [all]`` which includes the latest documentation. And another light target for periodically builds like ``make nightly`` which can skip building documentation.

What you think?

I had finished the process editing ``openscad.pro``. I just added this at the end:

```qmake

QMAKE_EXTRA_TARGETS += docs openscad_docs openscad_offliner doc_clean

docs.target = docs
docs.depends = openscad_offliner/openscad_docs

openscad_docs.target = openscad_offliner/openscad_docs
openscad_docs.depends = openscad_offliner

# ONLINE_DOCS =
isEmpty(ONLINE_DOCS) {
openscad_docs.commands = @ \
  cd openscad_offliner; \
  git pull origin master > /dev/null; \
  echo '"Unpaking stable documentation"'; \
  unzip -q openscad_docs.zip;
} else {
openscad_docs.commands = @ \
  cd openscad_offliner; \
  git pull origin master > /dev/null; \
  echo '"Donwloading online documents"'; \
  python openscad_offliner.py | tee openscad_offliner.log;
}

openscad_offliner.target = openscad_offliner
openscad_offliner.commands = @ \
  echo '"Cloning openscad_offliner"'; \
  rm -rf openscad_offliner; \
  git clone https://github.com/leoheck/openscad_offliner.git > /dev/null

doc_clean.target = clean-docs
doc_clean.commands = @ \
  rm -rf openscad_offliner

```

nairaner, 2016-07-06:
There already are white icons for dark themes. I guess you have to use Qt theme.
[image]

t-paul, 2020-02-16:
A number of things are now implemented. Closing this now as it's not really possible to track partial progress in one overarching issue.

closed 2020-02-16 by t-paul