openscad/openscad #1725: Script uni-get-dependencies.sh dont' install all dependencies

leoheck, 2016-07-02:
Running on Ubuntu 16.04.

Why do not to add all stuff into the script?

[image]

Installing missing regular packages:

```
sudo apt install libcgal-dev
sudo apt install libeigen3-dev
sudo apt search install libopencsg-dev
sudo apt search install libopencsg1v5
```

And, installing ``OpenCSG``:

```
git clone https://github.com/openscad/OpenCSG.git
cd OpenCSG
make 
sudo make install
```

[image]

t-paul, 2016-07-02:
It's supposed to add all packages, it's currently simply not possible to catch up with lots of distributions.

leoheck, 2016-07-02:
After a loong make process...

``itstool`` is a new missing dependency.

Solved with:

```
sudo apt install itstool
```

Terminal output... 

```
Compiling language files...
msgfmt -c -v -o ./locale/fr/LC_MESSAGES/openscad.mo ./locale/fr.po
268 translated messages.
msgfmt -c -v -o ./locale/ru/LC_MESSAGES/openscad.mo ./locale/ru.po
268 translated messages.
msgfmt -c -v -o ./locale/de/LC_MESSAGES/openscad.mo ./locale/de.po
268 translated messages.
msgfmt -c -v -o ./locale/cs/LC_MESSAGES/openscad.mo ./locale/cs.po
266 translated messages, 2 untranslated messages.
msgfmt -c -v -o ./locale/es/LC_MESSAGES/openscad.mo ./locale/es.po
225 translated messages, 41 fuzzy translations.
itstool missing, won't apply translations to openscad.appdata.xml
```

t-paul, 2016-07-02:
Yes, it's optional, so no error message. It's mainly interesting for distribution packaging.

WZ9V, 2016-07-02:
The Debian 7 dependency rule is missing also when running on Mint 17 or 18.  On Mint 18 I just had to install the dependencies myself in order to build 2015.03-2.  By the way the PPA still has 2015.03-1 that has syntax highlighting missing in the editor.

donbright, 2016-08-12:
please try again i just updated the ubuntu16 get dependencies script thanks.

kaefert, 2017-04-14:
Hey there! I'm running Linux Mint 18.1 (based on Ubuntu 16.04) and had the following problems:
after running `sudo ./scripts/uni-get-dependencies.sh` the dependency `fontconfig` was still reported as missing by the `check-dependencies.sh` script.
Installed it manually by running
`sudo apt-get install libfontconfig1-dev`

Afterwards my first run of
`qmake openscad.pro` reported `No package 'libzip' found`
had to install that manually with
`sudo apt-get install libzip-dev`

currently I'm waiting on the results of `make` :)

t-paul, 2019-11-22:
Fixed, script is used for building AppImages: https://github.com/openscad/docker-openscad/blob/master/appimage/appimage-x86_64-base/Dockerfile

closed 2019-11-22 by t-paul