openscad/openscad #1727: Preview not showing up after error

nairaner, 2016-07-04:
Everything works correctly until there is an issue in the code, then it stops showing preview, but it is still being rendered as you can see performance hit when rotating viewport. Happens with both Goldfeather and without it.

t-paul, 2016-07-04:
Please give some more details as mentioned in "[How do I report bugs](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs.3F)".

nairaner, 2016-07-04:
I'm using latest git version on Archliux.
OpenGL 3.3
```
include <MCAD/stepper.scad>;

module plate(nemaStandard = Nema17)
{
    motorDiameter=nemaStandard[NemaSideSize][1];
    
    screwDistance=(nemaStandard[NemaSideSize][1]-
    nemaStandard[NemaDistanceBetweenMountingHoles][1])/2;
    
    difference()
    {
        minkowski()
        {
            cube([motorDiameter-6,motorDiameter-6,2]);
            cylinder(2,3,3);
        }
    translate([screwDistance-3,screwDistance-3,-0.1])
        cylinder(4.2,2,2);
    translate([motorDiameter-screwDistance-3,screwDistance-3,-0.1])
        cylinder(4.2,2,2);
    translate([screwDistance-3,motorDiameter-screwDistance-3,-0.1])
        cylinder(4.2,2,2);
    translate([motorDiameter-screwDistance-3,motorDiameter-screwDistance-3,-0.1])
        cylinder(4.2,2,2);
    translate([motorDiameter/2-3,motorDiameter/2-3,-0.1])
        cylinder(h=4.2,r=nemaStandard[NemaRoundExtrusionDiameter][1]/2+1);
    }
}
plate();
```

MichaelAtOz, 2016-07-04:
`until there is an issue in the code` What do you mean by that? If there is an issue, i.e. bug, then it doesn't have anything to preview.
The code above works ok for me.

nairaner, 2016-07-04:
@MichaelAtOz I mean if there is any kind of the problem with compiling, then after fixing it preview dont show anything.
https://www.youtube.com/watch?v=N4JPlzeSYgk

MichaelAtOz, 2016-07-04:
The video shows v 20160630, later than the Development snapshot.
I'm on windows anyway, but the latest snapshot 20160608 does not have the issue.
Needs others on *nix to test.

nairaner, 2016-07-05:
Like i said i am on latest git version. I was able to narrow down this bug a little bit.

```
module part()
{
    translate([0,0,5])
    cube(4);
    mirror([0,0,1])
    translate([0,0,5])
    cube(4);    
    cylinder(r=4,h=10,center=true);
}

difference()
{
    part();
    cube(1); //change this
}
```

t-paul, 2016-07-05:
Nice, trimmed down examples help a lot. I think I saw this effect too, but could not really pinpoint the cause. I'll have a look if that example code triggers that for me too...

nairaner, 2016-07-06:
Also it seems like F12 preview works, but not F9.

nairaner, 2016-07-06:
Bug gets also triggered by resizing the preview, here's the example:
https://www.youtube.com/watch?v=gcUGOHI3SNc

t-paul, 2016-07-06:
Could you add the text from "Help -> Library Info" to see what library versions are used? I wonder if that issue got dragged in by a Qt upgrade.

Would you be able to test this with a manually built version against Qt4?

nairaner, 2016-07-06:
It doesn't happen on Qt4.
```
OpenSCAD Version: 20160630
System information: Linux 4.6.3-1-ARCH #1 SMP PREEMPT Fri Jun 24 21:19:13 CEST 2016 x86_64 Antergos Linux 4 CPUs 7.49 GB RAM
Compiler: GCC "6.1.1 20160602" 64bit
Boost version: 1_60
Eigen version: 3.2.8
CGAL version, kernels: 4.8, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.7.0
QScintilla version: 2.9.2
MingW build: No
GLib version: 2.48.1
Application Path: /usr/bin
Documents Path: /home/nairaner/.local/share
Resource Path: /usr/share/openscad
User Library Path: /home/nairaner/.local/share/OpenSCAD/libraries
User Config Path: /home/nairaner/.config/OpenSCAD
Backup Path: /home/nairaner/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/nairaner/.local/share/OpenSCAD/libraries
/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.13.0
OpenGL Version: 3.0 Mesa 11.2.2
GL Renderer: Mesa DRI Intel(R) Ivybridge Mobile 
GL Vendor: Intel Open Source Technology Center
RGBA(8880), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QOpenGLWidget

GL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_performance_monitor
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_trinary_minmax
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_vertex_array_object
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
[... 164 lines cut ...]
GL_INTEL_performance_query
GL_KHR_context_flush_control
GL_KHR_debug
GL_MESA_pack_invert
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
```

lewisjb, 2016-07-26:
I also get this problem: https://gist.github.com/lewisjb/551d60b5dfdf585d0fccd0a0a96522ce

donbright, 2016-09-09:
i agree, what @nairainer and @lewisjb are describing sounds like what i encountered in 1785

video here

https://www.youtube.com/watch?v=CYO-nqs6oiQ

nairaner, 2016-10-07:
Fixed with OpenCSG 1.4.2

closed 2016-10-07 by nairaner

karlkashofer, 2017-06-09:
This bug is present in Debian Jessie and Stretch, a update is in the works via -backports, and here is a workaround:
http://karl.kashofer.org/?p=339