openscad/openscad #1730: Opening dxf files

jaredabrandt001, 2016-07-06:
I'm trying to use OpensCad to convert PCB DXF's to extrudes and trouble.  OpensCad will open KiCad DXF's without an issue but that's the only program that OpensCad can open DXF's from (effectively).  Any idea why, and suggestions on a converter that might be compatible with OpensCad?
Thanks,
Jared

jaredabrandt001, 2016-07-07:
I wanted to post an update.  I've found that reaConverter DXF files will open in Openscad effectively.
I'm working with some reasonably intricate DXF's, and Openscad will open and render 1 single DXF quickly (usually in a few seconds).
Unfortunately anytime I try to open multiple DXF's, the render process is painfully long.  The preview operates in a few seconds, but rendering is taking several hours.  Like I mentioned, these same DXF's render in just a few seconds when I'm only rendering one at a time.  This additional time to render pretty much kills the usability of Openscad in this application:/  It seems like there's something going on under the hood that's not compatible with multiple DXF's.

Some background info:
I'm using Openscad to create .stl files for 3D printed circuit boards.  One .stl for the conductive filament, one .stl for the insulating filament.  A PCB may contain many layers (one DXF input per layer), which ultimately gets combined into 2 .stl's (one filled and the other the inverse).  The funny thing is, mechanically my 3D PCB process seems quite sound.  Processing an existing PCB (as 2 total .stl files) seems to be the crux of the project thus far, which is sad, and I'd assume easily overcome-able with a customized program.  
If each layer is taking 2 hours to create, I don't believe Openscad is usable in this application.  Maybe there are some special settings I'm missing, like part accuracy or maybe the program is writing to disk when it could be in RAM?  Do you have any ideas?

Thanks in advance for the help!
Jared

Oh...I guess this info is needed:
OS: Windows 7
OpenSCAD version 2015.03-2
Dell precision, 8-core 3.73ghz, Nvidia Quadro 2gb, 32gb ram
python version: no idea (not sure if OpenScad uses python)
ps: I'm an engineer, not a programmer...aka, I can program but not in depth.

Neon22, 2016-07-07:
I still recommend - from a practicality point of view - loading the dxf in inkscape and saving as openscad using the extension:
- http://www.thingiverse.com/thing:1065500
DXF is an old highly variable format with variable suport of lines,polylines, more complex curves. Typically he reader for it is cranky and difficult to make work with all possible variations of thw format.
Until the time is taken to make yet another DXF loader perfect - I recommend using one that has been worked on a lot.

I don't like most of the online converters as they break everything into discrete lines as its easier to code. but sometimes you want the curves to be curves. YMMV

jaredabrandt001, 2016-07-08:
Ok, I will give that extension a try, and be back in touch.  I've used an older version of that extension which will not generate the scad file from a 2400 dpi .png file.  For some reason it only works when I scale the png down.  I think Inscape or the extension is having trouble with the high resolution.
I presume I could include multiple .scad files to stack the PCB's and generate the .stl's using this method.  If the time to render the stack of .scad files is significantly shorter, I suppose we have success:)

jaredabrandt001, 2016-07-08:
...and to your point: 
"DXF is an old highly variable format with variable suport of lines,polylines, more complex curves." 
-Yes, DXF's are a pain in the butt; however I can't find anything else universal to both 3D CAD and PCB editors.

Ideally I'd like to read Gerber and go direct to STL, but I've hit nothing but dead-ends with that.  I've found some 3rd part programs that get me partially there, but typically they leave the image messy and un-editable.  
If this project gets funded I'll probably pay a programmer to write code to convert the files directly to stl, or better yet, STEP or IGS.

t-paul, 2016-07-08:
I'm going to close this issue, for discussions please use the [forum](http://forum.openscad.org/).

Feel free to open a new issue including examples if there's a specific issue.

Although it's a known problem that the DXF importer does not cover all types of entities, so depending on the generating program parts of the imported model might be missing. There are already issues for that (e.g. see #1307 and [project page](https://github.com/openscad/openscad/wiki/Project%3A-Improve-DXF-import-and-export)).

closed 2016-07-08 by t-paul

jaredabrandt001, 2016-07-08:
I tried the new inkscape extension, and it took considerably longer to render than the reaConverter output DXF.

I suppose I'd like to understand why OpenScad takes so 10+ min to render 2 totally non-intersecting DXF's, but seconds for each individual DXF.  This is likely a fault in OpenScad programming, which to my understanding has not before been documented.
Would OpenScad perform better if somehow the DXF's were contained within a single .scad file, then apply functions by using include<>  ?

t-paul, 2016-07-08:
How would we find out without having a chance to check a specific example?

2D operations are usually very fast (handled by the Clipper library). 3D operations with lots of vertices can be very slow as libCGAL does not seem to have much optimization for the cases we use it for.

jaredabrandt001, 2016-07-08:
I would think the issue would be easy to identify with nearly any two DXF's.  I would upload them, but this page will not allow me to upload a .zip file.  Are your github settings not allowing .zips?

jaredabrandt001, 2016-07-08:
Well, this is the .zip file, renamed to .png to get through your system.

[image]

t-paul, 2016-07-08:
Now what? Both files seem to import fine. Using both at the same time as 3D objects and combining them with CGS operations will be slow due to the large vertex count. What is this going to be? What is the OpenSCAD file trying to do?

jaredabrandt001, 2016-07-08:
The individual .dxf files here render in under 1s alone.  Rendering both (not intersecting) is taking my machine over 20 min and it's not finished.

I am simply trying to display both files at the same time, in a reasonable rendering time.  The objective would be to stack, for example 16 layers together, or have some way to convert the 16 layers to a single .stl.  I also need an inverse of this for each layer, which consists of a DXF for the board outline, minus the geometry included in the DXF's I sent you.  I explained this above already.  The icing on the cake would be to have something which could assemble hundreds of layers in a reasonable processing time (under a few hours).

I need these DXF's to combine into the 2 .stl files in a reasonable amount of time.  I don't know the inner programming of Openscad, but I would bet my house that the vertex count is not why this is taking so long.  I could combine both those DXF's into a single DXF, and OpenScad would probably render it in under 1 second.  

So the question is again: "Why does rendering more than 1 DXF in one .scad file take an incredibly long time to render?"

jaredabrandt001, 2016-07-08:
To check, I've imported 2 very simple DXF's (low vertex count) and rendering these together takes 1000's of times longer than the individual files.  Openscad shows the same inability to render 2 dxf's in one .scad file with a low vertex count.  Something in the program is just not effective at displaying multiple dxf's.

t-paul, 2016-07-08:
A single DXF file will not cause any CSG operation, it will just use the generated geometry. So that's basically never going to need any time. But even adding a single `cube()` would cause a considerable slowdown as an actual CSG operation is executed.

The current 3D CSG engine (libCGAL) does not handle large vertex numbers as fast as we would like, the hope is to find an alternative - so far there is none. There might be some options to speed-up  special cases, e.g. non-intersecting geometry, but up to now, nobody actually tried that.

jaredabrandt001, 2016-07-08:
hmmm, yea I see that if I add a cube, and one DXF the render times get awful.
So if I add multiple .scad files (previously DXF's) to an assembly would that invoke a CSG operation?  Any way around this?  I don't need cubes, but will need other DXFs.

Neon22, 2016-07-09:
shoudl probably move to forum as peple may be able to help you. I add this piece of info.
1. The dxf on load has a surplus of detail. Alas the default inkscape simplify is too agressive to maintain the shape.

2. The individual lines are not nested into entities. Here you can see I have manually combined the paths of the outline and holes to make a fillable shape. The openscad converter would work better with this.
Sorry - no immediate one click solution... :(
[image]
[image]

t-paul, 2016-07-09:
With the release version, there's not much chance to make it fast for combining multiple 3d objects with lots of vertices.

There's a branch that tries to skip at least the top-level union, which might help in this specific case.

Otherwise, it would need some optimizations to cover at least the non-intersecting case. Right now the dev resources are a bit limited, so I don't know if and when that happens. There was a long discussion about possible solutions on the forums lately. One mentions additional checks provided in libCGAL that maybe could be called before running the CSG union (although it would be much better if libCGAL would actually do that internally without bothering the application using the CSG operations).