openscad/openscad #1732: "Export as image" includes the menu dialog in the picture

rapenne-s, 2016-07-08:
Hello,

using the export to image menu exports the image with the menu on it which is not the expected behaviour when exporting an image.

[Picture of the problem here](http://i.zcraft.fr/938871467969821.png) (this is the result of the **Export as image**)

I use stumpwm window manager, which isn't a composite window manager (if that helps).

```
OpenSCAD Version: 2015.03-2
System information: OpenBSD 6.0 GENERIC.MP#2241 amd64 4 CPUs 7.91 GB RAM
Compiler: GCC "4.2.1 20070719 " 64bit
Boost version: 1_58
Eigen version: 3.2.2
CGAL version, kernels: 4.6.3, N4CGAL9CartesianINS_4GmpqEEE, N4CGAL18Extended_cartesianINS_4GmpqEEE, N4CGAL5EpeckE
OpenCSG version: OpenCSG 1.3.2
Qt version: 4.8.7
QScintilla version: 2.8
MingW build: No
GLib version: 2.48.1
Application Path: /usr/local/bin
Documents Path: /home/solene/.local/share
Resource Path: /usr/local/share/openscad
User Library Path: /home/solene/.local/share/OpenSCAD/libraries
User Config Path: /home/solene/.config/OpenSCAD
Backup Path: /home/solene/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/solene/.local/share/OpenSCAD/libraries
/usr/local/share/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.12.0
OpenGL Version: 3.0 Mesa 11.2.2
GL Renderer: Mesa DRI Intel(R) Haswell Mobile 
GL Vendor: Intel Open Source Technology Center
RGBA(8880), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget

GL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_trinary_minmax
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_vertex_array_object
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pipeline_statistics_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robustness
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counters
GL_ARB_shader_bit_encoding
GL_ARB_shader_clock
GL_ARB_shader_draw_parameters
GL_ARB_shader_image_load_store
GL_ARB_shader_image_size
GL_ARB_shader_objects
GL_ARB_shader_storage_buffer_object
GL_ARB_shader_texture_image_samples
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_barrier
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_draw_buffers
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_texture
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_gpu_program_parameters
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_polygon_offset
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shader_integer_mix
GL_EXT_shader_samples_identical
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texture
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_IBM_multimode_draw_arrays
GL_IBM_rasterpos_clip
GL_IBM_texture_mirrored_repeat
GL_INGR_blend_func_separate
GL_KHR_context_flush_control
GL_KHR_debug
GL_MESA_pack_invert
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
```

triffid, 2016-07-08:
This affects me also with openscad-2015.07.03, using kwin WM with compositing disabled.

See http://i.imgur.com/BA58BLE.png

kintel, 2016-07-08:
This is because "Export as Image" simply dumps the framebuffer of the 3D view to a file. If the view doesn't have its own buffer (e.g. using a non-composite WM), any overlay is likely to be visible.

Note: This can be solved by rewriting the image export to use an offscreen renderer. This is a bit tricky to get right across platforms, so it will take time unless someone new picks it up.