openscad/openscad #1734: Error in stl export

lfourrier, 2016-07-13:
I have a model :

MichaelAtOz, 2016-07-14:
Netfabb shows holes and degenerate faces in the stl.

I cut your code down to;
```
nbpas=35;
$fn=36;
module empreinte(){
		union(){
			for(t=[0:360/nbpas:360-360/nbpas]){
				rotate([0,0,t])rotate([36,0,0])
				translate([-6,-1,-5])cube([20,2,80.5]);
			}
			for(t=[0:360/nbpas:360-360/nbpas]){
				rotate([0,0,t])rotate([-36,0,0])
				translate([-6,-1,-5])cube([20,2,80.5]);
			}
		}
}
empreinte();
```
Which still has the problem; (note nbpas=35 shows issue more than 36)
[orange=degenerate, yellow=holes]
[image]

I suspect it doesn't like all the edges converging in that central point.
Tho IMO it should handle that.

I'm testing with an older version, but render is taking forever...
Latest dev version shows same problem.

lfourrier, 2016-07-21:
I understand (a little :-) ) the difficulties to correct the problem. And I can probably, I hope, live with it, by restructuring my models. But can we have some indicator of the problem, inside openscad, and without having to use netfabb ?
My tests, changing the rotated cubes for them to not converge in one point, seems to present the same problem of invalid model. Do you have any suggestions to prevent invalid models ?

kintel, 2016-07-21:
@lfourrier I think you're experiencing two related issues:

1) OpenSCAD doesn't clean STL output before _exporting_. Our geometry engine (CGAL) operates with higher precision than floating point numbers written to STL files can offer. We may lose some information or introduce issues when converting this, if you have very dense geometry. See #13 and #1621

2) OpenSCAD is slightly picky about _importing_ STL (well, CGAL is), and not all exported files are possible to re-import. See #945 and #1042

kintel, 2024-03-27:
Status as of 2024-03-24: This works when using Manifold (and as a bonus, renders in 0.4 seconds rather than 6 minutes).