openscad/openscad #1735: boost link issue

ec1oud, 2016-07-14:
I suspect that building with Qt 5 should be possible by now despite what the readme says, so I'm giving it a try.  The first issue I ran into is this link error (unrelated to Qt):

`g++ -Wl,-O1 -Wl,-O1,--sort-common,--as-needed,-z,relro -o openscad objects/scintillaeditor.o objects/scadlexer.o objects/AST.o objects/ModuleInstantiation.o objects/expr.o objects/function.o objects/module.o objects/UserModule.o objects/version_check.o objects/ProgressWidget.o objects/linalg.o objects/Camera.o objects/handle_dep.o objects/value.o objects/stackcheck.o objects/func.o objects/localscope.o objects/feature.o objects/node.o objects/context.o objects/modcontext.o objects/evalcontext.o objects/csgnode.o objects/CSGTreeNormalizer.o objects/CSGTreeEvaluator.o objects/Geometry.o objects/Polygon2d.o objects/clipper-utils.o objects/polyset-utils.o objects/GeometryUtils.o objects/polyset.o objects/polyset-gl.o objects/csgops.o objects/transform.o objects/color.o objects/primitives.o objects/projection.o objects/cgaladv.o objects/surface.o objects/control.o objects/render.o objects/text.o objects/dxfdata.o objects/dxfdim.o objects/offset.o objects/linearextrude.o objects/rotateextrude.o objects/printutils.o objects/fileutils.o objects/progress.o objects/parsersettings.o objects/stl-utils.o objects/boost-utils.o objects/PlatformUtils.o objects/LibraryInfo.o objects/nodedumper.o objects/NodeVisitor.o objects/GeometryEvaluator.o objects/ModuleCache.o objects/GeometryCache.o objects/Tree.o objects/DrawingCallback.o objects/FreetypeRenderer.o objects/FontCache.o objects/settings.o objects/rendersettings.o objects/highlighter.o objects/Preferences.o objects/OpenCSGWarningDialog.o objects/editor.o objects/GLView.o objects/QGLView.o objects/AutoUpdater.o objects/grid.o objects/hash.o objects/GroupModule.o objects/FileModule.o objects/builtin.o objects/calc.o objects/export.o objects/export_stl.o objects/export_amf.o objects/export_off.o objects/export_dxf.o objects/export_svg.o objects/export_nef.o objects/export_png.o objects/import.o objects/renderer.o objects/colormap.o objects/ThrownTogetherRenderer.o objects/svg.o objects/OffscreenView.o objects/fbo.o objects/system-gl.o objects/imageutils.o objects/lodepng.o objects/openscad.o objects/mainwin.o objects/OpenSCADApp.o objects/WindowManager.o objects/UIUtils.o objects/Dock.o objects/FontListDialog.o objects/FontListTableView.o objects/launchingscreen.o objects/legacyeditor.o objects/LibraryInfoDialog.o objects/clipper.o objects/bucketalloc.o objects/dict.o objects/geom.o objects/mesh.o objects/priorityq.o objects/sweep.o objects/tess.o objects/imageutils-lodepng.o objects/OffscreenContextGLX.o objects/OpenCSGRenderer.o objects/cgalutils.o objects/cgalutils-applyops.o objects/cgalutils-project.o objects/cgalutils-tess.o objects/cgalutils-polyhedron.o objects/CGALCache.o objects/CGALRenderer.o objects/CGAL_Nef_polyhedron.o objects/cgalworker.o objects/Polygon2d-CGAL.o objects/PlatformUtils-posix.o objects/qrc_openscad.o objects/moc_scintillaeditor.o objects/moc_ProgressWidget.o objects/moc_QGLView.o objects/moc_MainWindow.o objects/moc_OpenSCADApp.o objects/moc_WindowManager.o objects/moc_Preferences.o objects/moc_OpenCSGWarningDialog.o objects/moc_AboutDialog.o objects/moc_FontListDialog.o objects/moc_FontListTableView.o objects/moc_highlighter.o objects/moc_editor.o objects/moc_Dock.o objects/moc_AutoUpdater.o objects/moc_launchingscreen.o objects/moc_legacyeditor.o objects/moc_LibraryInfoDialog.o objects/moc_cgalworker.o objects/parser_yacc.o objects/lexer_lex.o   -lCGAL -lmpfr -lgmp -lopencsg -lGLEW -lboost_thread -lboost_program_options -lboost_filesystem -lboost_system -lboost_regex -lglib-2.0 -lharfbuzz -lfontconfig -lfreetype -L/usr/lib -lqscintilla2-qt5 -lQt5PrintSupport -lQt5Widgets -lQt5Gui -lQt5Concurrent -lQt5Core -lGL -lGLU -lX11 -lpthread 
objects/handle_dep.o: In function `__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > boost::re_detail_106000::re_is_set_member<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, char, boost::regex_traits<char, boost::cpp_regex_traits<char> >, unsigned int>(__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, __gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, boost::re_detail_106000::re_set_long<unsigned int> const*, boost::re_detail_106000::regex_data<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, bool)':
handle_dep.cc:(.text._ZN5boost16re_detail_10600016re_is_set_memberIN9__gnu_cxx17__normal_iteratorIPKcNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEjEET_SH_SH_PKNS0_11re_set_longIT2_EERKNS0_10regex_dataIT0_T1_EEb[_ZN5boost16re_detail_10600016re_is_set_memberIN9__gnu_cxx17__normal_iteratorIPKcNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEjEET_SH_SH_PKNS0_11re_set_longIT2_EERKNS0_10regex_dataIT0_T1_EEb]+0x1a0): undefined reference to `boost::re_detail_106000::cpp_regex_traits_implementation<char>::transform_primary(char const*, char const*) const'
handle_dep.cc:(.text._ZN5boost16re_detail_10600016re_is_set_memberIN9__gnu_cxx17__normal_iteratorIPKcNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEjEET_SH_SH_PKNS0_11re_set_longIT2_EERKNS0_10regex_dataIT0_T1_EEb[_ZN5boost16re_detail_10600016re_is_set_memberIN9__gnu_cxx17__normal_iteratorIPKcNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEjEET_SH_SH_PKNS0_11re_set_longIT2_EERKNS0_10regex_dataIT0_T1_EEb]+0x342): undefined reference to `boost::re_detail_106000::cpp_regex_traits_implementation<char>::transform(char const*, char const*) const'
collect2: error: ld returned 1 exit status
make: *** [Makefile:835: openscad] Error 1
`

Googled and found http://stackoverflow.com/questions/17588440/cannot-link-boost-regex which says it's because of libraries being linked in the wrong order.

ec1oud, 2016-07-14:
Why do you use boost regex instead of Qt?  Of course, when porting to Qt 5 you're supposed to use QRegularExpression, not QRegExp, so then it's not possible to build on Qt 4 anymore.

ec1oud, 2016-07-14:
But I'm on Arch Linux and have held back gcc to 5.7 because 6.1.1 causes some trouble, and something like this was fixed by rebuilding boost for gcc 6, so maybe the installed boost package is now incompatible with gcc 5.7.

https://bugs.archlinux.org/task/49248

ec1oud, 2016-07-14:
So nevermind, I rebuilt on another machine which has the gcc upgrade, and it works fine there.

closed 2016-07-14 by ec1oud

t-paul, 2016-07-14:
The OpenSCAD core does not depend on Qt, so it's not possible to use QRegExp/QRegularExpression or other Qt classes everywhere.