openscad/openscad #1738: editor section - text distortion

digitalvapor, 2016-07-16:
[image]

ubuntu studio 16.04. openscad is version `2015.03-1`. using `functions.scad` as an example after opening.

let me know if you need any other information, or know a section of the docs to point me to! thanks

t-paul, 2016-07-16:
Please check `Edit` -> `Preferences` -> `Editor` -> `Font` - which font is that?

Initial font selection is based on Qt font matching, which seems to have failed. I'm not sure if there's an easy way to find out why it returned this font.

It might help to have a look at the full list of installed font, but that can be quite long:
```
fc-list -v : | bzip2 -9 > /tmp/font-list.gz
```

digitalvapor, 2016-07-16:
thanks for the quick reply!

Editor Type: QScintilla Editor
Font: 10.15 Saturday Night BRK, 12
Color syntax highlighting: For Light Background

I swapped to good old `Monospace` and it looks normal again.

paulirotta, 2019-02-28:
[image]

I'm seeing similar distortion in Windows 10 when changing from laptop to external monitor. QScintilla editor, Courier 10.

This doesn't happen with Simple Editor.

t-paul, 2019-02-28:
That sounds much more related to #1624.

This one is about selecting the wrong font on Linux and I believe it's actually fixed by https://github.com/openscad/openscad/commit/cc50662036ae33a033247d3c180e9d4ac03be690 so I'm closing this issue here.

closed 2019-02-28 by t-paul