openscad/openscad #1740: SVG import issues

kintel, 2016-07-19:
This issue tracks remaining issues related to the experimental SVG import

TODO:
- [ ] Add some svg import test cases
- [ ] Clarify how to handle intersection of multiple imported paths
- [ ] Test on Windows
- [ ] Test lazy union export (see #350)
- [ ] Look into applying DPI when importing px or unitless designs

Might be considered as well:
- [ ] Viewport clipping -> I don't think this is important (@kintel)

subatomicglue, 2017-11-10:
So... to enable this, I would do this on the command line?  (MacOSX):  `/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD --enable='svg-import'`

I see the experimental `svg-import` [here](https://github.com/openscad/openscad/blob/master/src/feature.cc)

stuartpb, 2018-02-01:
Issue: SVGs are imported with centering, while DXFs are not.

kintel, 2018-02-01:
TODO list updated

stuartpb, 2018-02-01:
There's also the matter that imported SVGs, if centered, should (optionally) be centered on the center of the *viewbox*, not the center of the *paths*.

I'd handle this with something like a `viewbox` option that defaults to `true`, with this behavior:

|   | center = true | center = false |
| --- | --- | --- |
| **viewbox = true**  | centered on center of viewbox | bottom-left corner is bottom-left corner of viewbox |
| **viewbox = false** | centered on center of bounding box (current behavior) | bottom-left corner is bottom-left corner of bounding box |

t-paul, 2018-02-01:
What would be the reasoning for ignoring the view box?

t-paul, 2018-02-01:
That will potentially expose random transformations from the graphics program. I'm not sure that's a useful thing to have. I feel that's a big source of problems with random files. Handling SVG as CAD is complicated enough due to the visual nature, which is one of the reasons there's still a big amount of open questions for the import.

stuartpb, 2018-02-01:
> What would be the reasoning for ignoring the view box?

A couple of reasons:

- The viewbox may be defined for image presentation, placing space around the paths that should be ignored for the purposes of centering them (ie. `viewbox=false, center=true`).
- The viewbox may clip things in a certain way for image presentation where, for the paths' use in the SVG, a different kind of intersection is appropriate (ie. `viewbox=false, center=false`).

stuartpb, 2018-02-01:
> That will potentially expose random transformations from the graphics program[...]

I think you might misunderstand what I mean by "toggle clipping". I'm talking about clipping paths within the viewbox, the item currently listed as "Viewport clipping -> I don't think this is important" in the OP.

t-paul, 2018-02-01:
Hmm, so it would be more a `clip = true/false`.

stuartpb, 2018-02-01:
Well, it'd be `viewbox = true/false`, because it'd also affect the coordinate system (translating either the bottom-left or center of the viewbox to [0,0], depending on whether `center` is set or not).

stuartpb, 2018-02-01:
As for DPI, this stems from the `viewbox` as well (relative to the `width` and `height` attributes); however, these can really only be scaled to OpenSCAD units when *the units of OpenSCAD itself* are defined, so I think this should be handled by a third `units` parameter, which defaults to null (ie. no unit conversion / scaling, beyond perhaps aspect ratio scaling when `preserveAspectRatio` is `none`), and can be set to `"mm"` to scale the coordinates within the viewBox to millimeters (or `in` to scale them to inches, etc).

trustfm, 2019-06-22:
the version 2019.05 does not support svg layers import. Is there a way to import individual layers ?

stuartpb, 2019-06-22:
Well, there's no actual concept of "layers" in SVG (apps like Inkscape create their own layers-like abstractions out of specially-tagged `g` group elements), but, if there's a sufficiently-broad use case, OpenSCAD could add a feature like this by introducing a `root` argument to the import that would ignore all elements of an SVG that aren't under a given root

In the meantime, though, you could throw together a quick post-processing command with something like Node.JS's Cheerio: look up the ID attribute of the layer (or whatever object) you wish to individually import, then process the file with something like `node -e 'require("fs").writeFileSync("output_filename_here.svg","<?xml version=\"1.0\" encoding=\"utf-8\"?><svg version=\"1.1\" xmlns=\"http://www.w3.org/2000/svg\" xmlns:xlink=\"http://www.w3.org/1999/xlink\">"+require("cheerio")(require("fs").readFileSync("input_filename_here.svg"))("#id_of_layer_here").html+"</svg>")'`.

marcond05, 2020-03-27:
Not sure if this will help, but couple things I've noticed (2019.05). If you do 

```
$fn=50;
import(file = "coffee-CC0096.svg", center = true, dpi = 96);
import(file = "coffee-1A1A1A.svg", center = true, dpi = 96);
```
using the attached files (rename to .svg), they will result in different scales (one is larger), but they are centered.. That may sound uninteresting, but if you diff those two files, the only thing different is the svg FILL attribute.. The remaining file is identical. This seems related to $fn, which as I understand is an environment setting to adjust resolution/facets. Setting $fn to something divisible by 4 as I briefly saw in the docs they will be the same size.. just odd that the Fill color would cause them to be different at 50? 

Second, I'm often on noun project, and I know if you use their background feature, you'll generate offset issues in OpenScad. For example (also attached rename to .svg)

`import(file = "coffee-offset.svg", center = true, dpi = 96);

This svg will cause the vector to shift right and down (+x, -y)

`<rect width="100%" height="100%" fill="#FFFFFF"></rect>` is the culprit, and you can remove it to re-center.

Illustrator and various apps the icon is still centered in its bounding box, relatively to the box and itself. 

Very new to OpenSCAD but loving it.  Late to the game here, hope that helps.

-Marc

[coffee-1A1A1A.txt](https://github.com/openscad/openscad/files/4391430/coffee-1A1A1A.txt)
[coffee-CC0096.txt](https://github.com/openscad/openscad/files/4391431/coffee-CC0096.txt)
[coffee-offset.txt](https://github.com/openscad/openscad/files/4391432/coffee-offset.txt)

DavidPhillipOster, 2021-06-10:
The python qrcode package pyqrcode generates SVG using a series of short horizontal and vertical line segments.

Example of creating the .svg attached.

```
import pyqrcode
from pyqrcode import QRCode
getqrcode = pyqrcode.create("https://www.google.com")
getqrcode.svg("qrcode.svg", scale=10)
```

resulting in: [qrcode_svg.txt](https://github.com/openscad/openscad/files/6633274/qrcode_svg.txt) (filename changed since github won't let me directly upload a .svg here)

OpenSCAD version 2021.01, given the following .scad file:
```
import("qrcode.svg");
```
does not parse this correctly compared to Inkscape and Google Chrome. `qrcode.svg` could be worked up into a unit test.

t-paul, 2021-06-10:
@DavidPhillipOster could you put that in a separate issue please, that seems a bit unrelated from the scope of this issue.

The path looks certainly a bit funny `d="M4 4.5h7m4 0h2m2 0h2m1 0h1m1 0h1m1 0h7m-29` but I if other tools loading this ok, it's very likely some parsing probem.

chrysn, 2021-10-20:
The SVG import appears to follow the old (pre 1.0) Inkscape convention that the origin is the bottom left corner of the screen plane, X goes right and Y goes up.

This somehow aligns well with the default 3D view, but is confusing to no end when working with actual SVG coordinates (which, since Inkscape 1.0, are also used throughout Inkscape). Especially, to act on coordinates known from the SVG file, one needs to consider the size of the sheet to get to the right coordinates.

When the viewbox topic is revisited, it'd be nice if that would also allow using real SVG coordinates (with or without the mm conversion, I don't care much about that). Things would look "mirrored" at import, but that's OK because we're importing a 2D thing into 3D, and when loading an SVG natively one just needs to view it from negative Z direction to get the right picture (which, not incidentally, should also match the Z order of SVG)